PartialVolume/shredos.x86_64 #1: shredos doesn't work with USB keyboards

PartialVolume, 2020-04-13:
When running shredos on a system with a USB keyboard, the USB keyboard is unresponsive. It does however, work ok if you use a keyboard with a PS2 connector. Also works ok on laptops.

Both this version 2020.02 and the original 18.02 shredos at [https://github.com/nadenislamarre/shredos](https://github.com/nadenislamarre/shredos) both have the same problem.

Being investigated.

@nadenislamarre

PartialVolume, 2020-04-14:
Fixed by #3

closed 2020-04-14 by PartialVolume

vkumarsnn, 2023-11-03:
Hiya, I am trying to wipe the HP AIO 24 df0073na its an intel pentium gold. The ShredOS boots perfectly, however when I am trying to use the keyboard, it doesnt recognizes it. And because its an AIO, there only USB based I/O for Keyboard and Mice.
I have already tried the latest version https://github.com/PartialVolume/shredos.x86_64/releases/download/v2021.08.2_23_x86-64_0.34/shredos-2021.08.2_23_x86-64_0.34_20221231.img,

Also I have tried the earlier test releases in this post https://github.com/PartialVolume/shredos.x86_64/issues/25 but no use.

Can you please let me know if there is any fix for this?

PartialVolume, 2023-11-03:
> Hiya, I am trying to wipe the HP AIO 24 df0073na its an intel pentium gold. The ShredOS boots perfectly, however when I am trying to use the keyboard, it doesnt recognizes it. And because its an [...]

Hi, sounds like a missing driver. Can you run lspci and lsusb in the virtual terminal ALT F2 take photo and post here.  That will tell me what chipset I'm looking for. Thanks.

PartialVolume, 2023-11-03:
@vkumarsnn In the meantime, if you are in a rush to wipe this system you could always wipe it over the network from another PC using this technique https://github.com/PartialVolume/shredos.x86_64#how-to-wipe-drives-on-headless-systems-or-systems-with-faulty-display-hardware-for-use-on-secure-lans-only

vkumarsnn, 2023-11-03:
> Hi, sounds like a missing driver. Can you run lspci and lsusb in the virtual terminal ALT F2 take photo and post here. That will tell me what chipset I'm looking for. Thanks.

Hiya, Thanks for your quick revert, really appreciate that. Can you please provide a PS CLI? its windows with DOS
[image]

I have the ISUSB via ps cli that is attached here

PartialVolume, 2023-11-03:
Unfortunately, that doesn't provide the more detailed output of lsusb that ShredOS/Linux provides. I also would need the output of lspci.

As the keyboard doesn't work, I'd enable telnet server on ShredOS as described in the link above then telnet into ShredOS from another PC    (Windows, Linux or Mac) then run lspci and lsusb, take the snapshots and while you are logged in run nwipe in the terminal to wipe the disc on the HP system.

vkumarsnn, 2023-11-07:
Hiya, Please find below the screenshot via telnet. However, we couldnt run 1) the nwipe 2) lsusb both didnt work

[image]

[image]

[2023/11/07 11:47:20]  notice: processor-frequency = 3400 MHz
[2023/11/07 11:47:20]  notice: Opened entropy source '/dev/urandom'.
[2023/11/07 11:47:20]  notice: hwmon: Module drivetemp loaded, drive temperatures available
[2023/11/07 11:47:20]  notice: hwmon: nvme0n1 has temperature monitoring
[2023/11/07 11:47:20]   error: GUI.c,nwipe_gui_select(), loop runaway, did you close the terminal without exiting nwipe? Initiating shutdown now.
[2023/11/07 11:47:20]  notice: Program options are set as follows...
[2023/11/07 11:47:20]  notice:   autonuke = 0 (off)
[2023/11/07 11:47:20]  notice:   autopoweroff = 0 (off)
[2023/11/07 11:47:20]  notice:   banner   = ShredOS v2021.08.2_23_x86-64_0.34
[2023/11/07 11:47:20]  notice:   prng     = Isaac64
[2023/11/07 11:47:20]  notice:   method   = DoD Short
[2023/11/07 11:47:20]  notice:   quiet    = 0
[2023/11/07 11:47:20]  notice:   rounds   = 1
[2023/11/07 11:47:20]  notice:   sync     = 100000
[2023/11/07 11:47:20]  notice:   verify   = 1 (last pass)

******************************** Error Summary *********************************
!   Device | Pass Errors | Verifications Errors | Fdatasync I\O Errors
--------------------------------------------------------------------------------
********************************************************************************

********************************* Drive Status *********************************
!   Device | Status | Thru-put | HH:MM:SS | Model/Serial Number
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2023/11/07 11:47:22] Total Throughput   0 B/s, DoD Short, 1R+B+VL
********************************************************************************

[2023/11/07 11:47:22]    info: Nwipe successfully completed. See summary table for details.
# 1;0c

PartialVolume, 2023-11-07:
That's great thanks. What error did you get when you typed nwipe in the terminal. It should have displayed in the terminal.

vkumarsnn, 2023-11-07:
> That's great thanks. What error did you get when you typed nwipe in the terminal. It should have displayed in the terminal.
[image]

[2023/11/07 11:47:20]  notice: processor-frequency = 3400 MHz
[2023/11/07 11:47:20]  notice: Opened entropy source '/dev/urandom'.
[2023/11/07 11:47:20]  notice: hwmon: Module drivetemp loaded, drive temperatures available
[2023/11/07 11:47:20]  notice: hwmon: nvme0n1 has temperature monitoring
[2023/11/07 11:47:20]   error: GUI.c,nwipe_gui_select(), loop runaway, did you close the terminal without exiting nwipe? Initiating shutdown now.
[2023/11/07 11:47:20]  notice: Program options are set as follows...
[2023/11/07 11:47:20]  notice:   autonuke = 0 (off)
[2023/11/07 11:47:20]  notice:   autopoweroff = 0 (off)
[2023/11/07 11:47:20]  notice:   banner   = ShredOS v2021.08.2_23_x86-64_0.34
[2023/11/07 11:47:20]  notice:   prng     = Isaac64
[2023/11/07 11:47:20]  notice:   method   = DoD Short
[2023/11/07 11:47:20]  notice:   quiet    = 0
[2023/11/07 11:47:20]  notice:   rounds   = 1
[2023/11/07 11:47:20]  notice:   sync     = 100000
[2023/11/07 11:47:20]  notice:   verify   = 1 (last pass)

******************************** Error Summary *********************************
!   Device | Pass Errors | Verifications Errors | Fdatasync I\O Errors
--------------------------------------------------------------------------------
********************************************************************************

********************************* Drive Status *********************************
!   Device | Status | Thru-put | HH:MM:SS | Model/Serial Number
--------------------------------------------------------------------------------
--------------------------------------------------------------------------------
[2023/11/07 11:47:22] Total Throughput   0 B/s, DoD Short, 1R+B+VL
********************************************************************************

[2023/11/07 11:47:22]    info: Nwipe successfully completed. See summary table for details.

These are the screens we got !! A Blue screen followed by the comand line output as above!

PartialVolume, 2023-11-07:
Sorry, yes I see now. What telnet client are you using? I'd try putty as that looks like an issue with the telnet client you are using.

PartialVolume, 2023-11-07:
Other than putty does anybody else know of any descent terminals that run on windows that work well with ShredOS?

PartialVolume, 2023-11-07:
I'd maybe try a few of these and see which one renders nwipe properly.

https://www.puttygen.com/windows-terminal-emulators

PartialVolume, 2023-11-07:
As I don't tend to use windows much, it would be nice if somebody set up a discussion with a example of the best windows terminal emulators that render ShredOS correctly and a list of those to avoid.

vkumarsnn, 2023-11-07:
> Sorry, yes I see now. What telnet client are you using? I'd try putty as that looks like an issue with the telnet client you are using.

The putty worked for us, thanks!!! Thanks to Suma for the setup!!