PartialVolume/shredos.x86_64 #8: Secure Boot and keeping logs

WhiteDogBe, 2020-06-29:
Hello,

Really love this project, exactly what one expects when you go looking for a tool like this. I've been playing around with it a bit and hit two bumps at the moments:
- The USB key does not boot when Secure Boot is enabled. I think there are already systems out there which don't have Legacy Mode anymore and/or only support Secure Boot. I'm aware that Secure Boot is... complicated. But I was wondering if this is a limitation of buildroot which you are using and if there are any plans to support this in this future?
- Would it be possible to keep the logs permanently on the USB key? I'm aware that you provide instructions but I think it would be a good addition to have this by default somehow on the image you provide (or a more simple way to get the files off).

Keep safe and thanks for your time!

PartialVolume, 2020-06-29:
Thanks, much appreciated.

Regarding secure boot, it looks like a whole load of work to produce a secure boot version. However I don't know enough about secure boot to know for sure. Most people would disable secure boot and/or enable legacy mode on the bios to use ShredOS. I believe some distributions may support secure boot?  But I don't think buildroot does. Never say never but at the moment it's not in any of my plans. (Of course if there's anybody that knows all about secure boot signing with buildroot and is prepared to do the work it would likely get added)

In regards to the permanent keeping of the logs. Now that is a feature I want to add. Persistent storage. Not sure exactly when yet as I'm trying to get the 0.29 version of nwipe ready for official release and there's a few fixes I need to make to that first.

PartialVolume, 2021-01-08:
Working with UEFI secure boot systems as long as secure boot is disabled in the systems UEFI setup (aka bios) screen.

typesolar, 2022-01-31:
Apologies for the bump, but I saw that `shim` is included in /boot now. Are you planning on supporting secure boot in future versions of shredos?
I'm trying to sign the boot firmware so I can use this on secure boot machines in bulk, but preloader doesn't seem to support signing firmware ahead of time. If you have some support for shim planned, then that would be great; I'd offer to help, but I'm not quite sure how to yet.

PartialVolume, 2022-02-01:
Yes, although I've not spent any time looking at what issues need to be overcome in order to implement it. At the moment I'm focused on other parts of ShredOS but if anybody wants to open up a discussion and maybe even do a PR I'd be happy check it out and commit it. Even posting good links to tutorials on the subject would be useful and might change my priorities.

PartialVolume, 2022-02-05:
As a reminder to myself & others: I'm going to place a few links here so I can study this when I have some time:
[https://wiki.debian.org/SecureBoot/Testing](https://wiki.debian.org/SecureBoot/Testing)

PartialVolume, 2022-12-26:
> * Would it be possible to keep the logs permanently on the USB key? I'm aware that you provide instructions but I think it would be a good addition to have this by default somehow on the image you provide (or a more simple way to get the files off).

Currently adding this option to ShredOS, nwipe log files will be automatically copied over to the first FAT32 partition found (which should usually be the ShredOS USB stick. If no FAT32 partition found, such as the case might be if booting ShredOS via the network then the message "No FAT32 formatted drive found, unable to archive nwipe log file" will be displayed on exit from nwipe. The FAT32 formatted drive does not have to be a ShredOS drive, so if you are booting from the network just plug any old FAT32 formatted drive into the computer and ShredOS will copy the nwipe log files over to it on completion of the wipe. For reference the drive needs to be identified as FAT32 when using fdisk. I can open this up to other formats if necessary but a ShredOS thumb drive is identified as FAT32 by fdisk so thats what I've used for the time being.

PartialVolume, 2022-12-29:
> * Would it be possible to keep the logs permanently on the USB key? I'm aware that you provide instructions but I think it would be a good addition to have this by default somehow on the image you provide (or a more simple way to get the files off).

Archiving of the logs to USB has been commited by #121

PartialVolume, 2023-03-18:
In the next release, the PDF reports for each drive will also be sent across to the USB stick at the end of the wipes, along with the log file.

[image]

[image]

PartialVolume, 2023-11-10:
PDFs are also now saved to the USB flash drive [commit](https://github.com/PartialVolume/shredos.x86_64/commit/c9788147f26358ba49a62cf00e0af8d8b8462907)