PartialVolume/shredos.x86_64 #9: shredos doesn't work with PS/2 to USB adapter

basicmaster, 2020-09-29:
Maybe related to #1. I don't have an USB keyboard, but a PS/2 keyboard and a PS/2-to-USB adapter. While this adapter works e.g. with DBAN and in the BIOS, it doesn't work with shredos.

`lsusb` output on deskop Linux:
```
ID 0a81:0205 Chesen Electronics Corp. PS/2 Keyboard+Mouse Adapter
```

`dmesg` output on deskop Linux:
```
[  358.655665] usb 1-1.3: new low-speed USB device number 5 using ehci-pci
[  358.772943] usb 1-1.3: New USB device found, idVendor=0a81, idProduct=0205, bcdDevice= 0.10
[  358.772946] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  358.772948] usb 1-1.3: Product: PS2 to USB Converter
[  358.772950] usb 1-1.3: Manufacturer: CHESEN
[  358.780063] input: CHESEN PS2 to USB Converter as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:0A81:0205.0003/input/input6
[  358.844016] hid-generic 0003:0A81:0205.0003: input,hidraw2: USB HID v1.10 Keyboard [CHESEN PS2 to USB Converter] on usb-0000:00:1a.0-1.3/input0
[  358.851711] input: CHESEN PS2 to USB Converter Mouse as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:0A81:0205.0004/input/input7
[  358.911868] input: CHESEN PS2 to USB Converter System Control as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:0A81:0205.0004/input/input8
[  358.912050] input: CHESEN PS2 to USB Converter Consumer Control as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:0A81:0205.0004/input/input9
[  358.912322] hid-generic 0003:0A81:0205.0004: input,hidraw3: USB HID v1.10 Mouse [CHESEN PS2 to USB Converter] on usb-0000:00:1a.0-1.3/input1
```

PartialVolume, 2020-09-30:
Thanks for reporting, when I get a moment I'll dig out a PS2 keyboard and PS2 to USB adapter and see what's missing from the kernel.

PartialVolume, 2021-01-06:
@basicmaster I'm guessing this is probably working fine on the most recent releases of ShredOS, both i686 and x86_64 versions. Are you still in a position to confirm that?

basicmaster, 2021-01-07:
I tried with the new version "v2020.02.005_x86-64_0.30.001" you released some hours ago, but unfortunately it still doesn't work.

basicmaster, 2021-01-07:
BTW I have a second PS/2 to USB adapter that is affected the same way:

`lsusb` output on deskop Linux:
```
ID 13ba:0017 PCPlay PS/2 Keyboard+Mouse Adapter
```

`dmesg` output on deskop Linux:
```
[ 8655.644420] usb 1-1.4: new low-speed USB device number 8 using ehci-pci
[ 8655.760322] usb 1-1.4: New USB device found, idVendor=13ba, idProduct=0017, bcdDevice= 0.01
[ 8655.760324] usb 1-1.4: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 8655.760325] usb 1-1.4: Product: Generic USB K/B
[ 8655.762792] input: Generic USB K/B as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4:1.0/0003:13BA:0017.0008/input/input14
[ 8655.824711] hid-generic 0003:13BA:0017.0008: input,hidraw2: USB HID v1.10 Keyboard [Generic USB K/B] on usb-0000:00:1a.0-1.4/input0
[ 8655.826747] input: Generic USB K/B Mouse as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4:1.1/0003:13BA:0017.0009/input/input15
[ 8655.884566] input: Generic USB K/B System Control as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4:1.1/0003:13BA:0017.0009/input/input16
[ 8655.884727] hid-generic 0003:13BA:0017.0009: input,hidraw3: USB HID v1.10 Mouse [Generic USB K/B] on usb-0000:00:1a.0-1.4/input1
```

PartialVolume, 2021-01-07:
Thanks for letting me know, I'll order one of those adapters so I've got something to work with. If you recently purchased that adapter, do you have a link to where I might purchase the same one you have?

basicmaster, 2021-01-09:
The older one I bought in 2009, but it still seems to be available (item number is the same):
https://www.pollin.de/p/usb-adapterkabel-720685
(note that it has an annoying bug - it ignores the "<" key on a German keyboard)

I bought the other one a year ago:
https://www.conrad.com/p/renkforce-usb-ps2-keyboardmouse-adapter-1x-usb-11-connector-a-2x-ps2-socket-2000-cm-black-1360251

PartialVolume, 2021-01-11:
I've not yet bought either of those two adapters you mentioned, however I bought this one [UCEC USB to dual PS2 mouse keyboard adapter](https://www.amazon.co.uk/gp/product/B00X9QX9MS/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1).

`  idVendor           0x13ba PCPlay
  idProduct          0x0018 Barcode PCP-BCG4209
`

It works fine on a i7 laptop with either the 32 or 64 bit Shredos, but doesn't work on either a Pentium 4 or dual Xeon processors both of which already have existing PS2 ports. It's a bit of a puzzle at the moment.

basicmaster, 2021-01-11:
Does DBAN work with the P4 and Dual Xeon (I would suppose so)?

With a ThinkPad X220 I was able to use the adapters with Shredos, but with a ThinkPad T60 they don't work (DBAN works in both cases). I guess there is some PS/2 emulation done by the BIOS in case a USB keyboard is connected.

PartialVolume, 2021-01-17:
@basicmaster sorry for the delay getting back to you. I've only just downloaded DBAN and tried it out for the very first time ! I guess that may be surprising since I've been commiting code to nwipe for a couple of years now.

I know I'm biased but ShredOS/nwipe is definitely more refined than DBAN. While they are both very similar it's little things like USB devices being named as USB and not SCSI. Nwipe always shows the drive details, model etc where I found  DBAN doesn't. Nwipe, assuming ATA pass through compatible USB adapters always shows the serial number of drives, including USB drives, DBAN doesnt. Nwipe displays human readable throughout values like 25Mb/s where DBAN shows 25,000 kB/s. You can blank nwipes display during wiping , but not with DBAN. You can't gracefully abort a DBAN wipe other than switch off. With nwipe you can, a control-C is trapped by nwipe and it exits gracefully showing the user aborted wipes in the summary table. You can restart nwipe easily without rebooting, DBAN you can't. You just get that Blancco advert at the end. DBAN didn't seem to have any help info regarding options, unlike nwipe which has nwipe --help.

One thing I did like with DBAN was when it finished the wipe, it gave you the option to save logs to named USB drives. I'm going to add that feature to ShredOS however it should also allow FTP, SFTP and SMTP transfers as well as writing to disc and not only limited to USB devices. That should go nicely with the PDF drive certificates it will eventually issue.

But I digress, I will try DBAN on the P4 and Xeon and see if it works with the USB to PS2 adapter I have and let you know.

PartialVolume, 2021-01-22:
@basicmaster I'm releasing a new version of ShredOS over the weekend or early next week that will have about 70 vendor specific human interface device drivers (HIDs) . I'm hoping this will fix the USB-PS2 adapter issue.

PartialVolume, 2021-01-23:
@basicmaster Can you try this new release. My USB to PS2 adapter that previously was not working now works so hopefully your adapters now work too.

[https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2020.05.007_x86-64_0.30.001](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2020.05.007_x86-64_0.30.001)

PartialVolume, 2021-01-24:
@basicmaster New version now available for testing [https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2020.05.008_x86-64_0.30.001](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2020.05.008_x86-64_0.30.001)

basicmaster, 2021-01-28:
> I know I'm biased but ShredOS/nwipe is definitely more refined than DBAN. While they are both very similar it's little things like USB devices being named as USB and not SCSI. Nwipe always shows the [...]

In the past I was forced to use DBAN (instead of shredos), as the affected PC didn't have PS/2 - while I only had PS/2 spare keyboards. Usually I prefer software that is actively maintained and doesn't contain any advertising. Furthermore with DBAN the startup took some minutes, as it initially scanned for USB devices for one or two minutes....which was really annoying. So I would have used shredos in the first place, but wasn't able to, due to that driver issue.

> @basicmaster New version now available for testing https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2020.05.008_x86-64_0.30.001

I just tested that version and can confirm that now both PS/2 adapters work. Great, thank you!

closed 2021-01-28 by basicmaster