PartialVolume/shredos.x86_64 #10: just tried this on my Intel sandy bridge server board and the OS went to sleep...

mrjayviper, 2020-10-02:
as above. I do not know the progress of the wipe and I have a feeling it's not doing anything as l cannot hear any activity on the HDD.

some info:
1. I'm on a mac when I created the bootable USB
2. Booted using BIOS instead of UEFI

Thank you.

PartialVolume, 2020-10-02:
Can you post a photo of the screen ? i.e. did the drive/s appear, did you select it for wiping and the startvthe wipe. From the photo I can then see what part of the program you are at.

mrjayviper, 2020-10-03:
Nothing to see as the screen is blank. But I was able to select the drives that needs to be wiped and then I started it. An hour later when I checked on the machine, all I got is a blank screen. The PC wouldn’t respond to keyboard input.

PartialVolume, 2020-10-03:
Is it reproducible? Very strange that the whole screen is blank. No chance the blank screen key got inadvertently pressed? I don't remember whether you have to press the B key or any key to bring the screen back from blank mode. Just wondering if that was the case. Of course if you could hear no disc activity it would also mean the wipe would have finished by the time you got back to check it out. What size drive were you wiping?

PartialVolume, 2020-10-03:
`No chance the blank screen key got inadvertently pressed? I don't remember whether you have to press the B key or any key to bring the screen back from blank mode.`

Forget what I said above, I just checked and if you have blanked the screen with the 'B' key, then pressing any key should bring the screen back. So unlilkely to be that.

PartialVolume, 2020-10-03:
Not much help to you, but I have discovered a bug with the status bar. After starting the wipe the existing status bar content should be replaced with `B=Blank screen Ctrl-C=Quit`. I'll open an issue for that with nwipe and fix it.

mrjayviper, 2020-10-04:
I was wiping 3x 2TB HDDs so it’ll take more than 2 hours to finish it. I
doubt it was complete

PartialVolume, 2022-02-05:
Any chance it was related to this #64 as it sounds like similar symptoms.

PartialVolume, 2026-02-14:
Closing, if this is still an issue with https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2025.11_28_x86-64_0.40 then reopen issue. Thanks.

closed 2026-02-14 by PartialVolume