PartialVolume/shredos.x86_64 #11: Little mistake at the ShredOS img flash guide for MAC users

MichalPlays, 2020-12-08:
At
" Obtaining and writing shredos to a USB flash drive, the easy way ! "
under the section "Linux (and MAC) users" 

there is written something like 
_(where sdx is the device name of your USB drive, this can be obtained from the results of sudo fdisk -l)_

I don't know if that's is happening after the MacOS Big Sur update or something but the command sudo fdisk -l doesn't work on MacOS. It just gives me this error "fdisk: illegal option - l"

Please add to the installation guide that the fdisk command should be used on Linux and for MAC users should use the **diskutil list** command in order to display all storage devices connected to your machine.

Here is the screenshot, what happens is if I type the fdisk command from the readme file and what happens if you type the diskutil command I mentioned before, in case you don't own any device that runs MacOS and you might think that this command formats the hard drive

[image]

Sorry for my bad English and I hope to see some correction made in the readme file : )

PartialVolume, 2020-12-08:
Thanks for the update, I'll make the changes ASAP.

MichalPlays, 2020-12-08:
Alright, thanks for the Info

closed 2020-12-08 by MichalPlays

PartialVolume, 2020-12-08:
README.md updated. Thanks for your feedback, much appreciated.