PartialVolume/shredos.x86_64 #12: shred OS does not boot on old Notebook

peepls, 2020-12-23:
Hi,

I made a bootable USB-Stick with ShredOS. It worked fine on newer Laptops, but on an older one I do not get past a black screen with just one line: "ShredOS - booting". 

The computer in question is a Fujitsu Siemens AMILO Pro.

I can rule out the stick as it worked on other hardware. I can give more information if needed.

PartialVolume, 2020-12-23:
Is it booting the OS, windows/Linux from its internal drive ?

peepls, 2020-12-24:
Yes it does. Windows 7 boots.

PartialVolume, 2020-12-24:
That's very likely, I'm planing on updating shredos to use the latest 0.30.x release of nwipe shortly.  (Jan-Mar). When I do that, I'll look at the issue of 64 & 32 bit support and release a version for 32 bit processors once I've tested against older celeron M and Pentium 4 hardware.

peepls, 2020-12-24:
Sound promising. Thank you for your work!

PartialVolume, 2021-01-06:
I've produced a 32 version of shredos with nwipe 0.30.001 that should work on all versions of Pentium processor, i.e Pentium Pro, II, III, IV. etc Currently it's being tested on a Pentium 4 with 256 MB memory. If all good I'll release it later today. What's the processor on your old laptop?

peepls, 2021-01-06:
It's an Intel Pentium M 735. I am going to test the version once its released.

PartialVolume, 2021-01-06:
I've created a new repository for the 32 bit version [here](https://github.com/PartialVolume/shredos.i686). If you could download the latest and only release of the .img file and burn it to a USB stick and let me know if it works ok.

Expect it to take considerably longer to boot, i.e. it will remain at the "ShredOS - Booting" screen for probably 40 seconds+, at least that's how long it takes on my P4. I guess I've got USB V1 slots. In comparison, my 5 year old I7 laptop boots ShredOS in 5 seconds

peepls, 2021-01-07:
It is running right now on my old notebook. Everything seems to work. I am going to have a look at the hard disk afterwards, but until now it looks great. Thank you again!

I noticed one minor issue, nevertheless: on a 64-bit computer (intel core i5) it does not boot. I wanted to test it there first, because I was not sure if the old laptop would boot from USB at all. I do not see this as a bug, but wanted to mention it, because you said

> Note. ShredOS i686 (32 bit) will also run on x86_64 (64 bit) processors.

closed 2021-01-07 by peepls

PartialVolume, 2021-01-07:
Does that i5 laptop run the 64 bit version. I'm just wondering whether it didn't boot on the i5 due to it having UEFI secure boot enabled, so it will only boot Windows unless you disable it. I could be completely wrong about that, but worth checking.

peepls, 2021-01-11:
Hi, sorry for the delay. I hadn't have access to my computer until now. I just tried both versions to be sure:

The 64 bit version boots, the 32 bit version does not get past a black screen with the following two lines:

shred OS - booting ...
_

PartialVolume, 2021-01-11:
@peepls No problem,

So I thought I would recheck the 32bit edition (v2020.02.005.i386-0.30.001) on some 64 bit hardware.
| Manufacturer | Model | Processor | Status |
| ------------------ | -------- | -------------- | -------- |
| MacBook Pro 11,2 | A1398 | Intel i7 4770HQ CPU 2.2 GHz | 32 bit edition booted on 64 bit hardware |
| Medion Laptop | X7815 | Intel i7 2670QM CPU 2.2 GHz |  32 bit edition booted on 64 bit hardware |
| Novatech | N/A | Intel i5 4460 CPU 3.2GHz | 32 bit edition booted on 64 bit hardware (UEFI) |
| Unbranded | N/A | Xeon E5345 dual processors, 2.33GHz | Would only boot 64 bit edition and on only one USB port. |

So, the 32 bit edition does boot on 64 bit hardware but I'm guessing there are a few cases where it does not. Whether the 32 edition not booting on your i5 laptop is a bios/UEFI issue or maybe some setting in the bios to do with USB or maybe a obscure fault on the hardware, or a bug somewhere in the ShredOS software I couldn't say.

As the 64 bit edition works fine on the i5 I'm not too concerned about it. Even though I'd like to know the reason, it's probably not worth spending too much time on it as there are other things in nwipe I need to work on. However I do appreciate you reporting the problem so we can reopen it at a later date if required.

peepls, 2021-01-11:
> As the 64 bit edition works fine on the i5 I'm not too concerned about it. Even though I'd like to know the reason, it's probably not worth spending too much time on

I see it the same way. 

> However I do appreciate you reporting the problem so we can reopen it at a later date if required.

I will do that, adding some more information.

Thank you for investigating.