PartialVolume/shredos.x86_64 #16: 4 minute boot on Sony VAIO  Model PCG-3B1M while other laptops boot quickly.

PartialVolume, 2021-01-14:
Initially I thought ShredOS 64bit wouldn't boot on this device, while the exact same ShredOS on the exact same flash drive boots within six seconds on my I7 laptop. I can see from the activity light on the flash drive its reading the ShredOS image file for about 8 seconds then all activity stops, nothing. It just says ShredOS Booting. However wait for just under 4 minutes and nwipe appears. dmesg shows nothing that jumps out at me as being the cause. Win7 boots fine.

I'm not sure if it's hardware or software related at this stage. Thought I'd mention it here just in case anybody else has a ShredOS that takes several minutes to boot when it should be seconds.

Firminator, 2021-02-02:
Assuming ShredOS is Systemd based you could run
`systemd-analyze`
for a summary of your boot process. Then dig deeper to find out which service or device or daemon is taking 3 to 4 minutes to initialize with
`systemd-analyze blame`

Firminator, 2021-11-26:
According to the buildroot config ShredOS is using sysvinit and not systemd, so nevermind my comment.

PartialVolume, 2021-11-26:
>According to the buildroot config ShredOS is using sysvinit and not systemd, so nevermind my comment.

Yes, systemV, I did try systems but had issues with it so stayed with System V.

I haven't tried the Samsung laptop that takes over 2 minutes to boot with the latest 2021.08.2 version. I'll go grab it now and give it a try.

PartialVolume, 2021-11-26:
Ha ! It only works now ! 20 secs to boot and about 16 of those seconds are spend reading USB stick via the ancient old USB socket.

Fixed in 2021.08.2 build ! Release imminent.

closed 2021-11-26 by PartialVolume

Firminator, 2021-11-26:
Awesome! This is great, all these issues are gone with the new version!
But hold on... I wanted to report a bug with the serial number on a SAS drive that I just found. It could go into the release.  I'll open a new issues. This is probably an easy fix.

PartialVolume, 2021-11-26:
Ok, no hurry. I still need to stop telnetd starting at boot unless the user enables it via the kernel command line.