PartialVolume/shredos.x86_64 #17: Include arch in release filename

hameau, 2021-01-14:
Hello,

At present, both the x86_64 and the i686 versions have the same release download filename – shredos-20210106.img, at the time of reporting. This is problematic for easily identifying files after download and easy downloading by utilities such as wget.

For users that want/need both versions, running, for example, wget with the URL for one version would overwrite an existing copy of the other version. The only way of subsequently checking what's already downloaded would be by comparing checksums of local files with the release notes.

I suggest something like `shredos-<release-date>-i686.img` and `shredos-<release-date>-x86_64.img`.

Thanks for considering.

PartialVolume, 2021-01-14:
I'll adopt that naming convention for all future releases. Thanks for letting me know.

PartialVolume, 2021-01-15:
Closed by [cc03b29](https://github.com/PartialVolume/shredos.2020.02/commit/cc03b29b99f0d205e52f2e03e05c934c1bf70a39)

closed 2021-01-15 by PartialVolume