PartialVolume/shredos.x86_64 #18: Am I able to pick which drives to wipe?

zdat, 2021-01-16:
Am I able to select which disks to erase or is my only option to nuke everything?

There are no videos of this anywhere online (like DBAN) so I don't know what I'm going to see. Would appreciate any reassurance; screenshots? I do not want to wipe everything, just certain drives.

Thanks.

PartialVolume, 2021-01-16:
No, your drives won't be autonuked, you can select which drive to wipe. The only way you would nuke everything is to specify ./nwipe --autonuke on the command line. ShredOS's default mode is manual selection as shown in the video below. More information about nwipe can be found [here](https://github.com/martijnvanbrummelen/nwipe).

Once Shredos starts it will show all the drives on your system. You may have to do a Control-C to restart nwipe if you have USB drives as sometimes they are not displayed immediately after Shredos boots as the USB devices sometimes take longer to initialise than Shredos takes to boot. Shredos boots fast.

Use the up down arrow keys to highlight a drive then press spacebar to select it. Capital 'S' starts the wipe. Multiply drives can be selected and wiped simultaneously if required. 

[image]

PartialVolume, 2021-01-16:
@zdat Here's a you tube video of nwipe, which is what ShredOS boot straight into. You don't need to type anything on the command line to start ShredOS start nwipe automatically with no drives selected.

The version shown in this video is 0.21 and we are currently at 0.30 however you will get the idea.
[Nwipe Demo](https://youtu.be/B7skp80Ge3s)

zdat, 2021-01-16:
Thanks for the help. Hoping this works because I couldn't get DBAN to work at all!

PartialVolume, 2021-01-16:
If for any reason it doesn't work, let me know and I'll see if I can help out.

PartialVolume, 2021-01-22:
@zdat Closing issue, feel free to reopen if you have any issues.

closed 2021-01-22 by PartialVolume

zdat, 2021-01-24:
It worked, but I'm not 100% sure how to use the drives to store data again.

I know it's not to do with ShredOS but any help with that? Probably should've made sure it was a simple process beforehand.

PartialVolume, 2021-01-24:
Depends whether you plan on using the discs with windows or Linux. Because I use Linux I tend to use a program called gparted. Basically you need to create a partition table and the create a filesystem which could be NTFS for windows, ext4 or umpteen others for Linux. Perhaps I should put some  basic instructions in the README on how you would partition and create a filesystem on a disc. There will be loads of info on the internet or you tube on how you do that.

PartialVolume, 2021-01-24:
@zdat [Formatting a disc with Windows 10](https://youtu.be/ZxV6IcNvbek)

zdat, 2021-01-25:
Windows 7. I don't get that pop up when I open disk management on Windows 7. Tried using commands in diskpart but they didn't seem to work.

zdat, 2021-01-26:
Windows decided it likes the drives now... not sure why it didn't before. I thought there was some extra process to format a 0'd out drive but I guess it was just Windows being weird.