PartialVolume/shredos.x86_64 #22: ShredOS doesn't suspend to RAM

PartialVolume, 2021-01-25:
rtcwake -m mem -s 5 does indeed put the device to sleep but upon waking reboots ShredOS rather than the expected behaviour of restoring the running ShredOS from RAM.

PartialVolume, 2021-01-25:
https://github.com/PartialVolume/shredos.x86_64/issues/20#issuecomment-766871022

hameau, 2021-01-25:
'ShredOS fails to wake from suspend' might be a better title – the suspend part appears to be okay.

Referring to https://github.com/PartialVolume/shredos.x86_64/issues/20#issuecomment-766953990, ACPI support, or lack of, is the most likely culprit. That would also account for the waking without restarting the video hardware (https://github.com/PartialVolume/shredos.x86_64/issues/20#issuecomment-766871022).

PartialVolume, 2021-01-25:
>Apologies if you already know this stuff, but this Ubuntu wiki entry gives a useful overview.

Feedback and links to articles are always useful, thanks.

PartialVolume, 2021-01-27:
@hameau I just tried `rtcwake -m mem -s 5` multiple times using the latest i686 version of ShredOS (identical to the x86_64 version except for the change in architecture only) on a Pentium 4 and it woke up correctly after 5 seconds. All virtual terminals are fully functional.

So the problem would seem to be related to specific hardware.

hameau, 2021-01-28:
Wake from suspend has been a thorn in the side of Linux for ever!

I have tried with an MSI Wind (Atom) and it fails to resume – black screen, but the computer is running. The same result using `rtcwake` and manipulating `/sys/power/state`, with both the i686 and x86-64 images. Neither of my test cases have provoked a reboot, just no video.

That hardware works correctly under SysRescCD, but not Clonezilla (Debian version).

For a standard installed Linux distribution, it's usually a matter of pushing options to the video driver. Interesting that both of my test cases have integrated Intel graphics, which your P4 doesn't. (Of course, integration itself may not be significant – other things will also have changed over time.)

There is a kernel-level TRACE_RESUME feature: https://www.kernel.org/doc/Documentation/power/s2ram.txt

I'll set up the MSI Wind to test with that in the next few days, if I can. It's something that needs a bit of preparation and some clear bench space.

Also, this Ubuntu wiki page: https://wiki.ubuntu.com/DebuggingKernelSuspend (but needs a workaround for `pm-suspend`, which isn't in ShredOS – not a disaster).

hameau, 2021-01-28:
> ... but upon waking reboots ShredOS ...

What's the hardware for this one? CPU & graphics.

PartialVolume, 2021-01-28:
> > ... but upon waking reboots ShredOS ...
> 
> What's the hardware for this one? CPU & graphics.

HP Pavilion dm1 notebook. AMD E-450 APU (AMD Radeon HD 6320)

PartialVolume, 2021-01-28:
@hameau I've now got rtcwake working properly on my main laptop, which is an I7 with Nvidia graphics. The laptop now wakes correctly. Not only that but the LCD brightness controls on the keyboard now work too. I've had to expand the disc size from 38Mb to 41Mb to accommodate all the video, backlight, control drivers for Intel, AMD, ATI plus others. There is a couple of issues I need to work on, on the HP notebook the screen is alive in that it says 'ShedOS booting' but nwipe never displays. If you ALT F2 you can power off but the screen never changes from saying ShredOS booting.

The other problem is the text resolution on my I7, because it's now using the Nvidia video driver it defaults to the screens highest resolution which is too high for nwipe..
[image]

On the plus side the laptop now wakes properly from sleep.

PartialVolume, 2021-01-28:
nomodeset on the kernel command line gets me my original 80 x 25 character resolution back but reintroduces the rtcwake issue. So I definitely need to set a graphics mode, VGA=0x etc  is deprecated and set gfxpayload=1024x1024x24 still leaves me with the highest resolution. Onwards and upwards ..

hameau, 2021-01-29:
> I've had to expand the disc size from 38Mb to 41Mb ...

Yes, I was afraid of some growth. You're still pretty compact at around 7% of SysRescCD, so the wider usefulness is very much worthwhile, imo.

> set gfxpayload=1024x1024x24 still leaves me with the highest resolution.

I doubt if `1024x1024x24` is a valid mode. Typo? You probably need to find a reasonable 'lowest common denominator' to suit a wide range of hardware. Personally, I find an 80x25 terminal too low-res., even in nwipe, as the wipe method text description overflows the available space (e.g., for zero method). Even if not ideal, I would be okay with the resolution in your screenshot – you're unlikely to find a single setting that will fit _every_ case.

- Systutorials [kernel video mode](https://www.systutorials.com/configuration-of-linux-kernel-video-mode/) (may only be useful early in the boot process – I don't know if it continues to the end).
- Stackexchange [console video mode](https://unix.stackexchange.com/questions/583585/setting-the-most-basic-text-console-video-mode-on-linux).

Impressive progress, thanks.

PartialVolume, 2021-01-29:
@hameau thanks for those links, they were very useful. I think I'm going to have to spend a lot more time on making this reliable using video drivers rather than the fb driver, I have one generic I7/Nvidia laptop that works well but I need to work out a method by which the resolution can be easily changed by a user. While on a Sony VAIO the laptop freezes the frame buffer when switching drivers and the same with an HP.

This will be something I intend to do (or somebody else if they can can up with a good PR) but for the time being I need to fix a couple of bugs in nwipe and add HDA detection and correction and then I'll come back to this particular issue, therefore I'll leave this issue open. As always any help you can provide on this is appreciated.