PartialVolume/shredos.x86_64 #23: Is shredos USB stick able to be removed once a wipe is started?

DConfer99, 2021-01-28:
First of all, I have to say that I love this software; it is a very welcome upgrade from the now outdated DBAN. My organization is looking to use this software to sanitize a large number of HDDs, but I would like to know if it is safe to remove a shredos USB stick from a device after a wipe is started so that it can be used on another machine. I tested this out on a couple of desktops, and it seemed like everything was loaded to RAM and that removing the USB stick had no adverse effects. However, I thought that I would come here just to confirm that this was the case.

PartialVolume, 2021-01-28:
Thanks, for your kind words, it does take a considerable amount of time to maintain ShredOS and Nwipe, so it's appreciated when somebody finds its useful to them.

As regards pulling the USB stick out, that's not a problem, like you say it's all loaded to RAM. During testing I'm often removing the USB stick once it booted.

DConfer99, 2021-01-28:
Awesome, thanks for the quick response. Hopefully this thread will answer anyone else that has the same question. Keep up the good work! I'm sure this is going to become one of the more popular drive wiping solutions in the coming months/years, especially since some newer machines only support UEFI boot.

closed 2021-01-28 by DConfer99