PartialVolume/shredos.x86_64 #25: hit and miss success with USB keyboard use once booted in shred gui/cli

bforcier299, 2021-02-14:
I can force boot a computer to boot into ShredOS stick.

BUT...

Some computers will loose the connection to keyboard (also, power seems to go out from keyboard as in light turns of on it).

I have an Asus server with 10 USB ports (4 front, 6 back) and none of the USB ports will work.

It happens at blank screen before landing in ShredOS cli/gui which it does.

Please let me know what kinds of details are required so you can help me make sure it works 100% of the time.

PartialVolume, 2021-02-14:
What version of ShredOS are you running? I'm assuming you are using the latest v2020.05.008_x86-64_0.30.001 as printed on the nwipe title bar but if not then please try the latest version.

On the server where none of the USB ports work can you paste the output of `lsusb -v`. and also the nwipe log that will show the server basic details.

I'll need the same information for any other computer that has USB keyboard issues.

bforcier299, 2021-02-14:
Hello, I'm using the latest version.

Stupid question, how can I get the info you're asking if keyboard won't work?

Can I ssh in?

Thanks

PartialVolume, 2021-02-14:
No ssh isn't available. I would live boot a linux distro to obtain the lsusb output. Does this server have PS2 ports by any chance?

bforcier299, 2021-02-14:
Ok, will boot from a LiveCD USB stick and run the command then get back to you with the output.

P/S 2 answer is yes but I don't have a PS/2 keyboard though, it's a fairly recent server.

I'll get that to you tomorrow as I reverted to booting up with Gparted LiveCD and running 

sudo shred -v -n 8 -z /dev/sdX

from terminal on 6 drives so it'll take a while (2*120GB, 4*6TB)...

on a side note, and as a potential feature request, have you bought of a preflight menu on boot asking about keyboard locale etc.? (sort of like the questions asked by both Clonezilla Live and Gparted Live? That'd be nice as long as we can keep the image size under 128MB...

PartialVolume, 2021-02-14:
>on a side note, and as a potential feature request, have you thought of a preflight menu on boot asking about keyboard locale etc.? (sort of like the questions asked by both Clonezilla Live and Gparted Live?

Yes, I've thought about that and agree that it's something that should be included. I guess i could modify the launcher script to check the kernel command line for keyboard locale and if nothing found then it will run the keyboard locale selection and then write the locale back to the kernel command line so next time you run it you won't need to run keyboard selection.

PartialVolume, 2021-02-15:
I'll also checkout gparted live which should give me a clue as to what's missing for Asus. Does gparted live not include nwipe? If not it might be worth asking them to include it.

bforcier299, 2021-02-15:
Let me check on this (nwipe on gparted live) and get back to you.

bforcier299, 2021-02-15:
Here are an lsusb AND a lshw report from 1 of the computers in question (the 'Asus' server).

Let me know if doing the same report on the other one (an MSI clone H170I AC motherboard) would help as well.

[lshwreport.txt](https://github.com/PartialVolume/shredos.x86_64/files/5982231/lshwreport.txt)
[lsusbreport.txt](https://github.com/PartialVolume/shredos.x86_64/files/5982232/lsusbreport.txt)

Regards

bforcier299, 2021-02-15:
> Let me check on this (nwipe on gparted live) and get back to you.

It is not part of Gparted Live.

bforcier299, 2021-02-15:
> I'll also checkout gparted live which should give me a clue as to what's missing for Asus. Does gparted live not include nwipe? If not it might be worth asking them to include it.

I've made a feature request on their forum page.

PartialVolume, 2021-02-15:
@bforcier299 I just took a look at the ShredOS drivers and I'm missing the Asus HID driver. Just about every other HID driver is installed except that one!. Hopefully that will fix the problem. I'll recompile and send you a link to the image to check it out. What other systems were you having a problem with?

bforcier299, 2021-02-15:
Looking forward to testing it!

The other one is an MSI motherboard, let me connect the clone and grab the reports.

Thanks again!

bforcier299, 2021-02-15:
Just did the test with the MSI clone.

It seemed to detect the keyboard, must be making a mistake. I've got another one to test that maybe is the one giving me issues. I'll test that one as well and provide the reports if necessary.

Here are the reports for the MSI clone:

[msi_lshwreport.txt](https://github.com/PartialVolume/shredos.x86_64/files/5984342/msi_lshwreport.txt)
[msi_lsusbreport.txt](https://github.com/PartialVolume/shredos.x86_64/files/5984343/msi_lsusbreport.txt)

bforcier299, 2021-02-15:
So...the lenovo oldie (around 10 years alone) is the one not liking ShredOS once it lands in Nwipe in regards to keyboard usability,

Here are the two reports I've generated

[lenovolshw.txt](https://github.com/PartialVolume/shredos.x86_64/files/5984437/lenovolshw.txt)
[lenovolsusb.txt](https://github.com/PartialVolume/shredos.x86_64/files/5984438/lenovolsusb.txt)

Regards

bforcier299, 2021-02-17:
> I've made a feature request on their forum page.

The Gparted developers have published a revised version that includes nwipe for review.

So, at least there this option as well for wiping drives available on the same LiveCD that ships with Gparted so everything could be done with one LiveCD. 

Looking forward to a revised ShredOS LiveCD that will correct the usb keyboard driver issue.

Thanks again!

PartialVolume, 2021-02-17:
@bforcier299 That's great, thanks for taking the time to make the gparted guys aware of nwipe.

I've added the ASUS drivers and some drivers I'm hoping will fix the Lenovo problem. I hope to publish that some time today.

PartialVolume, 2021-02-17:
@bforcier299 Can you test the latest version  [shredos-2020.05.009_x86-64_0.30.001_20210217](https://github.com/PartialVolume/shredos.x86_64/releases/download/v2020.05.009_x86-64_0.30.001/shredos-2020.05.009_x86-64_0.30.001_20210217.img) on your ASUS server and Lenovo laptop. Fingers crossed, the keyboards now work correctly.

Thanks.

bforcier299, 2021-02-17:
I confirm the Lenovo oldie is working properly. Having some no related issues with the Asus ws/server, will make sure to test it and get back to you.

If both are confirmed to work, can we consider this revision Golden Master or 'ready for release'?

Thanks!

PartialVolume, 2021-02-24:
@bforcier299 No rush or anything, just wondering if you had a chance to check that ASUS server yet?

bforcier299, 2021-02-25:
Yeah, sorry. Haven't had a chance to test it on the Asus server but as I said it works on the old Lenovo so I'm confident it will work.

Thanks

PartialVolume, 2021-02-25:
No problem, In that case I'll close this issue, please reopen if you find an issue with the Asus at some future date.

bforcier299, 2021-02-25:
Finally had a chance to resolve the Asus Server booting issue (not related to shreds).

Keyboard disconnects while showing GRUB loading and once landed in Nwipe, keyboard does not work. 

Sorry for the delay.

PartialVolume, 2021-02-26:
Having had another look at the Asus files you sent, I'm missing the Intel xHCI USB role switch support driver from ShredOS. I'll fix that and send you another image to try and see if that resolves the issue.

PartialVolume, 2021-02-26:
@bforcier299 If you could download the latest release [v2020.05.010_x86-64_0.30.001](https://github.com/PartialVolume/shredos.x86_64/releases) and give that a try on your Asus server. Thanks.

bforcier299, 2021-02-26:
Just tested latest....

Keyboard does not work once booted into Nwipe on the Asus server.

Sorry

PartialVolume, 2021-02-26:
Thanks. I'll get back to you on this

PartialVolume, 2021-02-26:
This is a puzzle, it looks like the drivers that are needed are installed. So I'm missing something here. Just to double check, does ALT-F2 or ALT-F3 get you to a virtual terminal?, or CTRL+Z, CTRL+C, I'm just wondering if nwipe is hanging for some other reason. Do all the expected discs show up? Just wondering if libparted is hanging when trying to access the devices.

bforcier299, 2021-02-26:
Alt F2 or Alt F3 don't work

Ctrl Z or Ctrl C neither.

All 6 drives show up.

Is it missing a very basic non free USB driver?

Same behaviour, once it shows

Shred OS - booting ...

The keyboard light shuts off 1 or 2 seconds after and after another 2-3 seconds max it lands into Nwipe with keyboard not working.

PartialVolume, 2021-02-26:
From the files you provided, it has a P11C-X motherboard, with Intel's Cannon Lake PCH USB 3.1 xHCI Host Controller, using the xhci_hcd driver. The USB_XHCI_HCD driver is loaded in Shredos are per the info below. 
```
Symbol: USB_XHCI_HCD [=y]
Type  : tristate
Defined at drivers/usb/host/Kconfig:20
Prompt: xHCI HCD (USB 3.0) support
Depends on: USB_SUPPORT [=y] && USB [=y] && HAS_DMA [=y] && HAS_IOMEM [=y]
Location:
 -> Device Drivers
 (1)   -> USB support (USB_SUPPORT [=y])
```
The chipset inside the keyboard is made by SIGMACHIP and uses the usb_hid driver.
```
              *-usb:2
                   description: Keyboard
                   product: USB Keyboard
                   vendor: SIGMACHIP
                   physical id: 6
                   bus info: usb@1:6
                   version: 1.10
                   capabilities: usb-1.10
                   configuration: driver=usbhid maxpower=98mA speed=2Mbit/s
```
And in ShredOS the usb_hid driver is also loaded.
```
Symbol: USB_HID [=y]
Type  : tristate
Defined at drivers/hid/usbhid/Kconfig:5
Prompt: USB HID transport layer
Depends on: USB [=y] && INPUT [=y]
Location:
     -> Device Drivers
      -> HID support
 (1)     -> USB HID support
 Selects: HID [=y]
```

bforcier299, 2021-02-26:
This is weird.
I don't know what to say or to try next.

It's not the end of the world, it'd just be nice if we could resolve it.

PartialVolume, 2021-02-26:
The SATA section uses the Cannon Lake PCH SATA AHCI Controller and that's using the ahci driver.
```
        *-sata
             description: SATA controller
             product: Cannon Lake PCH SATA AHCI Controller
             vendor: Intel Corporation
             physical id: 17
             bus info: pci@0000:00:17.0
             logical name: scsi0
             logical name: scsi1
             logical name: scsi2
             logical name: scsi3
             logical name: scsi4
             logical name: scsi5
             version: 10
             width: 32 bits
             clock: 66MHz
             capabilities: sata msi pm ahci_1.0 bus_master cap_list emulated
             configuration: driver=ahci latency=0
```
The ahci driver is provided in ShredOS by the sata_ahci driver.
```
 Symbol: SATA_AHCI [=y]
 Type  : tristate
 Defined at drivers/ata/Kconfig:85
   Prompt: AHCI SATA support
   Depends on: ATA [=y] && HAS_DMA [=y] && PCI [=y]
   Location:
     -> Device Drivers
 (6)   -> Serial ATA and Parallel ATA drivers (libata) (ATA [=y])
```
The PCI bridge is a Cannon Lake PCH PCI Express Root Port #21 which uses the pcieport driver.
```
        *-pci:0
             description: PCI bridge
             product: Cannon Lake PCH PCI Express Root Port #21
             vendor: Intel Corporation
             physical id: 1b
             bus info: pci@0000:00:1b.0
             version: f0
             width: 32 bits
             clock: 33MHz
             capabilities: pci pciexpress msi pm normal_decode bus_master cap_list
             configuration: driver=pcieport
             resources: irq:121 memory:94d00000-94ffffff ioport:94600000(size=3145728)
```
### **Now this driver is not currently installed in ShredOS.**
```
 Symbol: PCIEPORTBUS [=n]
 Type  : bool
 Defined at drivers/pci/pcie/Kconfig:5
   Prompt: PCI Express Port Bus support
   Depends on: PCI [=y]
   Location:
     -> Device Drivers
 (1)   -> PCI support (PCI [=y])
```
Also the AST VGA controller is not loaded which is odd as we are getting a display. I'll add both those drivers and see where we go from there.

bforcier299, 2021-02-26:
> Also the AST VGA controller is not loaded which is odd as we are getting a display. I'll add both those drivers and see where we go from there.

The CPU is an Intel E-2176G with integrated Intel UHD Graphics P630 so maybe that's why and the motherboard detects that and deactivates the AST VGA controller?

PartialVolume, 2021-02-26:
When you get a chance can you try the latest release again.  [v2020.05.011_x86-64_0.30.001](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2020.05.011_x86-64_0.30.001)

 I've included the two extra drivers discussed above plus more USB drivers. If this doesn't work then if the server has a PS2 keyboard port we could use instead that should at least tell us whether it's a USB problem or something else.

bforcier299, 2021-02-27:
The AST VGA driver seems to have helped in regards to better displaying nwipe (it now shows it in 1920*1080 which is native resolution of the monitor instead of a blown 640*480). 

No luck on the usb keyboard though.

I don't have a PS/2 keyboard or a USB to PS/2 adapter handy so I can't test the PS/2 port.

As you said, this should not bar the build from being used in the future.

Thanks for giving it a try!

ps: at least we now have old Lenovo commercial desktops working ;-)

PartialVolume, 2021-12-28:
Closing, feel free to re-open issue if the problem still exists with the latest version of ShredOS.

closed 2021-12-28 by PartialVolume

nik6nik, 2022-03-24:
HI, I landed here because I had a machine with an ASUS motherboard and tried running ShredOS (latest ver v2021.08.2_21_x86-64_0.32.023) and was having the same thing. it would boot the keylock light would go off (set to be on at boot in bios) and no keys on keyboard would, no commands nothing. After several attempts at changing usb options in BIOS i checked ASUS for a firmware update ..... after the update keyboard worked! 

just an FYI

PartialVolume, 2022-03-24:
> HI, I landed here because I had a machine with an ASUS motherboard and tried running ShredOS (latest ver v2021.08.2_21_x86-64_0.32.023) and was having the same thing. it would boot the keylock light [...]

That's great, thanks for taking the time to let us know what worked on your hardware. Much appreciated.

vkumarsnn, 2023-11-03:
Hiya, I am trying to wipe the HP AIO 24 df0073na its an intel pentium gold. The ShredOS boots perfectly, however when I am trying to use the keyboard, it doesnt recognizes it. And because its an AIO, there only USB based I/O for Keyboard and Mice.
I have already tried the latest version https://github.com/PartialVolume/shredos.x86_64/releases/download/v2021.08.2_23_x86-64_0.34/shredos-2021.08.2_23_x86-64_0.34_20221231.img,

Also I have tried the earlier test releases in this post https://github.com/PartialVolume/shredos.x86_64/issues/25 but no use.

Can you please let me know if there is any fix for this?