PartialVolume/shredos.x86_64 #26: nwipe fails to build on Arch Linux

Repentinus, 2021-02-17:
I am running into a strange issue where `make nwipe` (likely `autogen.sh` in nwipe) manages to symlink `output/build/nwipe-v0.30.001/{compile,config.guess,config.sub,install-sh,missing,depcomp}` to the respective files in `/usr/share/automake-1.16/', which belong to a system package, are owned by root, and not writable by my normal user. Any clues what could be causing this? No other package is exhibiting the same issue.

```
[repentinus@Mandalore shredos.x86_64]$ make nwipe
nwipe-v0.30.001.tar.gz: OK (sha1: c784e1413543dd290296071db1a36e86b0d3d0f0)
>>> nwipe v0.30.001 Extracting
gzip -d -c /home/repentinus/Software/shredos.x86_64/dl/nwipe/nwipe-v0.30.001.tar.gz | /home/repentinus/Software/shredos.x86_64/output/host/bin/tar --strip-components=1 -C /home/repentinus/Software/shredos.x86_64/output/build/nwipe-v0.30.001   -xf -
>>> nwipe v0.30.001 Patching
(cd /home/repentinus/Software/shredos.x86_64/output/build/nwipe-v0.30.001 && cp ../../../package/nwipe/002-nwipe-banner-patch.sh /home/repentinus/Software/shredos.x86_64/output/build/nwipe-v0.30.001 && ./002-nwipe-banner-patch.sh  && ./autogen.sh);
configure.ac:12: installing './compile'
configure.ac:65: installing './config.guess'
configure.ac:65: installing './config.sub'
configure.ac:6: installing './install-sh'
configure.ac:6: installing './missing'
src/Makefile.am: installing './depcomp'
configure.ac:7: warning: AC_OUTPUT should be used without arguments.
configure.ac:7: You should run autoupdate.
>>> nwipe v0.30.001 Updating config.sub and config.guess
for file in config.guess config.sub; do for i in $(find /home/repentinus/Software/shredos.x86_64/output/build/nwipe-v0.30.001 -name $file); do cp support/gnuconfig/$file $i; done; done
cp: cannot create regular file '/home/repentinus/Software/shredos.x86_64/output/build/nwipe-v0.30.001/config.guess': Permission denied
cp: cannot create regular file '/home/repentinus/Software/shredos.x86_64/output/build/nwipe-v0.30.001/config.sub': Permission denied
make: *** [package/pkg-generic.mk:233: /home/repentinus/Software/shredos.x86_64/output/build/nwipe-v0.30.001/.stamp_patched] Error 1
```

I am seeing this on commit 87d0a4ef3580ef53caa6076dc503f79cf3fa4ec9.

PartialVolume, 2021-02-17:
Are you manually doing a `make nwipe` or was that output after having done a `make`. 

Just double checking but `make nwipe-dirclean` should proceed `make nwipe`. I've also found that it's often necessary to do a clean rebuild despite it taking considerably longer than just a make nwipe. 

So if you are getting any unexpected errors I would always do a clean rebuild and then diagnose the problem from there.
```
make clean
make shredos_defconfig
make
```
If you've done that, then I'd look at whether you've accidentally been running as sudo or su, sorry I don't use arch so I'm not sure whether you use the traditional su or like ubuntu sudo. That would have caused some directories to have the wrong permissions.

Doing a `make clean` should delete the output/build directories, if it doesn't  and it may not, if your directories were created as root you will need to manually delete them. However this can lead you down a rabbit hole chasing incorrectly owned directories. So if it was me, I'd delete the entire shredos folder, makesure you are **not** running as su or sudo. Download a fresh copy of shredos from github then do a 
```
make clean
make shredos_defconfig
make
```

Repentinus, 2021-02-17:
Thanks for the help.

> Are you manually doing a `make nwipe` or was that output after having done a `make`.

Both give the same results in a fresh clone after `make shredos_defconfig`.

> So if you are getting any unexpected errors I would always do a clean rebuild and then diagnose the problem from there.

That's what I've been doing.

> If you've done that, then I'd look at whether you've accidentally been running as sudo or su, sorry I don't use arch so I'm not sure whether you use the traditional su or like ubuntu sudo. That would have caused some directories to have the wrong permissions.

I have not used root for anything. All the files created in the build directory are owned by my user:
```
[repentinus@Mandalore shredos.x86_64]$ ls -la output/build/nwipe-v0.30.001/
total 416
drwxr-xr-x  7 repentinus repentinus   4096 Feb 17 08:44 .
drwxr-xr-x 84 repentinus repentinus   4096 Feb 17 08:44 ..
-rwxr-xr-x  1 repentinus repentinus    129 Feb 17 08:44 002-nwipe-banner-patch.sh
-rw-r--r--  1 repentinus repentinus  54591 Feb 17 08:44 aclocal.m4
-rw-r--r--  1 repentinus repentinus      0 Feb 17 08:44 .applied_patches_list
-rwxr-xr-x  1 repentinus repentinus    116 Dec 19 18:32 autogen.sh
drwxr-xr-x  2 repentinus repentinus   4096 Feb 17 08:44 autom4te.cache
-rw-r--r--  1 repentinus repentinus  10210 Dec 19 18:32 CHANGELOG.md
-rw-r--r--  1 repentinus repentinus   2292 Dec 19 18:32 .clang-format
lrwxrwxrwx  1 repentinus repentinus     32 Feb 17 08:44 compile -> /usr/share/automake-1.16/compile
lrwxrwxrwx  1 repentinus repentinus     37 Feb 17 08:44 config.guess -> /usr/share/automake-1.16/config.guess
-rw-r--r--  1 repentinus repentinus   3495 Feb 17 08:44 config.h.in
lrwxrwxrwx  1 repentinus repentinus     35 Feb 17 08:44 config.sub -> /usr/share/automake-1.16/config.sub
-rwxr-xr-x  1 repentinus repentinus 235804 Feb 17 08:44 configure
-rw-r--r--  1 repentinus repentinus   2101 Dec 19 18:32 configure.ac
-rw-r--r--  1 repentinus repentinus  18092 Dec 19 18:32 COPYING
lrwxrwxrwx  1 repentinus repentinus     32 Feb 17 08:44 depcomp -> /usr/share/automake-1.16/depcomp
drwxr-xr-x  3 repentinus repentinus   4096 Dec 19 18:32 .github
-rw-r--r--  1 repentinus repentinus    251 Dec 19 18:32 .gitignore
drwxr-xr-x  2 repentinus repentinus   4096 Dec 19 18:32 images
lrwxrwxrwx  1 repentinus repentinus     35 Feb 17 08:44 install-sh -> /usr/share/automake-1.16/install-sh
-rw-r--r--  1 repentinus repentinus    226 Dec 19 18:32 Makefile.am
-rw-r--r--  1 repentinus repentinus  25358 Feb 17 08:44 Makefile.in
drwxr-xr-x  2 repentinus repentinus   4096 Feb 17 08:44 man
lrwxrwxrwx  1 repentinus repentinus     32 Feb 17 08:44 missing -> /usr/share/automake-1.16/missing
-rw-r--r--  1 repentinus repentinus   9213 Dec 19 18:32 README.md
drwxr-xr-x  4 repentinus repentinus   4096 Feb 17 08:44 src
-rw-r--r--  1 repentinus repentinus      0 Feb 17 08:44 .stamp_downloaded
-rw-r--r--  1 repentinus repentinus      0 Feb 17 08:44 .stamp_extracted

```

The symlinks pointing to `/usr/share/automake-1.16/*` are created by autogen.sh. If I manually delete them and run the script, they are recreated. The buildroot package should make sure to use the autotools in `output/host/` when running the script, but it ends up using my system autotools instead. Hence, the subsequent buildroot autotools-package logic is unable to replace the files because it tries to remove the system files pointed to by the symlinks, rather than the symlinks itself. I do not know enough about buildroot to fix it, though.

PartialVolume, 2021-02-17:
Thanks for the patch. I'm still puzzled by this. Here's my output/build/nwipe-v0.30.001/ directory.
```
drwxr-xr-x   7 nick nick   4096 Feb 17 11:47 .
drwxr-xr-x 125 nick nick   4096 Feb 17 12:21 ..
-rwxr-xr-x   1 nick nick    129 Feb 17 11:47 002-nwipe-banner-patch.sh
-rw-r--r--   1 nick nick  54818 Feb 17 11:47 aclocal.m4
-rw-r--r--   1 nick nick      0 Feb 17 11:47 .applied_patches_list
-rwxr-xr-x   1 nick nick    116 Dec 19 18:32 autogen.sh
drwxr-xr-x   2 nick nick   4096 Feb 17 11:47 autom4te.cache
-rw-r--r--   1 nick nick  10210 Dec 19 18:32 CHANGELOG.md
-rw-r--r--   1 nick nick   2292 Dec 19 18:32 .clang-format
lrwxrwxrwx   1 nick nick     32 Feb 17 11:47 compile -> /usr/share/automake-1.15/compile
-rw-r--r--   1 nick nick   3613 Feb 17 11:47 config.h
-rw-r--r--   1 nick nick   3350 Feb 17 11:47 config.h.in
-rw-r--r--   1 nick nick  40276 Feb 17 11:47 config.log
-rwxr-xr-x   1 nick nick  37610 Feb 17 11:47 config.status
-rwxr-xr-x   1 nick nick 228224 Feb 17 11:47 configure
-rw-r--r--   1 nick nick   2101 Dec 19 18:32 configure.ac
-rw-r--r--   1 nick nick  18092 Dec 19 18:32 COPYING
lrwxrwxrwx   1 nick nick     32 Feb 17 11:47 depcomp -> /usr/share/automake-1.15/depcomp
-rw-r--r--   1 nick nick     57 Feb 17 11:47 .files-list.txt
drwxr-xr-x   3 nick nick   4096 Dec 19 18:32 .github
-rw-r--r--   1 nick nick    251 Dec 19 18:32 .gitignore
drwxr-xr-x   2 nick nick   4096 Dec 19 18:32 images
lrwxrwxrwx   1 nick nick     35 Feb 17 11:47 install-sh -> /usr/share/automake-1.15/install-sh
-rw-r--r--   1 nick nick  26667 Feb 17 11:47 Makefile
-rw-r--r--   1 nick nick    226 Dec 19 18:32 Makefile.am
-rw-r--r--   1 nick nick  24735 Feb 17 11:47 Makefile.in
drwxr-xr-x   2 nick nick   4096 Feb 17 11:47 man
lrwxrwxrwx   1 nick nick     32 Feb 17 11:47 missing -> /usr/share/automake-1.15/missing
-rw-r--r--   1 nick nick   9213 Dec 19 18:32 README.md
drwxr-xr-x   4 nick nick   4096 Feb 17 11:47 src
-rw-r--r--   1 nick nick      0 Feb 17 11:47 .stamp_built
-rw-r--r--   1 nick nick      0 Feb 17 11:47 .stamp_configured
-rw-r--r--   1 nick nick      0 Feb 17 11:47 .stamp_downloaded
-rw-r--r--   1 nick nick      0 Feb 17 11:47 .stamp_extracted
-rw-r--r--   1 nick nick     23 Feb 17 11:47 stamp-h1
-rw-r--r--   1 nick nick      0 Feb 17 11:47 .stamp_patched
-rw-r--r--   1 nick nick      0 Feb 17 11:47 .stamp_target_installed
```
If you take the file `compile`,  it is also symbolically linked to /usr/share/automake-1.15/compile the same as yours and looking at the ownership and permissions of compile it also is not writable by compile, yet it's not failing for me.
```
/usr/share/automake-1.15$ ls -al
total 520
drwxr-xr-x   4 root root   4096 Jan  8  2019 .
drwxr-xr-x 545 root root  20480 Feb  4 19:06 ..
drwxr-xr-x   2 root root   4096 Jan  8  2019 am
-rwxr-xr-x   1 root root   5826 Mar 26  2018 ar-lib
drwxr-xr-x   2 root root   4096 Jan  8  2019 Automake
-rwxr-xr-x   1 root root   7333 Mar 26  2018 compile
lrwxrwxrwx   1 root root     20 Mar 26  2018 config.guess -> ../misc/config.guess
lrwxrwxrwx   1 root root     18 Mar 26  2018 config.sub -> ../misc/config.sub
-rw-r--r--   1 root root  35147 Mar 26  2018 COPYING
-rwxr-xr-x   1 root root  23567 Mar 26  2018 depcomp
-rw-r--r--   1 root root  15756 Mar 26  2018 INSTALL
-rwxr-xr-x   1 root root  15155 Mar 26  2018 install-sh
-rwxr-xr-x   1 root root   6102 Mar 26  2018 mdate-sh
-rwxr-xr-x   1 root root   6872 Mar 26  2018 missing
-rwxr-xr-x   1 root root   3538 Mar 26  2018 mkinstalldirs
-rwxr-xr-x   1 root root   4671 Mar 26  2018 py-compile
-rwxr-xr-x   1 root root  19465 Mar 26  2018 tap-driver.sh
-rwxr-xr-x   1 root root   4640 Mar 26  2018 test-driver
-rw-r--r--   1 root root 323102 Mar 26  2018 texinfo.tex
-rwxr-xr-x   1 root root   6859 Mar 26  2018 ylwrap
```
However if your patch works ok on ubuntu and I don't see any reason why it shouldn't, I'll commit it.

Repentinus, 2021-02-17:
For some strange reason `config.guess` and `config.sub` in your `/usr/share/automake-1.15` (and the original Ubuntu package) are world writable symlinks. That is why you do not get a build failure, albeit you are still inconsistently mixing automake from /usr/share and output/host (which is what buildroot uses internally).

PartialVolume, 2021-02-17:
Closed by #27

closed 2021-02-17 by PartialVolume