PartialVolume/shredos.x86_64 #30: Shutdown / --autopoweroff  is broken

betaprinz, 2021-04-19:
my grub.cfg entry:

menuentry "ShredOS-ZERO-NoUSB-AutoOFF-AllDrives" {
	linux /boot/shredos console=tty3 loglevel=3 loadkeys=de nwipe_options="--method=zero --verify=off --noblank --nousb --autonuke --autopoweroff"
}

After wipping my disk i get a error/message 

"shutdown: -H and -P flags can only be used along with -h flag".
...
...
...

"Paused, press any key to restart nwipe."

Any Ideas?

PartialVolume, 2021-04-19:
@betaprinz Interesting, shutdown -P (which is what nwipe uses in it's code) works on Ubuntu 18.04LTS without any issues. I'll check it out on buildroot (ShredOS) and keep back to you this afternoon.

PartialVolume, 2021-04-19:
@betaprinz Confirmed, we have two different versions of the shutdown command that behave slightly differently.

The shutdown command that nwipe uses is `shutdown -P +1 "Broadcast shutdown message"`. This works fine on Ubuntu 18.04 LTS but. as you found, on ShredOS fails with the error `shutdown: -H and -P flags can only be used along with -h flag".`

Luckily there is a shutdown command with options that work correctly on both ShredOS and Ubuntu 18.04 LTS, this is `shutdown -Ph +1 "Broadcast shutdown message"`. i.e with the addition of -h.

I'll make the changes to nwipe and once that's done I'll release another version of ShredOS.

Thanks for reporting the problem.

PartialVolume, 2021-06-09:
Fixed by #37

closed 2021-06-09 by PartialVolume