PartialVolume/shredos.x86_64 #31: How to create bootable ISO from .IMG?

CoodyCZ, 2021-05-04:
Hi! I'm sorry for my question, but is there any option make from your .IMG bootable ISO for using in VirtualBox? Because latest IMG cannot boot.

Thanks for reply!

PartialVolume, 2021-05-04:
@CoodyCZ I've never tried this but maybe using convertdd to convert .IMG to.vdi will allow you to run in virtual box

[https://www.unixmen.com/linux-basics-convert-img-files-to-vdi-format-for-oracle-virtualbox/](https://www.unixmen.com/linux-basics-convert-img-files-to-vdi-format-for-oracle-virtualbox/)

CoodyCZ, 2021-05-04:
@PartialVolume thanks for quick answer! 
I have one more question... Will you release ISO version of ShredOS?

PartialVolume, 2021-05-04:
@CoodyCZ yes, that's the plan, I did have a go at it and was hoping it would be something I could implement quickly into the current build, but a had problems so has been delayed, currently I'm working on code in nwipe and when I'm through with that I may have time to look at an ISO build.

CoodyCZ, 2021-05-05:
@PartialVolume thank you for good news! ... btw... i've converted .img to .vdi... it was possible to boot into shredos, but he didn't see the vbox drive on which win10 is running.  That's why I asked for an ISO, it can be inserted "into the drive" and then it should normally see any virtual disk. Anyway, thanks for the answers.

PartialVolume, 2021-05-05:
No problem, in the meantime if anybody else should come across this thread and fancies doing a pull request to add an .iso build to the existing .img I'm happy to take a look at it.

PartialVolume, 2021-06-14:
64 bit legacy .iso for CD-R/DVD-R has been released. 64 bit legacy/UEFI .iso in development for imminent release. 32 bit version to follow.

closed 2021-06-14 by PartialVolume

deajan, 2021-06-30:
@PartialVolume Out of curiosity, I was using xorriso in order to try to change the ShredOS img to an iso (never succeeded :()
How did you do it ?

PartialVolume, 2021-06-30:
@deajan Basically enabling creation of iso9660 in buildroot, i.e. running `make menuconfig` and in the boot loaders section and enabling iso9660 creation. At a more detailed level buildroot uses genisoimage to do the actual creation if the .iso:
```
genisoimage -J -R -b boot/grub/grub-eltorito.img -no-emul-boot -boot-load-size 4 -boot-info-table  -o /alongoldpath/rootfs.iso9660 /anotherlongoldpath/rootfs.iso9660.tmp
```
There is a little more detail in terms of modifying the iso9660.mk file so that I can insert /boot/grub stuff into the iso
After this has been created, I simply rename the created .iso file to be consistent  with the filename used when creating the.img.

So the .iso is not actually created from the .img, it's created independently.

I've been working on the UEFI method today and to be honest I'm spending most of my time reading and examining the way other people have achieved it, like for instance with the gparted live CD. It's a steep learning curve but I'll get there in the end.

The current ShredOS source doesn't include the changes I made to create the legacy .iso file however as soon as I've got UEFI boot working on a DVD on my mac and Dell Optiplex 9010 then I'll commit the code.

For reference the iso9660.mk file that basically does the creation is below, it's had a few changes made by me, i.e adding UEFI stuff, which currently isn't working but that's basically how I did it. By the time I've finished I'm pretty sure it will look a lot different.

If you are looking for something that creates a UEFI bootable .iso from the .iso or img file I can't really help with that but I believe programs like E2B and creating a .imgPTN from the .img file look like they might do the job [https://www.easy2boot.com/add-payload-files/adding-uefi-images/](https://www.easy2boot.com/add-payload-files/adding-uefi-images/). Like I mentioned though, I've never used it so can't offer much help with that.

Hopefully though, once I've figured it out I should have a combined UEFI/Legacy with both Grub2 and isolinux/syslinux bootloaders working in the form of an .iso for CD/DVD. 

```
################################################################################
#
# Build the iso96600 root filesystem image
#
################################################################################

#
# We need to handle three cases:
#
#  1. The ISO9660 filesystem will really be the real root filesystem
#     itself. This is when BR2_TARGET_ROOTFS_ISO9660_INITRD is
#     disabled.
#
#  2. The ISO9660 filesystem will be a filesystem with just a kernel
#     image, initrd and grub. This is when
#     BR2_TARGET_ROOTFS_ISO9660_INITRD is enabled, but
#     BR2_TARGET_ROOTFS_INITRAMFS is disabled.
#
#  3. The ISO9660 filesystem will be a filesystem with just a kernel
#     image and grub. This is like (2), except that the initrd is
#     built into the kernel image. This is when
#     BR2_TARGET_ROOTFS_INITRAMFS is enabled (regardless of the value
#     of BR2_TARGET_ROOTFS_ISO9660_INITRD).

ROOTFS_ISO9660_BOOT_MENU = $(call qstrip,$(BR2_TARGET_ROOTFS_ISO9660_BOOT_MENU))

ROOTFS_ISO9660_DEPENDENCIES = host-cdrkit linux

ifeq ($(BR2_TARGET_ROOTFS_INITRAMFS),y)
ROOTFS_ISO9660_USE_INITRD = YES
endif

ifeq ($(BR2_TARGET_ROOTFS_ISO9660_INITRD),y)
ROOTFS_ISO9660_USE_INITRD = YES
endif

ifeq ($(ROOTFS_ISO9660_USE_INITRD),YES)
ROOTFS_ISO9660_TMP_TARGET_DIR = $(FS_DIR)/rootfs.iso9660.tmp
define ROOTFS_ISO9660_CREATE_TEMPDIR
	$(RM) -rf $(ROOTFS_ISO9660_TMP_TARGET_DIR)
	mkdir -p $(ROOTFS_ISO9660_TMP_TARGET_DIR)
endef
ROOTFS_ISO9660_PRE_GEN_HOOKS += ROOTFS_ISO9660_CREATE_TEMPDIR
else ifeq ($(BR2_TARGET_ROOTFS_ISO9660_TRANSPARENT_COMPRESSION),y)
ROOTFS_ISO9660_DEPENDENCIES += host-zisofs-tools
ROOTFS_ISO9660_TMP_TARGET_DIR = $(FS_DIR)/rootfs.iso9660.tmp
# This must be early, before we copy the bootloader files.
define ROOTFS_ISO9660_MKZFTREE
	$(RM) -rf $(ROOTFS_ISO9660_TMP_TARGET_DIR)
	$(HOST_DIR)/bin/mkzftree -X -z 9 -p $(PARALLEL_JOBS) \
[... 89 lines cut ...]
ROOTFS_ISO9660_PRE_GEN_HOOKS += ROOTFS_ISO9660_COPY_KERNEL
endif

ROOTFS_ISO9660_PRE_GEN_HOOKS += ROOTFS_ISO9660_DISABLE_EXTERNAL_INITRD

endif # ROOTFS_ISO9660_USE_INITRD

define ROOTFS_ISO9660_CMD
	$(HOST_DIR)/bin/genisoimage -J -R -b $(ROOTFS_ISO9660_BOOT_IMAGE) \
		-no-emul-boot -boot-load-size 4 -boot-info-table \
		$(ROOTFS_ISO9660_GENISOIMAGE_OPTS) \
		-o $@ $(ROOTFS_ISO9660_TMP_TARGET_DIR)
endef

ifeq ($(BR2_TARGET_ROOTFS_ISO9660_HYBRID),y)
define ROOTFS_ISO9660_GEN_HYBRID
	$(HOST_DIR)/bin/isohybrid -t 0x96 $@
endef

ROOTFS_ISO9660_POST_GEN_HOOKS += ROOTFS_ISO9660_GEN_HYBRID
endif

$(eval $(rootfs))

```