PartialVolume/shredos.x86_64 #32: The last 'verifying' work is shown as 'blanking' in GUI

alphanis, 2021-05-26:
Bug report.

In GUI, shredos shows current work as 'writing', 'blanking', 'verifying' and 'syncing' each drive.
But on the last verify AFTER blanking, current status does not change and still shows 'blanking'.
I guess it actually does verify properly. I've tested it on virtual machine by making currupt bit on drive right after shredos blanked it and right before shredos verify it. shredos detected the currupt bit and told me there was an error. Stdout log also shows verifying work was started, finished. (and errors if they exist)
So the only problem is displaying status in GUI.

To reproduce this bug:
1. Boot shredos (I've tested in UEFI boot)
2. Set Method: prng
3. Select drive and Start.
4. It shows 'writing' status and intermittently 'syncing'.
5. After arround 33%, status changes to 'blanking' and intermittently 'syncing'.
6. After arround 66%, status should change to 'verifying'. But it never changes and still shows 'blanking'. This time, 'syncing' does not happen.
7. You can check time lines on log screen when you quit shredos, that says it actually DOES verify.

PartialVolume, 2021-05-26:
Yes, I think you are correct, it's a display issue in nwipe. Can you open up an issue over at https://github.com/martijnvanbrummelen/nwipe and I'll deal with it over there. Thanks.

alphanis, 2021-05-27:
I did on nwipe issue. Thank you.

closed 2021-06-09 by PartialVolume