PartialVolume/shredos.x86_64 #34: When UEFI CSM disabled, screen does not appear but it boots and shreds.

alphanis, 2021-06-02:
With CSM(Compatibility Support Module), ShredOS works with both CSM UEFI and CSM BIOS. I can see large screen resolution with UEFI and small screen resolution with BIOS.

When I disable CSM in mother board firmware(del key on boot), my PC boots until POST, and string `recalbox` appears on screen, and it freezes. BUT I think ShredOS boots and works, because HDD LED blinks, and stops blinking when I hit Ctrl+c. I can even turn off my PC with Alt+F2, poweroff command.

I did not enable fast boot. Just disabled CSM.

Is this a problem of my own mother board, or ShredOS?

Test environment:
CPU Intel Celeron G5905 (Cometlake S) 
M/B ASUS PRIME H410M-A
RAM 4GiB*1
with following grub.cfg edit:
[grub.cfg.txt](https://github.com/PartialVolume/shredos.x86_64/files/6582550/grub.cfg.txt)

PartialVolume, 2021-06-02:
I've not had my second cup of coffee yet, but looking at the specs for the Intel H410M-A chipset, through my bleary eyes, this particular Intel chipset had a release date of March 2020? If that's correct I would guess it's just a video driver compatibility issue. ShredOS being currently built on buildroot 2020.02 almost certainly doesn't have the most up to date driver for that hardware. I will be updating ShredOS hopefully soon to the latest buildroot 2021-02 which should have the updated Intel drivers.

Does that tie in with what you know about your laptop's age?

I have a MAC and a older PC that both boot UEFI and recalbox appears and graphics come up in higher resolution with any problem, so this older hardware does work ok with that vintage of video hardware.

alphanis, 2021-06-02:
Thank you for your quick response.

Yes, Intel H410 chipset is quite new, launched 2020. I think your guess about driver would be right. I hope next buildroot will have latest intel drivers.

I'm sorry but I don't have other PC or laptop to test. I'll test and report if I have a chance.

Firminator, 2021-12-29:
This should be resolved with the latest version released ~2 or 3 weeks ago. Alphanis, if you don't mind could you test and report back here. Else this issues can probably be closed.