PartialVolume/shredos.x86_64 #35: IMG files not compatible with ImgBurn

cdoublejj, 2021-06-04:
when i try to burn the img file to CD IMG burn says it is not a compatible IMG file.

CoodyCZ, 2021-06-09:
@PartialVolume  @cdoublejj i have the same problem, i'm waiting for the ISO

PartialVolume, 2021-06-12:
I've built a .iso image of the latest ShredOS if anybody is interested in testing it? It only supports legacy bios booting at the moment, however, I'm working on a combined LegacyBios/UEFI version that I hope to release soon.

In the meantime, if you switch secure boot off and switch to legacy bios mode you should be able to boot from this CD/DVD. You can obtain the .iso [here.](https://github.com/PartialVolume/shredos.x86_64/releases/download/v2020.05.012_x86-64_0.31/shredos-2020.05.012_x86-64_0.31_20210612_legacy_bios.iso)

If you do try it, please let me know how how it went.

PartialVolume, 2021-06-12:
Nearly forgot, the above legacy ISO is for 64bit systems, a 32bit version will follow once I've built the combined LegacyUEFI root.

CoodyCZ, 2021-06-14:
@PartialVolume  thanks for quick release! awesome work, i'm testing it :) AND YES! 
It is absolutely great working :) (have tested on legacy bios, have tested with VENTOY, it's working!

I hope that the UEFI version will work as well, this is really important because most new computers often run only on uefi (it is not possible to switch to legacy). 

SO THANKS FOR YOUR WORK!

CoodyCZ, 2021-06-14:
@PartialVolume ... in last 5 minutes i have tested it on ONLY UEFI over app VENTOY, and it is booting in interactive mode, WORKING TOO!

Firminator, 2021-06-25:
Boots fine on a Dell Precision Workstation with BIOS (no UEFI / Tioancore / CSM).
Awesome. Thanks.

closed 2026-02-14 by PartialVolume