PartialVolume/shredos.x86_64 #36: Typo in README.md

alphanis, 2021-06-08:
https://github.com/PartialVolume/shredos.x86_64/blob/master/README.md#how-to-change-the-default-nwipe-options-so-the-change-persists-between-reboots

` --sync=NUM          Will perform a sync after NUM writes (default: 100000)`

I think default sync value is 10000. Not 100000.
refer: https://github.com/martijnvanbrummelen/nwipe/blob/master/man/nwipe.1

PartialVolume, 2021-06-08:
@alphanis The error is not in ShredOS but in nwipes man page. The default sync rate is 100000 as defined in https://github.com/martijnvanbrummelen/nwipe/blob/master/src/options.h , the error is in the man page file reporting it as 10000. https://github.com/martijnvanbrummelen/nwipe/blob/master/man/nwipe.1

I'll do a patch to nwipe to update that.

Thanks for reporting, much appreciated.

PartialVolume, 2021-06-08:
Fixed in nwipe by https://github.com/martijnvanbrummelen/nwipe/commit/47e523cf34025aa1fb12bdb4c113b4f67540fa1c

closed 2021-06-08 by PartialVolume