PartialVolume/shredos.x86_64 #38: hot swap support

ITJamie, 2021-07-23:
Is / Would it be possible to have hot swap support with auto wiping?

We want to build a system that we can pop drives into and have them automatically wipe and unmount when complete. we would remove them once there is no more drive activity.

PartialVolume, 2021-07-25:
Sort of related to [this](https://github.com/martijnvanbrummelen/nwipe/issues/264), but that was talking about updating the drive selection window, I think you may be talking about new drives appearing and being wiped automatically while other drives are still currently being wiped? I would imagine this being a scenario where you would have a lot of drives to wipe of differing sizes that complete their wipes at different times, so when a smaller drive completes a new drive can be hot swapped while the others are still wiping.

I think that's a feature that's worth having.

It will require changes to the way nwipe detects drives, it will probably still use libparted for drive detection, but once started will monitor  for drives using a method that doesn't hang drive detection when drives are being wiped. Alternatively libparted for drive detection may get ditched which would then allow multiple Instances of nwipe running simultaneously, something other uses have requested.

ITJamie, 2021-07-25:
yes exactly that. We basically want a server that we pop all drives into from decommissioned servers and have them autowipe. 

an additional nicety would be the ability to have it send a syslog event or even an email via a relay to say that drive has been wiped with its serial + wipe method used etc.
Being able to have it syslog or email like that to a permanent location would be fantastic for audit / legal reasons.

realistically i think (based on my quick scan through the code) the easier solution would be multiple instances of nwipe running and having some newer app or script watchdog watching for system events for new drive connections and then point a nwipe process at that specific drive.

ITJamie, 2021-09-23:
Just wondering if this is something you are still considering?

PartialVolume, 2021-09-24:
Yes, it's still a feature worth adding. Things are busy for me at the moment as I'm up to my eye balls with programming that pays the bills, so I'm not able to devote as much time as I would like to nwipe/ShredOS. Usually there's a gap between projects where I can spend a bit of time on this and hopefully that's coming up soon.

qedisme, 2021-09-24:
The is was be a killer feature, and would make our decom and donation process so much easier. @PartialVolume please consider bumping this up in your feature list. Already love this distro, but this would be a boon to anybody that has a donation program.