PartialVolume/shredos.x86_64 #39: Drivers for HW cards

skyline-65, 2021-09-01:
Is it possible to add drivers that support HW cards/adapters such as Dell Perc 310 so I can erase all 24 drives in my server case? Is there an easy way I could add to the USB boot stick?

PartialVolume, 2021-09-01:
I'm assuming your switching the hardware into non raid mode ? Anything else won't make much sense to nwipe. Doesn't nwipe then see the drives once the controller is in non raid mode?

What sort of drives are they? If the drivers are available in buildroot they can be added to ShredOS quite easily (but still requires a compile of buildroot) And if they are not available in buildroot but the source is available it may be possible to compile and built into ShredOS anyway.

skyline-65, 2021-09-01:
Yes non Raid. A typical Freenas/Truenas setup. I use an Intel E91267-203 RAID Expander Storage Adaptor which then feeds a PERC H310 which has been crossflashed to IT mode LSI 9211-8i. One set of 4 drives is directly attached to the Dell Perc and they cannot be seen either. All drives are a mix of sizes and are sata disks. I suspect it’s the LSI 9211-8i (aka Dell Perc) that doesn’t have the drivers.

Cheers

PartialVolume, 2021-09-01:
The megaraid SAS driver which supports Dell Perc, LSI, Intel, FSC, ACER devices is available.

I'll rebuild ShredOS with this driver if you can test it out for me?

skyline-65, 2021-09-01:
Yes no problem.

Firminator, 2021-09-14:
The drivers to enable a variety of 3rd party RAID controllers (in HBA mode) are:
* MEGARAID_LEGACY
* MEGARAID_SAS
* MEGARAID_MPT2SAS

These should enable DELL server support (DELL-branded LSI controllers, possibly IBM-branded LSI controllers and a bunch of others), and also the unbranded/original LSI controllers. Those were used in NHellFire's DBAN fork and they worked with Dell hardware.

PartialVolume, 2021-09-14:
@Firminator thanks, I'll get those added ASAP.

Firminator, 2021-09-14:
You're the man. Thanks! Let us know.

Firminator, 2021-09-14:
see
https://github.com/NHellFire/dban/wiki/Supported-Hardware
https://github.com/NHellFire/dban-buildroot/blob/dban/target/dban/linux-i586.config

Firminator, 2021-09-14:
If you build an ISO like you did with https://github.com/PartialVolume/shredos.x86_64/releases/download/v2020.05.012_x86-64_0.31/shredos-2020.05.012_x86-64_0.31_20210612_legacy_bios.iso I'll be able to test the on my wiping box.

derekschrock, 2021-09-27:
Any updates on this for an iso to test with?  I have a megaraid_sas device (9440-8i) I'd like to test with.

Firminator, 2021-09-27:
He's pretty covered right now accd. to a recent comment:
# https://github.com/PartialVolume/shredos.x86_64/issues/38#issuecomment-926252171

PartialVolume, 2021-10-04:
@Carl-Wilhelm much appreciated. I'm hopefully going to try and get a new release out by the end of this week, with a bit of luck.

Carl-Wilhelm, 2021-10-12:
> I'm currently working on the latest release, I'm just wondering whether adding iSCSI drivers would be of any practical use to somebody out there ?

Yes, absolutely if it is not to much extra work for you. 😊

Carl-Wilhelm, 2021-10-14:
> @skyline-65 @Firminator @Carl-Wilhelm See latest pre-release [v2020.05.013_x86-64_0.31](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2020.05.013_x86-64_0.31) and .img for writing to a USB flash drive. [2020.05.013_x86-64_0.31_20211013.img for USB flash drive](https://github.com/PartialVolume/shredos.x86_64/releases/download/v2020.05.013_x86-64_0.31/shredos-2020.05.013_x86-64_0.31_20211013.img). .ISO image for CD/DVD to follow.
> 
> If you could give it a try with your raid hardware and let me know how you get on. Many Thanks.

Did some quick testing now.
Tested with the following:
Dell H710 RAID controller with no configuration - Did not work
Dell H710 RAID controller with each disk set up with a RAID-0 configuration - Did not work
Dell H710 controller flashed to LSI SAS2308 IT-mode FW (HBA) - Did not work
Dell H310 controller flashed to LSI SAS9211 IT-mode FW (HBA) - Did not work

When i say "Did not work", i mean that no disk was visible in nwipe after booting up.

Booted up Ubuntu with the last controller, compiled nwipe and was able to run shedding on the disks.
Is there any info i can give you that can help your debugging/developing?
I also have a IBM HBA card i can try later.

PartialVolume, 2021-10-14:
@Carl-Wilhelm I've made some changes to config so that the drivers are builtin to the kernel rather than modules. I'm currently compiling and should have a new version for you to try later today. (Takes about 3 hours to compile).

Carl-Wilhelm, 2021-10-17:
> @Carl-Wilhelm That's great news, thanks for checking those out. If you haven't already done so can you also use hexedit /dev/xxxx on the Shredos disc or your preferred hex editor to check the disks are being wiped correctly. Maybe a single PRNG stream, no blanking, no verification and then check with the hex editor, the first and last sectors and random sectors in the middle of the disc contain random data and if ok follow up with a single zero pass, no blanking, no verification. Then do the same checks with the hexeditor to confirm the disc has been zeroed. There is no reason why the disc's shouldn't be wiped, but I like to test these things as much as possible. I'll release a official version shortly (version shredos-2020.05.015_x86-64_0.31) that contains the new RAID drivers plus more USB keyboard drivers and the start of adding network support so wipe status can be uploaded to a server via a file or possibly via TCP/IP socket to a KDE program that can monitor a farm of nwipe's.

Took me a couple of days, but checked now. After a PRNG i can confirm that the drive is random, and after blanking, it is only zero.
Works like a charm!

PartialVolume, 2021-10-31:
Implemented in official release [v2020.05.016_x86-64_0.32](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2020.05.016_x86-64_0.32)

closed 2021-10-31 by PartialVolume

Firminator, 2021-11-01:
LSI SAS2308 (FW 18.00.00/7.53) using the newly added mpt2sas driver: attached drives are recognized and the wiping starts.

However it shows UNK instead of SCSI/SAS/SATA/IDE in the main NWIPE GUI:
/dev/sda/ UNK (DriveSize GB) SEAGATE %Model#%

I see that SCSI is theoretically supported in https://github.com/martijnvanbrummelen/nwipe/blob/master/src/device.c but something doesn't play well.

PartialVolume, 2021-11-01:
@Firminator The device name for SAS is missing from the table of valid devices that are displayed in the GUI. So just a GUI issue for the SAS device.

I'm just wondering, but are you able to compile nwipe from source? The reason I ask is that I don't have any SAS drives to test with. What I want to do is add an extra line in the log code that displays the raw data returned by ioctl on HDIO_GET_IDENTITY. Once I know what value is being returned I can then add SAS to the device table.

If you've not compiled nwipe from source then no problem, I can do another pre-release of Shredos, it just takes longer to find out if I've fixed it properly or not.

Firminator, 2021-11-01:
I tried compiling once but it errored out somewhere in the process and there were at least a dozen reasons why it could have been so I had to drop it. Is there a command I can run on the ALT+F2 terminal maybe that would give you the info you need?

Firminator, 2021-11-01:
I see now. So SCSI support is there, but SAS is missing (just like with NVMe and VIRTUAL was up until you added both in v0.30).

PartialVolume, 2021-11-01:
On Shredos in one of the virtual terminals (ALT-F2) could you type
```
readlink -v /sys/block/sdb (or whatever your SAS device name is)
```
It should return something like..
```
../devices/pci0000:00/0000:00:1f.2/ata2/host1/target1:0:0/1:0:0:0/block/sdb
```

If you could past that, that should be all I need.

PartialVolume, 2021-11-01:
I just realised, I'd put some code in nwipe to make it easier next time we had an UNK device. In the Shredos ALT-F2 terminal, type `nwipe -v`. It will run nwipe then just Control-C the nwipe log text will show the result of the readlink command above.

```
[2021/11/01 18:28:15]    info: nwipe 0.32
[2021/11/01 18:28:15]    info: Linux version 5.11.0-37-generic (buildd@lcy01-am
                               d64-021) (gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.
                               3.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #41
                               ~20.04.2-Ubuntu SMP Fri Sep 24 09:06:38 UTC 202
                               1
[2021/11/01 18:28:16]   debug: Readlink: ../devices/pci0000:00/0000:00:1f.2/ata1/host0/target0:0:0/0:0:0:0/block/sda 
[2021/11/01 18:28:16]   debug: smartctl: smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.11.0-37-generic] (local build) 
[2021/11/01 18:28:16]   debug: smartctl: Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org 
[2021/11/01 18:28:16]   debug: smartctl: === START OF INFORMATION SECTION === 
[2021/11/01 18:28:16]   debug: smartctl: Device Model:     SanDisk SDSSDH3 500G 
[2021/11/01 18:28:16]   debug: smartctl: Serial Number:    2004DD455109 
[2021/11/01 18:28:16]   debug: smartctl: LU WWN Device Id: 5 001b44 4a8325368 
[2021/11/01 18:28:16]   debug: smartctl: Firmware Version: 401120RL 
[2021/11/01 18:28:16]   debug: smartctl: User Capacity:    500,107,862,016 bytes [500 GB] 
[2021/11/01 18:28:16]   debug: smartctl: Sector Size:      512 bytes logical/physical 
[2021/11/01 18:28:16]   debug: smartctl: Rotation Rate:    Solid State Device 
[2021/11/01 18:28:16]   debug: smartctl: Form Factor:      2.5 inches 
[2021/11/01 18:28:16]   debug: smartctl: Device is:        Not in smartctl database [for details use: -P showall] 
[2021/11/01 18:28:16]   debug: smartctl: ATA Version is:   ACS-4 T13/BSR INCITS 529 revision 5 
[2021/11/01 18:28:16]   debug: smartctl: SATA Version is:  SATA 3.3, 6.0 Gb/s (current: 6.0 Gb/s) 
[2021/11/01 18:28:16]   debug: smartctl: Local Time is:    Mon Nov  1 18:28:16 2021 GMT 
[2021/11/01 18:28:16]   debug: smartctl: SMART support is: Available - device has SMART capability. 
[2021/11/01 18:28:16]   debug: smartctl: SMART support is: Enabled 
[2021/11/01 18:28:16]  notice: Found /dev/sda,  ATA, SanDisk SDSSDH3, 500 GB, S/N=2004DD455109
```
From the above output, you will see the readlink output which is the line I need.

Firminator, 2021-11-01:
../devices/pci0000:00/0000:00:07.0/0000:03:00.0/host0/port-0:0/end-device-0:0/target0:0:0/0:0:0:0/block/sda

PartialVolume, 2021-11-01:
Can you type the 'nwipe -v' command as well.

Looks like the readlink output might be useful for ATA and USB etc but not very informative for your controller/driver. The smartctl output displayed above with nwipe -v might be more useful.

Firminator, 2021-11-01:
Where is nwipe -v dropping the logfile?

PartialVolume, 2021-11-01:
It doesn't write the log file by default, you have to use the -l or --logfile=FILE option
```
nwipe  -v --logfile=nwipe.log    ( Default is STDOUT )
```

Firminator, 2021-11-01:
```
[2021/11/01 10:37:46]   debug: Readlink: ../devices/pci0000:00/0000:00:07.0/0000:03:00.0/host0/port-0:0/end_device-0:0/target0:0:0/0:0:0:0/block/sda 
[2021/11/01 10:37:46]   debug: smartctl: smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.6.3] (local build) 
[2021/11/01 10:37:46]   debug: smartctl: Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org 
[2021/11/01 10:37:46]   debug: smartctl: === START OF INFORMATION SECTION === 
[2021/11/01 10:37:46]   debug: smartctl: Vendor:               HP 
[2021/11/01 10:37:46]   debug: smartctl: Product:              MBXXXXXXXXXXXXXXX
[2021/11/01 10:37:46]   debug: smartctl: Revision:             HPD7 
[2021/11/01 10:37:46]   debug: smartctl: Compliance:           SPC-3 
[2021/11/01 10:37:46]   debug: smartctl: User Capacity:        2,000,398,934,016 bytes [2.00 TB] 
[2021/11/01 10:37:46]   debug: smartctl: Logical block size:   512 bytes 
[2021/11/01 10:37:46]   debug: smartctl: Rotation Rate:        7200 rpm 
[2021/11/01 10:37:46]   debug: smartctl: Form Factor:          3.5 inches 
[2021/11/01 10:37:46]   debug: smartctl: Logical Unit id:      0x5000c50020eea123 
[2021/11/01 10:37:46]   debug: smartctl: Serial number:        XXXXXXXXXXXXXXXXXXXXXXXXX
[2021/11/01 10:37:46]   debug: smartctl: Device type:          disk 
[2021/11/01 10:37:46]   debug: smartctl: Transport protocol:   SAS (SPL-3) 
[2021/11/01 10:37:46]   debug: smartctl: Local Time is:        Mon Nov  1 10:37:46 2021 UTC 
[2021/11/01 10:37:46]   debug: smartctl: SMART support is:     Available - device has SMART capability. 
[2021/11/01 10:37:46]   debug: smartctl: SMART support is:     Enabled 
[2021/11/01 10:37:46]   debug: smartctl: Temperature Warning:  Enabled 
[2021/11/01 10:37:46]  notice: Found /dev/sda,  UNK, HP MBXXXXXXXXXXXXXX,   2 TB, S/N=[note: this was empty]
[2021/11/01 10:37:46]   debug: Readlink: ../devices/pci0000:00/0000:00:07.0/0000:03:00.0/host0/port-0:1/end_device-0:1/target0:0:1/0:0:1:0/block/sdb 
[2021/11/01 10:37:46]   debug: smartctl: smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.6.3] (local build) 
[2021/11/01 10:37:46]   debug: smartctl: Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org 
[2021/11/01 10:37:46]   debug: smartctl: === START OF INFORMATION SECTION === 
[2021/11/01 10:37:46]   debug: smartctl: Vendor:               SEAGATE 
[2021/11/01 10:37:46]   debug: smartctl: Product:              STXXXXXXXXXXXXXXXX
[2021/11/01 10:37:46]   debug: smartctl: Revision:             YS0D 
[2021/11/01 10:37:46]   debug: smartctl: Compliance:           SPC-4 
[2021/11/01 10:37:46]   debug: smartctl: User Capacity:        300,000,000,000 bytes [300 GB] 
[2021/11/01 10:37:46]   debug: smartctl: Logical block size:   512 bytes 
[2021/11/01 10:37:46]   debug: smartctl: Rotation Rate:        15000 rpm 
[2021/11/01 10:37:46]   debug: smartctl: Form Factor:          2.5 inches 
[2021/11/01 10:37:46]   debug: smartctl: Logical Unit id:      0x5000c50054b40017 
[2021/11/01 10:37:46]   debug: smartctl: Serial number:        XXXXXXXXXXXXXXXXXXX
[2021/11/01 10:37:46]   debug: smartctl: Device type:          disk 
[2021/11/01 10:37:46]   debug: smartctl: Transport protocol:   SAS (SPL-3) 
[2021/11/01 10:37:46]   debug: smartctl: Local Time is:        Mon Nov  1 10:37:46 2021 UTC 
[2021/11/01 10:37:46]   debug: smartctl: SMART support is:     Available - device has SMART capability. 
[2021/11/01 10:37:46]   debug: smartctl: SMART support is:     Enabled 
[2021/11/01 10:37:46]   debug: smartctl: Temperature Warning:  Disabled or Not Supported 
[2021/11/01 10:37:46]  notice: Found /dev/sdb,  UNK, SEAGATE STXXXXXXXXXX, 300 GB, S/N=[note: this was empty]
[2021/11/01 10:37:46]   debug: Readlink: ../devices/pci0000:00/0000:00:07.0/0000:03:00.0/host0/port-0:2/end_device-0:2/target0:0:2/0:0:2:0/block/sdc 
[2021/11/01 10:37:46]   debug: smartctl: smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.6.3] (local build) 
[2021/11/01 10:37:46]   debug: smartctl: Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org 
[2021/11/01 10:37:46]   debug: smartctl: === START OF INFORMATION SECTION === 
[2021/11/01 10:37:46]   debug: smartctl: Vendor:               SEAGATE 
[2021/11/01 10:37:46]   debug: smartctl: Product:              STXXXXXXXXXXXXXXXXX
[2021/11/01 10:37:46]   debug: smartctl: Revision:             YS0D 
[2021/11/01 10:37:46]   debug: smartctl: Compliance:           SPC-4 
[2021/11/01 10:37:46]   debug: smartctl: User Capacity:        300,000,000,000 bytes [300 GB] 
[2021/11/01 10:37:46]   debug: smartctl: Logical block size:   512 bytes 
[2021/11/01 10:37:46]   debug: smartctl: Rotation Rate:        15000 rpm 
[2021/11/01 10:37:46]   debug: smartctl: Form Factor:          2.5 inches 
[2021/11/01 10:37:46]   debug: smartctl: Logical Unit id:      0x5000c5005a009057 
[2021/11/01 10:37:46]   debug: smartctl: Serial number:        XXXXXXXXXXXXXx
[2021/11/01 10:37:46]   debug: smartctl: Device type:          disk 
[2021/11/01 10:37:46]   debug: smartctl: Transport protocol:   SAS (SPL-3) 
[2021/11/01 10:37:46]   debug: smartctl: Local Time is:        Mon Nov  1 10:37:46 2021 UTC 
[2021/11/01 10:37:46]   debug: smartctl: SMART support is:     Available - device has SMART capability. 
[2021/11/01 10:37:46]   debug: smartctl: SMART support is:     Enabled 
[2021/11/01 10:37:46]   debug: smartctl: Temperature Warning:  Disabled or Not Supported 
[... cut ...]
[2021/11/01 10:37:46]   debug: smartctl: /dev/sde: Unknown USB bridge [0x1370:0x2168 (0x200)] 
[2021/11/01 10:37:46]   debug: smartctl: Please specify device type with the -d option. 
[2021/11/01 10:37:46]   debug: smartctl: Use smartctl -h to get a usage summary 
[2021/11/01 10:37:46] warning: /dev/sde USB bridge, no passthru support
[2021/11/01 10:37:46]  notice: Found /dev/sde,  USB, SWISSBIT Twist, 261 MB, S/N=(no ATA pass thru)
[2021/11/01 10:37:46]    info: Automatically enumerated 5 devices.
```

PartialVolume, 2021-11-01:
Thanks, that's useful. Before issuing an UNK I need to read the transport protocol and report that as the device.

Firminator, 2021-11-01:
Yup that's what I thought. I checked real quick but the MEGARAID/PERC option @ https://www.smartmontools.org/browser/trunk/smartmontools/smartctl.8.in seem like a rabbithole. Maybe you have the ability to grep out
`smartctl: Transport protocol: `
somewhere and be done :)

PartialVolume, 2021-11-01:
Yes, I'll definitely just strip the info from the output of smartctl.

I've got too many other things to add to nwipe to waste a load of time figuring out how smartmontools gets that info. I've got eight or so new wipe methods to add to nwipe and my personal pet feature I've always wanted to add. A really nice professional looking pdf wipe certificate that contains the details of the drive and wipe.

I'll make the changes to nwipe in the next day or so and let you know when it's ready to test. Thanks for reporting, much appreciated.

Firminator, 2021-11-01:
We all appreciate you working on this I dare to say. Thanks so much.

PartialVolume, 2021-11-01:
@Firminator Thanks, I've just commited the fix upstream on nwipe [https://github.com/martijnvanbrummelen/nwipe/pull/350](https://github.com/martijnvanbrummelen/nwipe/pull/350)

If you are able to compile the nwipe code and confirm that SAS as the bus type now appears against the appropriate drive. (When compiling the code I don't mean the Shredos code as that takes a lot longer with a lot more that can go wrong).

Compiling nwipe from source on a Ubuntu or any Debian based Linux distro should be fairly straight forward [https://github.com/martijnvanbrummelen/nwipe#debian--ubuntu-prerequisites](https://github.com/martijnvanbrummelen/nwipe#debian--ubuntu-prerequisites)

and

[https://github.com/martijnvanbrummelen/nwipe#compilation](https://github.com/martijnvanbrummelen/nwipe#compilation)

Firminator, 2021-11-02:
I'm using this dedicated standalone box with no OS installed. Just the controller and the drives. Maybe someone else in this thread running their actual live (Ubuntu/Debian) OS off of a SAS controller connected disk can compile nwipe and test it? If you publish another beta/test USB .IMG I can test it within seconds today.

PartialVolume, 2021-11-02:
No problem, I can do a ShredOS pre-release that uses the nwipe master. I'll let you know when it's ready.

Firminator, 2021-11-02:
I just tried https://github.com/martijnvanbrummelen/nwipe#automating-the-download-and-compilation-process-for-debian-based-distros and then copied the freshly compiled nwipe binary with SAS support (v32.001) onto a flash drive. Then mounted the flash drive into a live session of ShredOS; copied nwipe from flash drive to /usr/bin to replace the old v0.32.000 with v32.001 then ran nwipe_launcher, but it error'd out with
`/usr/bin/nwipe: error while running shared libraries: libtinfo.so.6: cannot open shared object file: No such file or directory`

PartialVolume, 2021-11-03:
@Firminator Here's Shredos with the master version of nwipe with the fix for the  SAS instead of UNK. [https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2020.05.016_x86-64_0.32.001](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2020.05.016_x86-64_0.32.001)

Firminator, 2021-11-03:
Interesting.
SAS drives now show as SAS (instead of UNK), so that works. However the second disk I have attahced as a test is SATA SSD that can be connected to to SAS controller and it still shows UNK. That's a legit scenario though since SAS connectors are compatible with SATA drives. I'll work on getting the logs...

Firminator, 2021-11-03:
>I wonder if SATA drives can also be attached to this controller

Yes. If you are thinking SCSI drives... they have a different connector not compatible to SAS connectors. Also you can't connect SAS drives to SATA ports/cables btw :)

Firminator, 2021-11-03:
```
[2021/11/03 07:38:37]   debug: Readlink: ../devices/pci0000:00/0000:00:07.0/0000:03:00.0/host0/port-0:0/end_device-0:0/target0:0:0/0:0:0:0/block/sda 
[2021/11/03 07:38:37]   debug: smartctl: smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.6.3] (local build) 
[2021/11/03 07:38:37]   debug: smartctl: Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org 
[2021/11/03 07:38:37]   debug: smartctl: === START OF INFORMATION SECTION === 
[2021/11/03 07:38:37]   debug: smartctl: Vendor:               HP 
[2021/11/03 07:38:37]   debug: smartctl: Product:              MBXXXXX 
[2021/11/03 07:38:37]   debug: smartctl: Revision:             HPD7 
[2021/11/03 07:38:37]   debug: smartctl: Compliance:           SPC-3 
[2021/11/03 07:38:37]   debug: smartctl: User Capacity:        2,000,398,934,016 bytes [2.00 TB] 
[2021/11/03 07:38:37]   debug: smartctl: Logical block size:   512 bytes 
[2021/11/03 07:38:37]   debug: smartctl: Rotation Rate:        7200 rpm 
[2021/11/03 07:38:37]   debug: smartctl: Form Factor:          3.5 inches 
[2021/11/03 07:38:37]   debug: smartctl: Logical Unit id:      0x5000c50020d352df 
[2021/11/03 07:38:37]   debug: smartctl: Serial number:        XXXXX 
[2021/11/03 07:38:37]   debug: smartctl: Device type:          disk 
[2021/11/03 07:38:37]   debug: smartctl: Transport protocol:   SAS (SPL-3) 
[2021/11/03 07:38:37]   debug: smartctl: Local Time is:        Wed Nov  3 07:38:37 2021 UTC 
[2021/11/03 07:38:37]   debug: smartctl: SMART support is:     Available - device has SMART capability. 
[2021/11/03 07:38:37]   debug: smartctl: SMART support is:     Enabled 
[2021/11/03 07:38:37]   debug: smartctl: Temperature Warning:  Enabled 
[2021/11/03 07:38:37]  notice: Found /dev/sda,  SAS, HP MBXXXXX,   2 TB, S/N=[not populated for some reason]
[2021/11/03 07:38:37]   debug: Readlink: ../devices/pci0000:00/0000:00:07.0/0000:03:00.0/host0/port-0:1/end_device-0:1/target0:0:1/0:0:1:0/block/sdb 
[2021/11/03 07:38:37]   debug: smartctl: smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.6.3] (local build) 
[2021/11/03 07:38:37]   debug: smartctl: Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org 
[2021/11/03 07:38:37]   debug: smartctl: === START OF INFORMATION SECTION === 
[2021/11/03 07:38:37]   debug: smartctl: Vendor:               SEAGATE 
[2021/11/03 07:38:37]   debug: smartctl: Product:              STXXXXX 
[2021/11/03 07:38:37]   debug: smartctl: Revision:             YS0D 
[2021/11/03 07:38:37]   debug: smartctl: Compliance:           SPC-4 
[2021/11/03 07:38:37]   debug: smartctl: User Capacity:        300,000,000,000 bytes [300 GB] 
[2021/11/03 07:38:37]   debug: smartctl: Logical block size:   512 bytes 
[2021/11/03 07:38:37]   debug: smartctl: Rotation Rate:        15000 rpm 
[2021/11/03 07:38:37]   debug: smartctl: Form Factor:          2.5 inches 
[2021/11/03 07:38:37]   debug: smartctl: Logical Unit id:      0x5000c50054b40017 
[2021/11/03 07:38:37]   debug: smartctl: Serial number:        XXXXX 
[2021/11/03 07:38:37]   debug: smartctl: Device type:          disk 
[2021/11/03 07:38:37]   debug: smartctl: Transport protocol:   SAS (SPL-3) 
[2021/11/03 07:38:37]   debug: smartctl: Local Time is:        Wed Nov  3 07:38:37 2021 UTC 
[2021/11/03 07:38:37]   debug: smartctl: SMART support is:     Available - device has SMART capability. 
[2021/11/03 07:38:37]   debug: smartctl: SMART support is:     Enabled 
[2021/11/03 07:38:37]   debug: smartctl: Temperature Warning:  Disabled or Not Supported 
[2021/11/03 07:38:37]  notice: Found /dev/sdb,  SAS, SEAGATE STXXXXX, 300 GB, S/N=[not populated for some reason]
[2021/11/03 07:38:37]   debug: Readlink: ../devices/pci0000:00/0000:00:07.0/0000:03:00.0/host0/port-0:2/end_device-0:2/target0:0:2/0:0:2:0/block/sdc 
[2021/11/03 07:38:37]   debug: smartctl: smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.6.3] (local build) 
[2021/11/03 07:38:37]   debug: smartctl: Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org 
[2021/11/03 07:38:37]   debug: smartctl: === START OF INFORMATION SECTION === 
[2021/11/03 07:38:37]   debug: smartctl: Model Family:     Marvell based SanDisk SSDs 
[2021/11/03 07:38:37]   debug: smartctl: Device Model:     SanDisk SDXXXXX
[2021/11/03 07:38:37]   debug: smartctl: Serial Number:    XXXXX 
[2021/11/03 07:38:37]   debug: smartctl: LU WWN Device Id: 5 001b44 c716a546e 
[2021/11/03 07:38:37]   debug: smartctl: Firmware Version: X31000RL 
[2021/11/03 07:38:37]   debug: smartctl: User Capacity:    240,057,409,536 bytes [240 GB] 
[2021/11/03 07:38:37]   debug: smartctl: Sector Size:      512 bytes logical/physical 
[2021/11/03 07:38:37]   debug: smartctl: Rotation Rate:    Solid State Device 
[2021/11/03 07:38:37]   debug: smartctl: Form Factor:      2.5 inches 
[2021/11/03 07:38:37]   debug: smartctl: Device is:        In smartctl database [for details use: -P show] 
[2021/11/03 07:38:37]   debug: smartctl: ATA Version is:   ACS-2 T13/2015-D revision 3 
[2021/11/03 07:38:37]   debug: smartctl: SATA Version is:  SATA 3.2, 6.0 Gb/s (current: 6.0 Gb/s) 
[2021/11/03 07:38:37]   debug: smartctl: Local Time is:    Wed Nov  3 07:38:37 2021 UTC 
[2021/11/03 07:38:37]   debug: smartctl: SMART support is: Available - device has SMART capability. 
[2021/11/03 07:38:37]   debug: smartctl: SMART support is: Enabled 
[2021/11/03 07:38:37]  notice: Found /dev/sdc,  UNK, SanDisk SDXXXXX, 240 GB, S/N=XXXXXXXXXXXXXXXX
[... cut ...]
[2021/11/03 07:38:37]   debug: smartctl: /dev/sde: Unknown USB bridge [0x090c:0x1000 (0x1100)] 
[2021/11/03 07:38:37]   debug: smartctl: Please specify device type with the -d option. 
[2021/11/03 07:38:37]   debug: smartctl: Use smartctl -h to get a usage summary 
[2021/11/03 07:38:37] warning: /dev/sde USB bridge, no passthru support
[2021/11/03 07:38:37]  notice: Found /dev/sde,  USB, FLASH Drive SM_USB20,   2 GB, S/N=(no ATA pass thru)
[2021/11/03 07:38:37]    info: Automatically enumerated 5 devices.
```

Firminator, 2021-11-03:
Yes exactly to all three.

PartialVolume, 2021-11-03:
With SATA looks like Transport protocol doesn't exist, instead I need to search for "SATA Version is:"

Firminator, 2021-11-03:
Well, should it show SATA in the GUI or ATA (like it shows now). This is quite a conundrum as you could also show the version number of each protocol to give some info about expected throughput (3GBit compared to 6GBit for example). SAS could also show SPL-2 or SPL-3 (https://www.t10.org/drafts.htm#SCSI3_SAS).

Once this is all figured out I can provide you a screenshot for documentation that shows SAS, SATA, ATA, USB on one screen... plus if I find an old IDE drive I might be able to get that drive shown as well. All non-virtualized on real hardware. Ideal for testing.

Carl-Wilhelm, 2021-11-03:
> Well, should it show SATA in the GUI or ATA (like it shows now). This is quite a conundrum as you could also show the version number of each protocol to give some info about expected throughput [...]

I do have Machines with SCSI, IDE and all kinds of SAS. Also have external USB drives. I have one box with debian and compiled nwipe, plus one machine that i boot from usb with shredos. Can do whatever testing you want me to.

PartialVolume, 2021-11-03:
@Carl-Wilhelm That will be useful, thanks.

PartialVolume, 2021-11-03:
@Firminator 

>This is quite a conundrum 

That it is, nothing is ever simple. All the hardware I've come across so far have been consistent in the way they name the /sys/block/.. path, but not these controllers/drivers. Currently most IDE and SATA devices that follow the /sys/block.. naming convention are just end up being called ATA, but with the controller @Firminator  has (sorry, I don't remember if you told me what the make/model was) I can identify SATA or SAS from smartctl but if I followed the previous naming convention, SATA would be called ATA and SAS would be called SCSI. So although technically correct they don't quite describe the drive correctly in terms of what we recognise them as.

I think I might change the terminology previously used, i.e  ATA was displayed for both SATA and IDE assuming smartctl is installed and provides the relevant field I would replace ATA with SATA. If smartmontools is not installed on a system (which can happen on a distro, although maybe most distros install smartmontools as default) nwipe will revert to it's fallback method which is ATA,USB, SCSI and UNK for unknown. If smartctl is available then hopefully the naming will be SATA, SAS, USB and maybe IDE or more correctly PATA. This would be initially a bit of an experiment but hopefully we can find some consistency irrespective of the controller and driver used. In a perfect world I would have liked to retrieve this info via /sys/block or ioctl calls but I think I would be trying to reinvent the wheel when it's so easy to leverage smartmontools to do the work. I hope most of that makes sense. By all means if you don't agree or I've missed something then let me know. It may be a week or so before I can code this as I'm busy for the rest of the week, but hopefully sometime next week I can do another release. And thanks for all you input, it's good to know Shredos and nwipe are useful.

PartialVolume, 2021-11-04:
@Firminator @Carl-Wilhelm https://github.com/PartialVolume/shredos.x86_64/issues/46

Firminator, 2021-11-04:
>LSI SAS2308-IT (FW 18.00.00/7.35.00.00)
>
It was an LSI SAS2308-IT (with FW 18.00.00/7.35.00.00)

We have to figure out what we are trying to show to the user here. We have the protocol, the bus, the interface, the formfactor and the actual physical connectors. I'd argue what we want is to see is what interface the device is using and what the expected (theoretical max) speed of that interface is. That would be beneficial for the user although we have some constraints with GUI space and what can be shown on the main nwipe GUI screen without having to create and invoke a page 2/second screen somewhere.

Then we have naming confusions.
The ATA/IDE name was superseded by PATA (parallel ATA) whenever SATA (Serial ATA) hit the market. So we shouldn't be using ATA/IDE anymore. But that's just the theory. In reality different names for these standards were established and I'd argue everyone of us has heard about (PATA/IDE) UDMA-100 which is an accepted name for ATAPI-6/ATA-6 apparently (accd. to https://en.wikipedia.org/wiki/Parallel_ATA#Speed_of_defined_transfer_modes ).
 
Each interface/bus/standard comes with a certain theor. max bandwidth in MBit/s or GBit/s. I've seen a list somewhere on Wikipedia that shows all this. Here's a list of the ones we would be interested in:
* SCSI-1 (40 MBit/s)
* SCSI-2 (80 MBit/s)
* more SCSI
* Ultra-320 (2560 Mbit/s)
* Ultra-640 (5120 MBit/s)
* USB1.1 (12 MBit/s)
* USB2.0 (480 MBit/s)
* USB3.0 (5 GBit/s)
* USB3.1 (10 GBit/s)
* USB3.2 (20 GBit/s)
* ATA-1
* ATA-2
* ... more transfer rates in **MB**/s @  )
* ATA-5 (UDMA/66) 66 **MB**/s
* ATA-6 (UDMA/100) 100 **MB**/s
* ATA-7 (UDMA/133) 133 **MB**/s
* SATA-1 (1.5 GBit/s)
* SATA-2 (3-GBit/s)
* SATA-3 (6 GBit/s)
* SAS-1 (3 Gbit/s)
* SAS-2 (6 GBits)
* SAS-3 (12 GBit/s)
* SAS-4 (22.5 GBit/s)
* SAS-5 (45 GBit/s)

These naming conventions are fortunate since having 'InterfaceName-Version' gives us some optical consistency in the GUI.

I'd also argue that we want to show MB or GB per second instead of MBit/s and GBit/s for the interfaces as we are wiping disks and pertaining to this we need to know how fast each interface is. No one wants to convert the raw data rate of the interface (GBit/s) into the actual value that we want to compare to. nwipe GUI also uses MB to show the total 'throughput' of the wipe.

Some commands that could be useful to get this info are:
hdparm -I /dev/XXXX | grep -i 'speed'
dmesg | grep -i ATA | grep - i 'link up'
dmesg | grep -i SATA | grep - i 'link up'
dmesg | grep -i SATA | grep - i 'SATA max'
lshw -businfo -C storage -C disk
lshw -short -C storage -C disk
but of course there are intricacies when the device is not SATA (dev/sda) and let's say NVMe (/dev/nvme0n1). NVMe devices would benefit from having nvme-cli installed in ShredOS. I'm sure with megacli or other vendor cli tools you can read out additional info behind a RAID controller in HBA mode as well.

In other words we don't necessarily have to rely on just smartmontools.

Haven't checked if lshw and hdparm is available in ShredOS.

PartialVolume, 2021-11-04:
I'm thinking of a new interface something like this... We would still keep the original (smartmontools might not be installed, so we need a fall back) but maybe by pressing 'I' for info, the default drive selection changes to the screenshot below or maybe you could select this type of display as default from an option.
.
[image]
.

Firminator, 2021-11-05:
Oh my. This is awesome. I'm all for it.

Yes definitely need a fallback. Best would be to derive the data from kernel/systemtools and only use 3rd-party/additional tools as fallback.

What sticks out to me is that it shows SATA 3.0 with a max of 3 GBit/s, but that's wrong. SATA-3 has 6 GBit/s. The drive must be using SATA-2 with a max of 3 GBit/s[**1**] and of course is backward compatible with the older SATA-1 (1.5GBit/s). There might be actually a jumper on that drive where you decide to run it in 1.5 or 3 GBit/s mode :)

If there is GUI space I'd would like to see something like this (using the WD drive in your screenshot)
2. /dev/sdb SATA-2 (3.0 GBit/s) (2 TB) (33 C)

[**1**] https://en.wikipedia.org/wiki/Serial_ATA#SATA_revision_2.0_(3_Gbit/s,_300_MB/s,_Serial_ATA-300)

PartialVolume, 2021-11-05:
> What sticks out to me is that it shows SATA 3.0 with a max of 3 GBit/s, but that's wrong. SATA-3 has 6 GBit/s. The drive must be using SATA-2 with a max of 3 GBit/s[1] and of course is backward [...]

I wouldn't read too much into those numbers, the SATA references are fake, in reality the smartctl data was indeed from the drive mentioned but that drive was attached via a USB 3 port and a very old SATA to USB adapter on a RPI4. running nwipe. I guess that old SATA/USB adapter may have something to do with it? I should hook the drive up to an actual SATA connector on my Dell test system and see whether the smartctl reports differently.

PartialVolume, 2021-11-05:
>If there is GUI space I'd would like to see something like this (using the WD drive in your screenshot)
2. /dev/sdb SATA-2 (3.0 GBit/s) (2 TB) (33 C)

Looks good, I like it. It's a little longer than I'd like as I need to have some leading spaces on the disc capacity to keep the columns aligned for varying sizes in the list. However I don't think we need the 1., 2., 3 numbering, it's not like we reference that anywhere. Getting rid of the numbering would free up 3 characters so ” /dev/sdb SATA-2 (3.0 GBit/s) (2 TB) (33 C)" should fit in nicely. And if not I can always move the drive info over a character or two.

PartialVolume, 2021-11-05:
Now I've counted the characters I think " /dev/sdb SATA-2 (3.0 GBit/s) (2 TB) (33 C)" is 19 characters too long for a 80 character column, but fine for 100+ columns. I would probably dynamically check column width and if it falls below 100 columns  move the "3.0 GBit/s” and "(33 C)" into the info pane. What do you think?

Firminator, 2021-11-05:
> Now I've counted the characters I think " /dev/sdb SATA-2 (3.0 GBit/s) (2 TB) (33 C)" is 19 characters too long for a 80 character column, but fine for 100+ columns. I would probably dynamically [...]

Sounds perfect.... although I have no understanding how it is determined if the layout is just 80 characters (which is what my test box uses) and when you get a 100+ character layout. Does that have something to do with the monitor resolution or is that a video driver thing?

Firminator, 2021-11-05:
> What sticks out to me is that it shows SATA 3.0 with a max of 3 GBit/s, but that's wrong. SATA-3 has 6 GBit/s. The drive must be using SATA-2 with a max of 3 GBit/s[1] and of course is backward [...]
 
> I wouldn't read too much into those numbers, the SATA references are fake, in reality the smartctl data was indeed from the drive mentioned but that drive was attached via a USB 3 port and a very [...]

Well I think your example setup couldn't have been better. It very well has proven why you would want to show the interface max speed... it will put attention to a non-optimized setup using for example slow/bad/incompatible interface adapters like the old one you were using limiting throughput to 1.5GBit/s. For example I have an adapter that will connect 40/80pin IDE drives to SATA ports and it works for wiping, but I'm not sure if I reach full wiping speed due to faulty/buggy/imperfect interface translation from PATA to SATA. I could buy a second brand and then compare :)

PartialVolume, 2021-11-05:
>Sounds perfect.... although I have no understanding how it is determined if the layout is just 80 characters (which is what my test box uses) and when you get a 100+ character layout. Does that have something to do with the monitor resolution or is that a video driver thing?

From what I've found on the different systems I have.

If you boot on a legacy bios (no UEFI) Shredos will display nwipe in 80 column mode using the frame buffer (no Nvidia/Intel video drivers are used).

If you boot UEFI Shredos will boot at full screen resolution assuming Shredos has a video driver that supports your graphics card, which for me, makes the text too small but even so, it's still legible. I started loading some graphics drivers into Shredos but  realised real quick that the boot image was growing very fast so had second thoughts about that. Adding drivers for graphics cards was a rabbit hole I didn't want to go down until after all the wiping related features have been added to nwipe.

So on my DELL optiplex 9010 it will boot high resolution on either an Nvidia (don't remember the model) or it's onboard graphics, but if I switch to legacy boot in the DELL configuration it runs Shredos in 80 column mode. Personally I like the 80 column mode as I can read the text from across the room, but others prefer the high resolution mode. Each to their own.

PartialVolume, 2021-11-05:
>. it will put attention to a non-optimized setup using for example slow/bad/incompatible interface adapters like the old one you were using limiting throughput to 1.5GBit/s

The old SATA/USB adapter did indeed limit throughtput. Here's a snapshot of smartctl -i on the same model of disk attached to the SATA connectors on the Dell motherboard. Doubles current speed from 1.5 Gb/s to 3.0 Gb/s

[image]

PartialVolume, 2021-11-05:
However, in reality the actual sustained wipe speed (zero fill, no blanking, no verify) is about 120-140MB/s (0.96-1.12 Gb/s) so the old SATA/USB adapter might be an issue for small amounts of data being read/written to disk but when it comes to a sustained write to disk where the limiting factor is how fast we can write to the platter rather than the disc cache, then the old 1.5Gb/s adapter possibly has little affect for our particular use.

PartialVolume, 2021-11-06:
@Firminator For the time being I've added bus type as ATA for a SATA (currently UNK) that is plugged into your LSI controller. We'll revisit this once the selection GUI layout has been changed, but for the time being I'll leave this as USB, SAS, ATA, NVME and UNK as the common names. Committed upstream in nwipe [Check smartctl for unresolved bus types SATA #358](https://github.com/martijnvanbrummelen/nwipe/pull/358)