PartialVolume/shredos.x86_64 #44: ISO  - v2020.05.016_x86-64_0.32

CoodyCZ, 2021-11-01:
Hi! In version 0.31 you have released legacy bios ISO (it was working with UEFI too)! Will you release a new ISO?

PartialVolume, 2021-11-01:
@CoodyCZ  Yes, hopefully I should be compiling the .iso tomorrow and testing it. If all good, I'll do a new .iso release tomorrow night.

Thanks for letting me know the previous legacy version also worked with UEFI, I'll need to update the README.md file.

CoodyCZ, 2021-11-02:
@PartialVolume thanks, i'm waiting for it, i'l be testing!

PartialVolume, 2021-11-02:
See assets section for .iso in [latest release](https://github.com/PartialVolume/shredos.x86_64/releases/latest)

Firminator, 2021-11-02:
works.

PartialVolume, 2021-11-02:
@Firminator That's good, works in legacy bios mode on my Dell box, but doesn't appear as a UEFI device.

I'm not sure if that's an issue with the Dell UEFI config or something missing from ShredOS. The .iso doesn't have the EFI directory, like the USB .img, that may be the issue, or maybe an ISO does things differently with UEFI. The ISO does contain /boot/grub-eltorito.img

CoodyCZ, 2021-11-03:
@PartialVolume 
Legacy BIOS - works (tested on desktop, notebook, virtualbox)

UEFI - working too, but only with VENTOY 

RUFUS - shows error -  not bootable medium, or is used compression that is not supported in Rufus

PartialVolume, 2021-11-03:
@CoodyCZ Thanks, Currently I'm having to build separate versions of ShredOS for USB and .iso. But to speed releases up I'm wondering whether there is a tool out there that will take the 64bit legacy/UEFI .img file and produce a legacy/UEFI . iso?

Firminator, 2021-11-03:
It might be worth a try to contact Pete from Rufus as he's the guru regarding USB and ISO. I'm pretty sure he can tell you in a minute what has to be done to produce legit BIOS and UEFI .IMG and .ISO

PartialVolume, 2021-11-03:
I guess this might be one way, although from my brief look at it, I don't see any mention of whether UEFI is supported. [https://ostechnix.com/create-iso-bootable-usb-drive-linux/](https://ostechnix.com/create-iso-bootable-usb-drive-linux/)

Firminator, 2021-11-03:
I think you might be right since that sounds exactly what [Passmark's Memtest](https://www.memtest86.com/download.htm) is doing for a number of years. They don't provide ISOs anymore and only provide the USB flash drive IMG file which only supports UEFI. If you have a BIOS box you are stuck with the older v4 of Memtest.

CoodyCZ, 2021-11-03:
I was looking for a way, and the only thing I found was this -
`https://stackoverflow.com/questions/43381612/buildroot-building-iso-image`

BUT... no mention of UEFI anywhere

CoodyCZ, 2021-11-04:
@PartialVolume i have read manual for BuildRoot... there is a point 

14.1.2. Live CD
To build a live CD image, enable the iso image option in the Filesystem images menu. Note that this option is only available on the x86 and x86-64 architectures, and if you are building your kernel with Buildroot.

You can build a live CD image with either IsoLinux, Grub or Grub 2 as a bootloader, **but only Isolinux supports making this image usable both as a live CD and live USB (through the Build hybrid image option).**

PartialVolume, 2021-11-04:
Thanks, that's the way I currently build the .iso but without isolinux, using grub2, I had issues with it before when I tried isolinux, I might have to revisit it. Unfortunately I don't think it mentions anything about UEFI. To achieve UEFI in the USB .IMG we do it bit of custom stuff with the .efi file, maybe I'll need to do the same with .iso, the .efi code isn't ending up in the current .iso in /EFI maybe it's as simple as achieving that and the iso will boot with grub on UEFI.

CoodyCZ, 2021-11-04:
@PartialVolume  good luck in development ... i know you can't satisfy everyone ... thanks so much for any ISO. 
As I wrote above, I use VENTOY, there ISO loads on both Legacy and UEFI. Shredos is a fantastic tool!

PartialVolume, 2026-02-15:
@sorenstoutner no problem, the .iso's in the latest release should pretty much boot on every x86-64 or i686 system and the full compliment of .isos and .imgs will always be provided for every release due to the new 'build_shredos.sh all' script. So no more missing .isos 😊