PartialVolume/shredos.x86_64 #45: Update SmartMonTools (smartctl) from 7.1 to 7.2

Firminator, 2021-11-03:
Not a bug, but a suggestion. Not sure how complicated to implement.

ShredOS comes with smartctl v7.1 2019-12-30 r5022
A newer release of smartctl v7.2 2020-12-30 r5155 is out with a number of bugfixes and also an updated drive database: https://www.smartmontools.org/browser/tags/RELEASE_7_2/smartmontools/NEWS

PartialVolume, 2021-11-03:
Some time soon I'll need to upgrade the buildroot I'm currently using from 2020.05 to 2021.xx or even 2022.xx, I expect that will include the latest smartmontools.

Firminator, 2021-11-26:
I checked and the 2021-08-02 buildroot build from https://github.com/PartialVolume/shredos.x86_64/discussions/59#discussioncomment-1705166 has smartmontool v7.2 (2020-12-30) included. This issue can be closed.

Firminator, 2021-12-08:
Closing this... especially now that the newer ShredOS has been released (it also comes with smartmontools v7.2).

closed 2021-12-08 by Firminator