PartialVolume/shredos.x86_64 #46: Request for smartctl -i info

PartialVolume, 2021-11-04:
If anybody is seeing UNK in the bus type field as opposed to USB or ATA in the GUI, can you post the output from either smartctl -i /dev/... Or nwipe -v

Also if you have an IDE or SCSI drive, can you also post the smartctl -i /dev/... Or nwipe -v output. And tell me what's in the bus type field in the GUI.

Thanks

Firminator, 2021-11-05:
Just some notes for others who might need a workflow how to get this file off an non-networked/air-gapped box
* ALT+F2
* `mkdir /media/usbdrive`
* attach flash drive, then `lsblk` to find identifier
* `mount /def/sdX /media/usbdrive` where X is the identifier from the previous step
* `nwipe -v --logfile=/media/usbdrive/shredos-debug.log`
* CTRL+C to end NWIPE
* `umount /media/usbdrive`

```
Connected to LSI SAS controller:
[2021/11/05 06:14:13]   debug: Readlink: ../devices/pci0000:00/0000:00:07.0/0000:03:00.0/host0/port-0:0/end_device-0:0/target0:0:0/0:0:0:0/block/sda 
[2021/11/05 06:14:13]   debug: smartctl: smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.6.3] (local build) 
[2021/11/05 06:14:13]   debug: smartctl: Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org 
[2021/11/05 06:14:13]   debug: smartctl: === START OF INFORMATION SECTION === 
[2021/11/05 06:14:13]   debug: smartctl: Vendor:               HP 
[2021/11/05 06:14:13]   debug: smartctl: Product:              MBxxx 
[2021/11/05 06:14:13]   debug: smartctl: Revision:             HPD7 
[2021/11/05 06:14:13]   debug: smartctl: Compliance:           SPC-3 
[2021/11/05 06:14:13]   debug: smartctl: User Capacity:        2,000,398,934,016 bytes [2.00 TB] 
[2021/11/05 06:14:13]   debug: smartctl: Logical block size:   512 bytes 
[2021/11/05 06:14:13]   debug: smartctl: Rotation Rate:        7200 rpm 
[2021/11/05 06:14:13]   debug: smartctl: Form Factor:          3.5 inches 
[2021/11/05 06:14:13]   debug: smartctl: Logical Unit id:      xxxxxxxxxxxxxxxxxxxxxxxx
[2021/11/05 06:14:13]   debug: smartctl: Serial number:        xxxxxxxxxxxxxxxxxxxxxxxxxx 
[2021/11/05 06:14:13]   debug: smartctl: Device type:          disk 
[2021/11/05 06:14:13]   debug: smartctl: Transport protocol:   SAS (SPL-3) 
[2021/11/05 06:14:13]   debug: smartctl: Local Time is:        Fri Nov  5 06:14:13 2021 UTC 
[2021/11/05 06:14:13]   debug: smartctl: SMART support is:     Available - device has SMART capability. 
[2021/11/05 06:14:13]   debug: smartctl: SMART support is:     Enabled 
[2021/11/05 06:14:13]   debug: smartctl: Temperature Warning:  Enabled 
[2021/11/05 06:14:13]  notice: Found /dev/sda,  SAS, HP MBxxxx,   2 TB, S/N=

Connected to LSI SAS controller:
[2021/11/05 06:14:13]   debug: Readlink: ../devices/pci0000:00/0000:00:07.0/0000:03:00.0/host0/port-0:1/end_device-0:1/target0:0:1/0:0:1:0/block/sdb 
[2021/11/05 06:14:13]   debug: smartctl: smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.6.3] (local build) 
[2021/11/05 06:14:13]   debug: smartctl: Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org 
[2021/11/05 06:14:13]   debug: smartctl: === START OF INFORMATION SECTION === 
[2021/11/05 06:14:13]   debug: smartctl: Vendor:               SEAGATE 
[2021/11/05 06:14:13]   debug: smartctl: Product:              STxxxxxxxxxx
[2021/11/05 06:14:13]   debug: smartctl: Revision:             YS0D 
[2021/11/05 06:14:13]   debug: smartctl: Compliance:           SPC-4 
[2021/11/05 06:14:13]   debug: smartctl: User Capacity:        300,000,000,000 bytes [300 GB] 
[2021/11/05 06:14:13]   debug: smartctl: Logical block size:   512 bytes 
[2021/11/05 06:14:13]   debug: smartctl: Rotation Rate:        15000 rpm 
[2021/11/05 06:14:13]   debug: smartctl: Form Factor:          2.5 inches 
[2021/11/05 06:14:13]   debug: smartctl: Logical Unit id:      xxxxxxxxxxxxxxxxxxxxxxx
[2021/11/05 06:14:13]   debug: smartctl: Serial number:       xxxxxxxxxxxxx
[2021/11/05 06:14:13]   debug: smartctl: Device type:          disk 
[2021/11/05 06:14:13]   debug: smartctl: Transport protocol:   SAS (SPL-3) 
[2021/11/05 06:14:13]   debug: smartctl: Local Time is:        Fri Nov  5 06:14:13 2021 UTC 
[2021/11/05 06:14:13]   debug: smartctl: SMART support is:     Available - device has SMART capability. 
[2021/11/05 06:14:13]   debug: smartctl: SMART support is:     Enabled 
[2021/11/05 06:14:13]   debug: smartctl: Temperature Warning:  Disabled or Not Supported 
[2021/11/05 06:14:13]  notice: Found /dev/sdb,  SAS, SEAGATE STxxxxxxx, 300 GB, S/N=

Connected to LSI SAS controller.
[2021/11/05 06:14:13]   debug: Readlink: ../devices/pci0000:00/0000:00:07.0/0000:03:00.0/host0/port-0:4/end_device-0:4/target0:0:4/0:0:4:0/block/sdc 
[2021/11/05 06:14:13]   debug: smartctl: smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.6.3] (local build) 
[2021/11/05 06:14:13]   debug: smartctl: Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org 
[2021/11/05 06:14:13]   debug: smartctl: === START OF INFORMATION SECTION === 
[2021/11/05 06:14:13]   debug: smartctl: Device Model:     INTEL SSDxxxxxxxxxxx 
[2021/11/05 06:14:13]   debug: smartctl: Serial Number:    xxxxxxxxxxx 
[2021/11/05 06:14:13]   debug: smartctl: LU WWN Device Id: xxxxxxxxxxxxxxxxxxxxxx
[2021/11/05 06:14:13]   debug: smartctl: Firmware Version: 4PC10365 
[2021/11/05 06:14:13]   debug: smartctl: User Capacity:    160,041,885,696 bytes [160 GB] 
[2021/11/05 06:14:13]   debug: smartctl: Sector Size:      512 bytes logical/physical 
[2021/11/05 06:14:13]   debug: smartctl: Rotation Rate:    Solid State Device 
[2021/11/05 06:14:13]   debug: smartctl: Device is:        Not in smartctl database [for details use: -P showall] 
[2021/11/05 06:14:13]   debug: smartctl: ATA Version is:   ATA8-ACS T13/1699-D revision 4 
[... 55 lines cut ...]
[2021/11/05 06:14:13]   debug: smartctl: Serial Number:    xxxxxxxxxxxxxxxxxxx 
[... cut ...]
[2021/11/05 06:14:13]   debug: smartctl: /dev/sdg: Unknown USB bridge [0x090c:0x1000 (0x1100)] 
[2021/11/05 06:14:13]   debug: smartctl: Please specify device type with the -d option. 
[2021/11/05 06:14:13]   debug: smartctl: Use smartctl -h to get a usage summary 
[2021/11/05 06:14:13] warning: /dev/sdg USB bridge, no passthru support
[2021/11/05 06:14:13]  notice: Found /dev/sdg,  USB, FLASH Drive SM_USB20,   2 GB, S/N=(no ATA pass thru)
[2021/11/05 06:14:13]    info: Automatically enumerated 7 devices.
```

Firminator, 2021-11-08:
IDE drive (connected to PCI IDE adapter via super-short 40-pin ribbon cable; PITA to connect; a co-worker of mine found it in his stash hidden in an USB enclosure):

>[2021/11/08 13:57:21]   debug: Readlink: ../devices/pci0000:00/0000:00:1e.0/0000:06:05.0/**ata9**/host9/target9:0:0/9:0:0:0/block/sdg 
[2021/11/08 13:57:21]   debug: smartctl: smartctl 7.1 2019-12-30 r5022 [x86_64-linux-5.6.3] (local build) 
[2021/11/08 13:57:21]   debug: smartctl: Copyright (C) 2002-19, Bruce Allen, Christian Franke, www.smartmontools.org 
[2021/11/08 13:57:21]   debug: smartctl: === START OF INFORMATION SECTION === 
[2021/11/08 13:57:21]   debug: smartctl: Model Family:     Western Digital Caviar 
[2021/11/08 13:57:21]   debug: smartctl: Device Model:     WDC WDxxxxxxx
[2021/11/08 13:57:21]   debug: smartctl: Serial Number:    WD-xxxxxxxxxxxxx 
[2021/11/08 13:57:21]   debug: smartctl: Firmware Version: 05.01C05 
[2021/11/08 13:57:21]   debug: smartctl: User Capacity:    80,026,361,856 bytes [80.0 GB] 
[2021/11/08 13:57:21]   debug: smartctl: Sector Size:      512 bytes logical/physical 
[2021/11/08 13:57:21]   debug: smartctl: Device is:        In smartctl database [for details use: -P show] 
[2021/11/08 13:57:21]   debug: smartctl: **ATA Version is:   ATA/ATAPI-6** (minor revision not indicated) 
[2021/11/08 13:57:21]   debug: smartctl: Local Time is:    Mon Nov  8 13:57:21 2021 UTC 
[2021/11/08 13:57:21]   debug: smartctl: SMART support is: Available - device has SMART capability. 
[2021/11/08 13:57:21]   debug: smartctl: SMART support is: Enabled 
[2021/11/08 13:57:21]  notice: Found /dev/sdg,  ATA, WDC WDxxxxxxxxxxx,  80 GB, S/N=WD-xxxxxxxxxxxxx

This confirms the reference table from https://github.com/PartialVolume/shredos.x86_64/issues/39#issuecomment-961244517:

- SCSI-1 (40 MBit/s)
- SCSI-2 (80 MBit/s)
- ... more SCSI
- Ultra-320 (2560 Mbit/s)
- Ultra-640 (5120 MBit/s)
- USB1.1 (12 MBit/s)
- USB2.0 (480 MBit/s)
- USB3.0 (5 GBit/s)
- USB3.1 (10 GBit/s)
- USB3.2 (20 GBit/s)
- ATA-1
- ATA-2
- ... more ATA (i.e. ATA-3, ATA-4)
- ATA-5 (UDMA/66) 66 MB/s
- **ATA-6 (UDMA/100) 100 MB/s**
- ATA-7 (UDMA/133) 133 MB/s
- SATA-1 (1.5 GBit/s)
- SATA-2 (3 GBit/s)
- SATA-3 (6 GBit/s)
- SAS-1 (3 Gbit/s)
- SAS-2 (6 GBits)
- SAS-3 (12 GBit/s)
- SAS-4 (22.5 GBit/s)
- SAS-5 (45 GBit/s)

Checked further and found double confirmation in `dmesg`. Here's the relevant output:
>[    2.340545] ata9.00: **ATA-6**: WDC WDxxxxxxxxxx, 05.01C05, max **UDMA/100**
[    2.340548] ata9.00: 156301488 sectors, multi 0: LBA 
[    2.340553] ata9.00: **limited to UDMA/33 due to 40-wire cable**
[    2.343856] scsi 9:0:0:0: Direct-Access     ATA      WDC WDxxxxxxxxx 1C05 PQ: 0 ANSI: 5
[    2.343989] scsi 9:0:0:0: Attached scsi generic sg6 type 0
[    2.344096] sd 9:0:0:0: [sdg] 156301488 512-byte logical blocks: (80.0 GB/74.5 GiB)
[    2.344105] sd 9:0:0:0: [sdg] Write Protect is off
[    2.344106] sd 9:0:0:0: [sdg] Mode Sense: 00 3a 00 00
[    2.344119] sd 9:0:0:0: [sdg] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    2.353691]  sdg: sdg1
[    2.354267] sd 9:0:0:0: [sdg] Attached SCSI disk
[    2.370031] scsi host10: **pata_legacy** [aka IDE]
[    2.370106] ata11: PATA max PIO4 cmd 0x170 ctl 0x376 irq 15

recap:
ATA/ATAPI-6 [smartctl -i]
equals
ATA-6 (UDMA/100) [dmesg]
equals
ATA-6 (Ultra ATA/100) [[Wikipedia](https://en.wikipedia.org/wiki/Parallel_ATA#Features_introduced_with_each_ATA_revision)]