PartialVolume/shredos.x86_64 #47: New package request: nvme-cli

Firminator, 2021-11-04:
This would be used in the future if NVMe Secure Erase can be implemented into SHredOS (as already suggested @ https://github.com/PartialVolume/shredos.x86_64/issues/20#issuecomment-778616843 ).

For now it could be used to implement a new low-footprint feature showing the temperature of NVMe drives with:
nvme smart-log /dev/nvme0 | grep -i '^temperature'

PartialVolume, 2021-11-06:
nvme-cli isn't provided in buildroot, however, that's not a problem.

I'll create a custom package and pull in the latest release from [https://github.com/spdk/nvme-cli](https://github.com/spdk/nvme-cli)

PartialVolume, 2021-11-06:
Correction to the above github link, this is the main branch for nvme-cli [https://github.com/linux-nvme/nvme-cli](https://github.com/linux-nvme/nvme-cli)

Firminator, 2021-11-26:
I think this can be closed as I found the nvme-cli package already selected in the 2020-05 ShredOS buildroot under Target packages -> Hardware handling -> nvme

>NVME utility, a utility for interacting with standard NVM
Express (optimized PCI Express SSD interface) devices.
>
>https://github.com/linux-nvme/nvme-cli

PartialVolume, 2021-11-26:
Yes, I'm pretty sure I also included the nvme package in the 2021.08 version of ShredOS that you are testing. As it s a newer buildroot I also think smartctl is also a newer version 7.2?

closed 2021-11-26 by PartialVolume

Firminator, 2021-11-26:
Yes, I checked and the 2021-08-02 buildroot build from #59 (comment) has smartmontool v7.2 (2020-12-30) included.
Also nvme-cli v1.12 is included. This issue can also be closed :)