PartialVolume/shredos.x86_64 #52: Boot hanging/failing to launch the OS

Ryan-A-Bradley, 2021-11-17:
shredOS hangs at the black launch screen. Is the laptop's hardware too new?
example:

RecalBox
_

I've tried booting with USB; the 32-bit and 64-bit .img files both hang at recal with _.
I've tried using an external cd/dvd writer/reader with the 64-bit .iso and disc spins up but fails to boot to the black screen.

ACER Predator Helios 300 15.6" Gaming Laptop - Intel® Core™ i7, RTX 3070, 1 TB SSD

Processor | - Intel® Core™ i7-11800H Processor- Octa-core- 2.4 GHz / 4.6 GHz- 24 MB cache
RAM - 16 GB DDR4 (2133 MHz)- 32 GB maximum installable RAM
GPU - NVIDIA GeForce RTX 3070- 8 GB GDDR6
Storage - 1 TB SSD

USB Type-C x 1- USB 3.1 x 3

A link to the laptop where you can access all specifications.
https://www.currys.co.uk/gbuk/gaming/pc-gaming/gaming-laptops/acer-predator-triton-300-15-6-gaming-laptop-intel-core-i7-rtx-3070-1-tb-ssd-10224271-pdt.html

Firminator, 2021-11-17:
Is there an option in the BIOS/UEFI setup to change from UEFI mode to Legacy/BIOS mode?

Ryan-A-Bradley, 2021-11-17:
shredOS works on my other laptop and on my desktop pc (both have older hardware than the new laptop). I was hoping to use shredOS to wipe all 3 machines. :o

I just don't have much knowledge about how hardware and bootable software needs to be programmed to work.

Laptop is UEFI boot by default (tested boot with secure boot on/off). For some reason, I'm unable to switch it from UEFI so I'm under the impression this BIOS doesn't support legacy? I'll look into this now.

Ryan-A-Bradley, 2021-11-17:
Apparently, I should be able to change the boot setting to legacy after setting a password (which I tried). However, I'll try a bit more right now as I have the laptop currently with me on the desk.

Firminator, 2021-11-17:
Sure. Give it a try and let us know.

PartialVolume, 2021-11-17:
If this document applies to your model you may have to enable a setting that trusts the USB device to boot UEFI, not only that but the actual file name.
[image]

Ryan-A-Bradley, 2021-11-17:
Also, this is the BIOS except mine says v2.02. I got this image from google
[image]
.

PartialVolume, 2021-11-17:
Acer A315-53G Service Manual [https://www.manualslib.com/manual/1780777/Acer-A315-53g.html?page=55#manual](https://www.manualslib.com/manual/1780777/Acer-A315-53g.html?page=55#manual)

PartialVolume, 2021-11-17:
Possible under 'advanced settings'

Ryan-A-Bradley, 2021-11-17:
It says, on this BIOS system you can press CTRL+S on the tab "Main" in advanced settings to access special settings regarding SATA mode (which is apparently needed to switch it to legacy). I've changed it from RAID to AHCI and still unable to change the boot mode to Legacy.

Ryan-A-Bradley, 2021-11-17:
https://community.acer.com/en/discussion/615614/no-sata-mode-in-my-bios-predator-helios-300-ph315-53

tried this for help but doesn't seem to let me switch the boot mode. :/

PartialVolume, 2021-11-17:
> If this document applies to your model you may have to enable a setting that trusts the USB device to boot UEFI, not only that but the actual file name. [image]

Any luck finding this setting that lets you set a trusted UEFI device?

Ryan-A-Bradley, 2021-11-17:
I've added the .efi to the trusted files for execution and it does the same thing. It boots to a black screen and then shows:

RecalBox
_

Ryan-A-Bradley, 2021-11-17:
EFI file boot 0: shredos

it's now displaying the file in the boot priority. i've named it "shredos". set shredos to 1st slot in priority order.

selected f12 to choose from boot order and attempted booting shredos.

loads a black screen and then shows:

RecalBox
_

PartialVolume, 2021-11-17:
Has it got two discs in raid configuration or two separate device in non raid mode. Nwipe in shredos will only work in non raid mode.

It's also possible shredos doesn't have the latest driver, however if you loaded ubuntu 18.10 on it and it worked then shredos should work too.

Ryan-A-Bradley, 2021-11-17:
I'm able to switch it between the SATA modes:

ACHI
Optane without RAID
Optane with RAID

Same issue happens on all settings. And I'm unable to switch it from UEFI to Legacy no matter what SATA mode I choose.

I haven't tried installing Ubuntu 18.10 or later so I will download Ubunutu now and format it onto a USB to see if it works.

Note: This laptop was bought with Windows preinstalled on it and has the product key on the hardware (It wasn't manufactured with Linux on it and has never been tested with Linux on it). I'll begin the process now.

Ryan-A-Bradley, 2021-11-17:
Currently downloading https://old-releases.ubuntu.com/releases/18.10/

Ryan-A-Bradley, 2021-11-17:
Just a quick example of what the hang looks like when I boot. I made it in Microsoft Paint while waiting for Ubuntu to download.
[image]

Ryan-A-Bradley, 2021-11-17:
Update regarding Ubuntu 18.10: Currently using Rufus to format Ubuntu onto a USB.

Firminator, 2021-11-17:
This is basically a known issue. See https://github.com/PartialVolume/shredos.x86_64/issues/34 and https://github.com/PartialVolume/shredos.x86_64/issues/48#issuecomment-969185245

So far it's happening only on very new devices unless more reports are coming in from other people reporting the opposite.

Ryan-A-Bradley, 2021-11-17:
Ubuntu 18.10 successfully installed.

I will check the comment links now, Firminator. Thank you to both of you for helping me try to solve this issue and then further explaining that it's likely due to it being a very new device.

PartialVolume, 2021-11-17:
I've noticed that shredos is lacking the /device/fb0 ( the frame buffer ). That used to work as I had used fbgrab etc to grap screenshots in the past. I've got a feeling the frame buffer may have got disabled when I started loading video drivers. Before I build tonights shredos I'll take a look at enabling the frame buffer. That would certainly cause this problem.

Firminator, 2021-11-17:
Ryan,
Do you have an NVMe drive installed in that Acer laptop?

Ryan-A-Bradley, 2021-11-17:
Just wiped Ubuntu off it.

I'm also not sure what an NVMe drive is. If NVMe means "do you have an SSD in your laptop?", I have a 1TB SSD installed on the laptop.

Ryan-A-Bradley, 2021-11-17:
By the way, the main post of this thread is like half wrong because I accidentally posted specs for a different laptop.

My laptop is the ACER Predator Helios 300 15.6" Gaming Laptop - Intel® Core™ i7, RTX 3070, 1 TB SSD.

I accidentally posted Acer Predator Triton, which isn't the same laptop. I've edited the main post now so it's correct.

PartialVolume, 2021-11-17:
Can you try this version of shredos. It's a version prior to me adding the video drivers and so the frame buffer may be working
[https://github.com/PartialVolume/shredos.x86_64/releases/download/v2020.05.009_x86-64_0.30.001/shredos-2020.05.009_x86-64_0.30.001_20210217.img](https://github.com/PartialVolume/shredos.x86_64/releases/download/v2020.05.009_x86-64_0.30.001/shredos-2020.05.009_x86-64_0.30.001_20210217.img)

Ryan-A-Bradley, 2021-11-17:
I will try it now. :D

Ryan-A-Bradley, 2021-11-17:
Crashed again at these screen in the same way.

RecalBox
_
[image]

Firminator, 2021-11-17:
Yes same here.
What's your CPU and GPU?

PartialVolume, 2021-11-17:
I just checked, the frame buffer isn't available in that version either. I'll go back further and check an earlier release.

Firminator, 2021-11-17:
Mine is 11th gen Intel Core aka Tigerlake and it comes with this GPU:
https://en.wikipedia.org/wiki/Tiger_Lake#GPU
I don't have a dedicated GPU in this box.

Ryan-A-Bradley, 2021-11-17:
CPU: Intel® Core™ i7
GPU: RTX 3070 notebook (since it's a laptop it uses notebook version of drivers).

I'm not sure what the exact specification of the CPU is other than that. I'll look around for some documents in my drawer regarding the laptop. 1 moment.

Firminator, 2021-11-17:
It should show on the upper left main screen of your BIOS setup program :)

Ryan-A-Bradley, 2021-11-17:
oh yeah, ahaha. I was about to look it all up on acer using serial number. xD i told you i'm newby.

Ryan-A-Bradley, 2021-11-17:
Intel core i7-10750H cpu

Ryan-A-Bradley, 2021-11-17:
here you go, you guys! i looked up the SN on Acer's website.

https://snlookup.com/acer-predator-ph315-53-notebook-nh-qatek-001-p382646?sn=NHQATEK001120026843400#ffs-tabbed-17|tab6

this contains the specifications of my laptop!

Firminator, 2021-11-17:
So 10th and 11th gen CPUs and their GPU are affected
* i7-10750H
* i7-11800H

Ryan-A-Bradley, 2021-11-17:
I'll update the description a bit more in the main post so other people who have this problem can see easier if this thread is the same issue they're having.

PartialVolume, 2021-11-17:
@Ryan-A-Bradley I'm just compiling a new shredos that contains 30 or more drivers for frame buffer support. Hopefully that will fix this problem. ( @Firminator It will also have the latest nwipe v0.32.012 ). I'll release it sometime tomorrow.

Ryan-A-Bradley, 2021-11-17:
looking forward to seeing if it fixes it! i don't use many non-microsoft tools except for the ones i really need and shredOS is an absolute must-have in the toolbelt so I'm thankful to all who are involved in the development of this piece of kit.

there's nothing more reassuring than shredOS for completely sanitizing storage devices. it cleans the storage and cures the OCD. ahahhahaa.

PartialVolume, 2021-11-18:
@Ryan-A-Bradley Can you try this version, I've enabled frame buffer support so fingers crossed should now work. Let me know if it doesn't. Thanks.

[v2020.05.018_x86-64_0.32.014](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2020.05.018_x86-64_0.32.014)

Ryan-A-Bradley, 2021-11-18:
I'll get right on it now!

Ryan-A-Bradley, 2021-11-18:
Tested.

It works but the graphical interface is shrunk to the top of the screen. I hit ctrl+c which proved to me that it is working! when i hit ctrl+c, the blue bar at the top with tiny white text changed to the black screen and i could see it loaded the usual text on that screen too.

i can't take a picture but i can show another example made in paint so you know what's happening!

[image]

Ryan-A-Bradley, 2021-11-18:
i would imagine if i was to hit shift+S it would begin wiping using the default settings as usual:

DoD short
1 cycle
1 blanking pass after main wipe

I also just tried pressing enter on the black screen and it did, in fact, launch the wiping interface again. it's working but currently has that visual issue.

Ryan-A-Bradley, 2021-11-18:
a more accurate example of what it looks like:

[image]

PartialVolume, 2021-11-18:
Thanks, if nothing else we at least know it's a graphics problem. I'll look up the specs on that system and see whether I can load a new driver for it.

If you ALT-F2 or ALT-F3 do you get the same thing? You may have to type some random text. ALT-F1 takes you back to the main screen.

Ryan-A-Bradley, 2021-11-18:
so, hard to see what's going on with alt-f2 but here's what happens:

blinking white dots at the very top of the screen, which means there's something loaded and working but unfortunately unable to display it.

as for alt-f3, not white dots but text still shrunk into a bar at the very top of the screen.

Ryan-A-Bradley, 2021-11-18:
here's an example of what alt-f3 looks like:

[image]

this is also what it looks like when pressing ctrl-c to switch from the main interface back to the launch screen.

PartialVolume, 2021-11-18:
I'm wondering whether editing the resolution may fix the problem. What's the screen aspect ratio ? 16:9 or 16:10 ?

See the info here on editing grub.cfg on how to change the terminal resolution. [how-to-make-a-persistent-change-to-the-terminal-resolution](https://github.com/PartialVolume/shredos.x86_64#how-to-make-a-persistent-change-to-the-terminal-resolution)

Maybe change it to 1280x800 or 1152×648

Ryan-A-Bradley, 2021-11-18:
actually, from looking deeper at alt-f3 i can see the blinking cursor from within the very tiny unreadable text.

Ryan-A-Bradley, 2021-11-18:
i should also note:

the shrunken text seems to be copied in sections across the top of the screen. hard to explain but perhaps this can help.

text | text | text | text | text |

each 'text' represents the entire interface so try to imagine it loading the interface text multiple times next to each other in a shrunken format across the top of the screen.

i will try changing resolution now. :)

Ryan-A-Bradley, 2021-11-18:
aspect resolution is 16:9 by the way for this laptop.

Ryan-A-Bradley, 2021-11-18:
regarding changing resolution in grub.cgf, changed it to:

set default="0"
set timeout="0"
gfxpayload=1920x1080x16
menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3
}

i have done this for both boot and EFI grub.cfg.

problem hasn't fixed, still same issue with the shrunk text inside a bar across the top of the screen.

Ryan-A-Bradley, 2021-11-18:
i've also updated the main post to contain part 2 of this issue (if requested i will make the graphical issue as a separate issue).

PartialVolume, 2021-11-18:
Can you try putting nomodeset on the kernel line in /EFI/BOOT/grub.cfg.
```
set default="0"
set timeout="0"
gfxpayload=1920x1080x16
menuentry "shredos" {
linux /boot/shredos console=tty3 loglevel=3 nomodeset
}
```

Ryan-A-Bradley, 2021-11-18:
added nomodeset on the kernel line in EFI/BOOT/grub.cfg.

didn't fix the issue.

PartialVolume, 2021-11-18:
This is probably a long shot but can you remove the `nomodeset` and replace with `i915.alpha_support=1`

Ryan-A-Bradley, 2021-11-18:
replaced nomodeset with i915.alpha_support=1

didn't fix the issue.

PartialVolume, 2021-11-18:
I think I'm correct in saying your Intel core i7-10750H cpu is using  a Intel UHD Graphics 630 chipset.

I'm going to build shredos with `Laptop Hybrid Graphics - GPU switching support` and de-install the Intel KMS frame buffer, I'm wondering whether that's causing some conflict with the Linux fbdev driver. I'll build this version overnight and hopefully it will be ready tomorrow.

Ryan-A-Bradley, 2021-11-18:
alright! today's update has been excellent. we're able to now see shredOS is at least functioning, so it seems, which is good.

looking forward to seeing if tomorrow's update fixes the GUI! thank you for doing this by the way. ^^ and thank you to everyone else who's helped with making shredOS.

PartialVolume, 2021-11-18:
This sounds very similiar !

>KMS Issue: **console is limited to small area**
One of the low-resolution video ports may be enabled on boot which is causing the terminal to utilize a small area of the screen. To fix, explicitly disable the port with an i915 module setting with video=SVIDEO-1:d in the kernel command line parameter in the bootloader. See Kernel parameters for more info.
If that does not work, try disabling TV1 or VGA1 instead of SVIDEO-1. Video port names can be listed with xrandr.

Found [here](https://wiki.archlinux.org/title/intel_graphics#KMS_Issue:_console_is_limited_to_small_area)

You could try their suggestion put `video=SVIDEO-1:d` on the linux command line and see if that fixes it.

[65 more replies]

closed 2021-11-27 by PartialVolume