PartialVolume/shredos.x86_64 #53: minor issue: when making bootable USB, the drive volume isn't auto assigned a letter

Ryan-A-Bradley, 2021-11-18:
I'm on windows 10 home, updated to the latest stable release.

to gain GUI access to the volume:

diskpart
list disk
select disk 2 (disk 2 was the USB with shredOS on for me)
list volume
select volume 4
assign (auto assigns a drive letter)

this assigned a drive letter to the volume containing shredOS which then allowed me to edit the grub.cgf file easier.

PartialVolume, 2021-11-20:
This sounds like a Windows 10 issue rather than shredos. [windows-10-not-detecting-usb](https://www.google.com/amp/s/www.diskpart.com/windows-10/windows-10-not-detecting-usb.amp.html)

PartialVolume, 2021-11-21:
@Firminator I just found this option buried away in the kernel setup, disabled in my case. I think I'll enable this with the new 5.13 kernel (upgraded from 5.6) and see if it works on my older systems, and then we can see whether it solves the blank screen problem on some newer systems. I'll double check but I think simplefb was already installed.

```
CONFIG_X86_SYSFB:

  │ Firmwares often provide initial graphics framebuffers so the BIOS,            │
  │ bootloader or kernel can show basic video-output during boot for              │  
  │ user-guidance and debugging. Historically, x86 used the VESA BIOS             │  
  │ Extensions and EFI-framebuffers for this, which are mostly limited            │  
  │ to x86.                                                                       │  
  │ This option, if enabled, marks VGA/VBE/EFI framebuffers as generic            │  
  │ framebuffers so the new generic system-framebuffer drivers can be             │  
  │ used on x86. If the framebuffer is not compatible with the generic            │  
  │ modes, it is advertised as fallback platform framebuffer so legacy            │  
  │ drivers like efifb, vesafb and uvesafb can pick it up.                        │  
  │ If this option is not selected, all system framebuffers are always            │  
  │ marked as fallback platform framebuffers as usual.                            │  
  │                                                                               │  
  │ Note: Legacy fbdev drivers, including vesafb, efifb, uvesafb, will            │
  │ not be able to pick up generic system framebuffers if this option             │  
  │ is selected. You are highly encouraged to enable simplefb as                  │  
  │ replacement if you select this option. simplefb can correctly deal            │  
  │ with generic system framebuffers. But you should still keep vesafb            │  
  │ and others enabled as fallback if a system framebuffer is                     │  
  │ incompatible with simplefb.                                                   │  
  │                                                                               │  
  │ If unsure, say Y.                                                             │  
  │                                                                               │  
  │ Symbol: X86_SYSFB [=n]                                                        │  
  │ Type  : bool                                                                  │  
  │ Defined at arch/x86/Kconfig:2846                                              │  
  │   Prompt: Mark VGA/VBE/EFI FB as generic system framebuffer                   │
  │   Location:                                                                   │  
  │     -> Bus options (PCI etc.)                                                 │

Ryan-A-Bradley, 2021-11-22:
@PartialVolume i think you've wrote this comment in the wrong subject. xD

Ryan-A-Bradley, 2021-11-22:
as for it being a windows 10 problem, i wouldn't know. i've made other bootable USB images and they appear in file explorer once created with Rufus (they are automatically assigned a drive letter).

i'm not sure why shredOS isn't automatically assigned a drive letter.