PartialVolume/shredos.x86_64 #55: Question: SAS Drive Support?

github-pba, 2021-11-23:
Hi all!

**Question:** 
Does the current version of shredOS support SAS drives? Thank you!

**Background** (no need to read)
I have to wipe the disks of multiple IBM System X3650 M4 (https://lenovopress.com/tips1102-system-x3650-m4-bd) which are equipped with SAS/SATA-Drives. Unfortunatly, I cannot access the servers myself, as they are in a datacenter where I do not have access to. So I have to send multiple USB sticks to a field engineer who will insert them for me.

I simply want to make sure I can expect everything is working, so I do not have to exchange the USB sticks over and over again.

Thank you

PartialVolume, 2021-11-23:
Although I don't test Shredos on SAS raid hardware, apparently it does work on a number of raid controllers. However, whether it works with the IBM H1110, 1115, 5110 or N2115 raid controllers, I've no idea. There isn't a specific driver for those controllers that I can find in buildroot, however maybe it's generic and therefore compatible with the raid drivers we already have installed. Having said that I think red hat have a driver specifically for the IBM raid controllers.

Unless somebody else can tell us that they have specifically tested on IBM System X3650 M4, you may have to try it and see if any SAS/SATA discs are recognised. You'll also have to disable raid before it will recognise the discs.

Sorry I can't be of more help. Perhaps somebody who works on this type of hardware can provide more info.

github-pba, 2021-11-23:
Thank you for the reply. At least you're not pessimistic, and this lets me give it a shot.

If I have to break the RAID device (if any are configured), that's fine too. I want to purge these disks anyway, because the servers get recycled.

I will report back on my experience.

Firminator, 2021-11-23:
Support for a bunch of server SAS RAID controllers (including IBM ServeRAID) has been added recently: https://github.com/PartialVolume/shredos.x86_64/pull/40
So it should work. If it doesn't boot try to disable UEFI (set it to Legacy BIOS mode) and also disable SecureBoot. Since that server seems to be from around pre 2010 I'd assume ShredOS boots fine. Send us a screenshot of the wiping screen :)

github-pba, 2021-11-23:
Thanks again. 

It will take some time before I can do the wiping session, but I will send you screenshot when I'm working on it.

Firminator, 2021-11-23:
Just tried on an HP box from 2010 with 12 SAS drives and HP RAID controller and it's recognized. I had to use the older ISO though ( https://github.com/PartialVolume/shredos.x86_64/releases/download/v2020.05.017_x86-64_0.32.003/shredos-2020.05.017_x86-64_0.32.003_20211112.iso ). The  latest USB image ( https://github.com/PartialVolume/shredos.x86_64/releases/download/v2020.05.018_x86-64_0.32.014/shredos-2020.05.018_x86-64_0.32.014_20211118.img ) doesn't output video.

PartialVolume, 2021-11-23:
Is there some way I can share the new build with you, 60MB .img I don't want to make it public as it's definitely not fully debugged yet but it will give me an idea whether the new frame buffer code is working. (Different code to rev 18 .img above, that you had the no video problem with)

Networking doesn't work but drive detection is working. It's not for production use though, just for testing the video/UEFI issue.

Firminator, 2021-11-23:
I have to reword this after further testing:
If a RAID volume is present using HP RAID controller (HP SmartArray P410) ShredOS detects it. That's of course not a valid way to wipe. (In my case nwipe shows one 6TB volume as UNK). So you have to dismantle the RAID volume and hope that the controller provides pass-through or IT mode. The controller on that HP box doesn't provide it so nothing is detected after deleting the logical volume.
But it was worth testing this as I found ISO to IMG difference. Yes you can share the IMG with me somewhere.

Firminator, 2021-11-23:
Send me the link to ivno69@yadim.dismail.de and I delete the email right away. It's a disposable email service.

PartialVolume, 2021-11-23:
@Firminator Done.

Firminator, 2021-11-23:
Got it. Thanks.

Firminator, 2021-11-23:
Still doesn't work on 3 different systems. But as mentioned in the other issue I can confirm you can 'blindly' ALT+F2 to the other console and type `reboot` and it will reboot.

PartialVolume, 2021-11-23:
@Firminator I've moved discussion to [here](https://github.com/PartialVolume/shredos.x86_64/discussions/59#discussion-3702119)

Firminator, 2021-12-08:
@github-pba I would think you definitely have to break the RAID, but even then the IBM RAID controller might not present the drives to the OS. The controller has to firmware-flashed to IT-mode/pass-through/JBOD/HBA mode. Newer server RAID controllers have this setting in their firmware. It's then just a setting to change the 'personality' of the RAID controller.

For example a newerish DELL PERC 9:
>PERC 9 series of cards support two personality modes.
• RAID mode: RAID mode is commonly used and the controllers are mostly shipped from the factory in RAID mode. This mode allows the creation and operation of RAID virtual disks and non-RAID disks.
• HBA mode: In the HBA mode, PERC controller operates as Host Bus Adapter (HBA). This mode does not contain virtual disks or the ability to create them. All physical disks function as non-RAID disks under operating system control. The PERC card acts as a conduit between the host server and the physical disks. Input and output requests originate from the host and are passed through the controller to the physical drives.

I would think newerish IBM RAID controller have a similar setting.

PartialVolume, 2021-12-08:
Closing issue, please open a new issue if you have any further questions. Thanks.

closed 2021-12-08 by PartialVolume

github-pba, 2021-12-09:
@Firminator Thank you for your knowledge. I'm currently waiting to have the USB stick with ShredOS inserted to the server and then I will see what I get. I will report here for further references.