PartialVolume/shredos.x86_64 #60: Kernel Panic on the latest version

jay18001, 2021-11-24:
Looks like there is a kernel panic when booting from a usb image on `v2020.05.018_x86-64_0.32.014`. It boots fine on `v2020.05.017_x86-64_0.32.003`.

PartialVolume, 2021-11-24:
Thanks for reporting.

Were there any messages displayed? What are the system details? Make/model, CPU, GPU. Are you booting legacy or UEFI? Did you checksum the .img? Have you re-flashed the USB stick a second time?

jay18001, 2021-11-24:
The checksums were the same. I tried reflashing the USB drive a few times. I used Rufus on windows to flash it. I'm using UEFI. Also here's the error.

System info:
Motherboard: Z390 Aorus Ultra
CPU: Intel i7-9700k
GPU: Nvidia 3090

[image]

Firminator, 2021-11-25:
PartialVolume, am I reading this right? It's using the NV20 (super ancient NVIDIA card) for the framebuffer console. Then the kernel tries to use a function of that driver to read the clock speeds of the GPU and that's what's crashing the kernel.
I would assume that the added drivers in commit https://github.com/PartialVolume/shredos.x86_64/commit/8199551e5e8a87275468d2e8f9c46fc505f09951url specifically for this case
```
+CONFIG_FB_NVIDIA=y
+CONFIG_FB_NVIDIA_I2C=y
```
are causing the issue.

Riva is also NVIDIA (for example Riva TNT... that's the GPU before GeForce; I had one of those back in the days). Should go, too. I think all of these GPUs are basically non-existent these days so I'm not sure if these drivers do any good. NeoMagic and Voodoo GPU? Last seen 20 years+ ago.

The only one I see on that list that we might want to keep is Matrox. Some servers have these. The G200 if I remember correct. It comes in some DELL servers to provide VGA console output.

I think the previous VESA, EFI, and SIMPLE mode was working just fine.
CONFIG_FB_VESA=y
CONFIG_FB_EFI=y
CONFIG_FB_SIMPLE=y

On a second thought I'd like to check what drivers NHellFire was using in his fork....

Firminator, 2021-11-25:
He's using the same as ShredOS (without all the abcient FBdrivers added
CONFIG_FB_VESA=y
CONFIG_FB_EFI=y
CONFIG_FRAMEBUFFER_CONSOLE=y
but not CONFIG_SIMPLE.

Additionally he has defined
CONFIG_FB=y
CONFIG_FRAMEBUFFER_CONSOLE_DETECT_PRIMARY=y
see https://github.com/NHellFire/dban-buildroot/blob/dban/target/dban/linux-i586.config

PartialVolume, 2021-11-25:
You're right, those drivers seem to cause more trouble than they are worth. I'll pull them out of the build leaving.

CONFIG_FB_VESA=y
CONFIG_FB_EFI=y
CONFIG_FRAMEBUFFER_CONSOLE=y
Plus
CONFIG_SIMPLE=y

CONFIG_FB=y
CONFIG_FRAMEBUFFER_CONSOLE_DETECT_PRIMARY=y

PartialVolume, 2021-11-25:
@jay18001 I'm working on a new build without some problematic frame buffer drivers. I'll send you a link when it's ready to test on you hardware.

PartialVolume, 2021-11-26:
@jay18001 Can you download this .img file. [Shredos 2021.08.2 legacy and UEFI bootable](https://drive.google.com/file/d/1-vQYBDBs_HUgKwaIgVfvhw1C5f2g3uFn/view?usp=sharing) It's the latest 2021.08.2 development release. It should hopefully resolve this issue. Please let us know how it goes. Thanks.

jay18001, 2021-11-27:
It boots. I don't have a drive I can test the rest of the features on though.

PartialVolume, 2021-11-27:
That's great, thanks for letting us know. I'm confident it will function fine but if you should have any other problem then feel free to open a new issue.

PartialVolume, 2021-11-27:
Just to clarify for others reading this. This problem is fixed in the development version 2021.08.2 of ShredOS that hasn't been officially released yet. The development version  will be officially released in the next few days.

closed 2021-11-27 by PartialVolume