PartialVolume/shredos.x86_64 #61: Question: Does DoD Short wipe an SSD securely? If not, what methods should be used for an SSD?

Ryan-A-Bradley, 2021-11-27:
Does DoD Short wipe an SSD securely? If not, what methods should be used for an SSD?

PartialVolume, 2021-11-27:
This is just my own personal opinion and others may agree or disagree but I wouldn't generally do a DOD short wipe on an SSD.

There is a huge amount of discussion on this subject if you look through the open/closed issues on the nwipe GitHub pages but for what it's worth here's what I would do.

Do a ATA secure erase using hdparm first (nwipe currently doesn't do a ATA secure erase but will be able to do one in an upcoming release.) Then I would use hexedit to quickly check whether zeros or ones are present on the wiped disc. Then I would use ShredOS/nwipe to verify the drive using the Verify ones or verify zeros method.

Alternatively some people say you should do a 'write ones' method followed by a verify zeros.

If you are paranoid about wiping the disc maybe a secure erase, then a PRNG stream with blanking and final pass verification. I would try to keep such wipes as infrequently as possible to avoid shortening the life of the drive to much. I would probably only use this on disposal or transfer of the drive.

pogue, 2021-12-10:
I still like this idea as an alternative. I've seen it recommended SO MANY times to encrypt an SSD with a random key then format it.

[Secure Wiping of SSD Drives using Encryption & Formatting](https://bitbin.it/aJOhwukR/)

Sorry I had to post this on bitbin, I keep getting the error from Github "You can't comment at this time."

Firminator, 2021-12-28:
Probably wanna move this to https://github.com/PartialVolume/shredos.x86_64/discussions/categories/q-a