PartialVolume/shredos.x86_64 #66: Add temperature support for USB devices

PartialVolume, 2021-12-08:
Add temperature support for USB devices.

USB memory sticks don't provide temperature support, at least all the sticks I've come across don't. However a 2.5", 3" or SSD that are connected to the USB port with a USB to SATA adapter that supports 'ATA pass through' do provide temperature data.

USB temperature support to be scheduled for 0.33 release.

B3rn475, 2024-11-11:
Hi, sorry for jumping on this thread without being a user of ShredOS, which actually looks really cool.

I'm currently encountering the same issue, i.e. "`drivetemp` doesn't recognize USB SATA enclosures".
With the side effect that `hwmon` doesn't expose it, which means `lm-sensors` doesn't expose it, which means `<long chain of dependencies>` cannot use it...

**Do you know if something changed on this topic?**

I was only able to find references to this issue here and in some "highly passive aggressive" kernel patchmail thread from 2018.
**Do you know where official documentation/statements about this can be found?**

PartialVolume, 2024-11-12:
In the past I was submitting a few bug reports upstream but the guy who appeared to maintain it, no longer does so, so I've no idea who maintains it if anybody now. See #81 and I was submitting to https://github.com/groeck/drivetemp/issues/1#issuecomment-1000128093 however the https://github.com/groeck/ still exists but he no longer seems to support drivetemp. Maybe see if you can contact https://github.com/groeck/ and see if he knows who maintains it now.