PartialVolume/shredos.x86_64 #68: Update NWipe's Wiping Methods

pogue, 2021-12-08:
So, the documentation on your site says NWipe uses the following wiping techniques to wipe HDDs. However, all of them are from the 1990s, before there was SSD & Flash Memory. So I think it's time to update and use more advanced and optimized shredding techniques.

NWipe uses the following methods:

- Quick erase - Fills the device with zeros, one round only.
- RCMP TSSIT OPS-II - Royal Candian Mounted Police Technical Security Standard, OPS-II
- DoD Short - The American Department of Defense 5220.22-M short 3 pass wipe. 1,2,& 7.
- DoD 5220.22M - The American Department of Defense 5220.22-M full 7 pass wipe. 1-7
- Gutmann Wipe - Peter Gutmann's method. (Secure Deletion of Data from Magnetic and Solid-State Memory)
- PRNG Stream - Fills the device with a stream from the PRNG.
- Verify only - This method only reads the device and checks that it is all zero.
- HMG IS5 enhanced - Secure Sanitisation of Protectively Marked Information or Sensitive Information

Unfortunately, most of these are outdated and one (the Gutmann method) is a flat out waste of time.

The RCMP level 3 wipe has been superseded by CSEC ITSG-06.

> What Does CSEC ITSG-06 Do?
All data sanitization methods are similar, but what sets them apart from each other are the small details. For example, Write Zero is a method that only uses one pass of zeros. Gutmann overwrites the storage device with random characters, possibly up to dozens of times. However, the CSEC ITSG-06 data sanitization method is a little different in that it uses a combination of zeros and random characters, plus ones. It's usually implemented in the following way:

>Pass 1: Writes a one or zero
Pass 2: Writes the complement of the previously written character (e.g. one if Pass 1 was zero)
Pass 3: Writes a random character and verifies the write

So, it's just a 3 wipe algorithm, but it's much more advanced. Here is what I believe is the white paper on it: [IT Media Sanitization (ITSP.40.006)](https://cyber.gc.ca/en/guidance/it-media-sanitization-itsp40006)

Next is DoD 5220.22M ECE (7 wipes) and DoD 5220.22M aka DoD Short (3 wipes). These also have been superseded by standards from NIST. 

The DoD 5220.22M was developed during a time when we were still using tape backup drives, optical memory, EEPROM, cathode ray tubes and the like. You can read an interesting history of it on [White Canyon's DoD 5220.22-M Relevancy & The Evolution Of Wipe Standards](https://www.whitecanyon.com/white-papers/dod-relevancy). Apparently, the DoD is working on a 5200.48 standard, which they leaked the white paper on here: [DoD Instruction 5200.48 "Controlled Unclassified Information CUI](https://www.esd.whs.mil/Portals/54/Documents/DD/issuances/dodi/520048p.PDF?ver=2020-03-06-100640-800) (PDF). However, it may be in alpha/beta stages and not fully ready for prime time. That's why they let NIST, a very capable agency (they picked AES aka Rijndael as the standard encryption scheme, as opposed to a host of others, and it has proven to be a very fast and versatile & fast encryption standard.  [Advanced Encryption Standard Process & Competition](https://en.wikipedia.org/wiki/Advanced_Encryption_Standard_process)

[NIST Special Publication 800-88 Revision 1 Guidelines for Media Sanitization: https://nvlpubs.nist.gov/nistpubs/SpecialPublications/NIST.SP.800-88r1.pdf](url)

These are called _NIST 800-88 Clear_ and _NIST 800-88 Purge_. Both of these are described in the above document.

Finally, we have the Gutmann Wipe, which it's inventor,  Peter Gutmann, has said it is outdated and unnecessary.

>Most of the patterns in the Gutmann method were designed for older MFM/RLL encoded disks. Gutmann himself has noted that more modern drives no longer use these older encoding techniques, making parts of the method irrelevant. He said "**In the time since this paper was published, some people have treated the 35-pass overwrite technique described in it more as a kind of voodoo incantation to banish evil spirits than the result of a technical analysis of drive encoding techniques**"

[Source](https://en.wikipedia.org/wiki/Gutmann_method)

I searched and searched for an updated version of HMG IS5 enhanced aka "HMG Infosec Standard 5", which is what the British (apparently still even currently) use to scrub sensitive data. Either that, or they are keeping their secure wipe methods very close to their chest, which is what intelligence agencies usually do anyway. There is apparently a HMG IS6 but it's still either in development or classified by the UK government. In light of that, IS5 should be fine, as I found many currenct documentats saying something along the lines of the "The British Agent will use Infosec Standard 5 to wipe the file" 

So, the documentation on your site says NWipe uses the following wiping techniques to wipe HDDs. However, all of them are from the 1990s, before there was SSD & Flash Memory. So I think it's time to update and use more advanced and optimized shredding techniques.

I searched and searched for an updated version of HMG IS5 enhanced aka "HMG Infosec Standard 5", which is what the British (apparently still even currently) use to scrub sensitive data. Either that, or they are keeping their secure wipe methods very close to their chest, which is what intelligence agencies usually do anyway. There is apparently a HMG IS6, and I was approached by two gentlemen in suites and manhandled a little bit about finding this document. But, they eventually gave up and left. The documents read like they're only at the [green or blue paper stage](https://en.wikipedia.org/wiki/White_paper#Variants), and not [full white paper](https://en.wikipedia.org/wiki/White_paper) stage.

[Use of HMG Cryptography Policy](https://security-guidance.service.justice.gov.uk/use-of-hmg-cryptography-policy/#use-of-hmg-cryptography-policy-1) left. The documents read like they're only at the [green or blue paper stage](https://en.wikipedia.org/wiki/White_paper#Variants), and not [full white paper](https://en.wikipedia.org/wiki/White_paper) stage.

[Use of HMG Cryptography Policy](https://security-guidance.service.justice.gov.uk/use-of-hmg-cryptography-policy/#use-of-hmg-cryptography-policy-1)

Quick Erase would be good for SSD drives, as would PRNG Stream (although it doesn't specify the levels it does that at) they have a finite number of reads/writes they can do before they are completely destroyed. It's quit a lot, but if you're wiping your SSD HDD with the Gutmann method multiple times, running CCleaner with different level wipes, wiping free space, defragging often, you're more likely to encounter SSD failure (especially on how cheap or expensive the brand of the drive is).  

As far as the two PSRNGs, it appears Mersenne Twister is not out of date, as much as there are better PSRNGs available, not to mention they were created in the 90s.

[Letting Go Of The Mersenne Twister?](https://medium.com/asecuritysite-when-bob-met-alice/letting-go-of-the-mersenne-twister-580121f29152)

The author recommends instead, using CRNGs or Crypto Random Number Generators like [url=https://docs.microsoft.com/en-us/windows/win32/api/wincrypt/nf-wincrypt-cryptgenrandom]CryptGenRandom[/url] which was developed by Microsoft, but is now deprecated by Windows other CRNGs built into the OS, you couldn't use an API call from Windows to generate the random numbers in a bootable CD/USB environment.

[CryptGenRandom - Wikipedia](https://en.wikipedia.org/wiki/CryptGenRandom) - Not to mention, the source code to wincrypt.h is available on the [Wine Project's GitHub - WinCrypt.H](https://github.com/wine-mirror/wine/blob/master/include/wincrypt.h). As well as one of MIcrosoft's alternatives, [BCrypt.H](https://github.com/tpn/winsdk-10/blob/master/Include/10.0.14393.0/shared/bcrypt.h), another CRNG. Bcrypt was developed using Blowfish, Blowfish is a cryptographic algorithm developed by the famous Bruce Schneier, who also went on to make it's successor, Twofish. 

> Blowfish
General
Designers	Bruce Schneier
First published	1993
Successors	Twofish
Cipher detail
Key sizes	32–448 bits
Block sizes	64 bits
Structure	Feistel network
Rounds	16
[... 17 lines cut ...]

[RC4-based random number generators](https://en.wikipedia.org/wiki/RC4#RC4-based_random_number_generators)

> Several attacks on RC4 are able to distinguish its output from a random sequence.
[Source](https://link.springer.com/content/pdf/10.1007%2F3-540-45473-X_13.pdf)

Ain't that a bitch?

[... cut ...]  try to find Python implementations of those new cryptographic algorithms to use.

And just searching Google, I found a TON of people making  their own Crypto RNGs just using the calls in Python's OS.

Anyway, that was a long write up for me, but if you want me to break them into individual posts for each algorithm I can do so if you are actually interested in trying them.

I hope that might give you a few ideas and it wasn't too boring or tedious to read.
Thanks and keep up the good work!
pogue

pogue, 2021-12-10:
> @pogue I read that study some time ago. I'm not sure what you mean by 'it only does a one pass'. Their study shows that no single method can necessarily be trusted on its own. However it did actually show that on some drives it took two traditional random wipes to remove any sign of the fingerprint which to be honest, I thought was impressive. Even the ATA secure wipe can't necessarily be trusted from their evidence.

I'm with on you the Secure Erase. It seems like a solid solution. However, I would like to get some Forensic Software off somewhere *cough*Bittorrent*cough and test it and see if there is anything left on the drive afterwards.

> 
> So much depends on the drive/make/model/firmware rev. hence why I propose for nwipe a secure wipe followed by a verification, then a random stream, then random stream verification, followed by an optional ones write and verification for the highest security level. For medium security a ATA secure wipe then verification. Based on their findings maybe a single ATA secure erase and verification should be classed as low security. So much depends on the make/model of the drive and how they implement the erasure in firmware. That's why I use a wipe method that incorporates both a drive initiated wipe and a traditional random stream wipe that should hopefully cover all possibilities.

Sorry, what is a random stream and random stream verification?
> 
> It also depends on how valuable your data is and what lengths others might go to recover it. From what I've seen when it comes to military grade wiping nothing less than total destruction of the drive, i.e physically shredded having never left site is the only option. That includes engineers entering a military operation with a brand new spare drive. No hard drives leave site and if you brought it in you won't be taking it out, whether you used it or not !

Yeah, from what I've read about Secure Wipe, it all depends on if the HDD mfg implemented it properly to do the job it was meant to do. However, with SSDs, it is incredibly hard to physically destroy them because, as mentioned, they are often now soldered onto laptops and motherboards made by vendors and to remove them you basically have to destroy the computer to get them out. That's why I still prefer the method I listed below. I've seen it recommended MANY MANY times, more than I can count. Just encrypt the drive with a PSRNG for the password string, encrypt it, then just format it. That is good enough for SSDs.

ATAs we've known how to wipe for a long time now.
> 
> However unless you were actually disposing of the drive, I would just do a secure wipe & verification.
> 
> The issues with secure wipe, as is pointed out in the article, is how can you verify the disc has really been wiped, unless you take out all the chips and read them. I think one of the points they were also making was that the drive itself should have some verification function. However they didn't really go into whether the drives could be compromised by malicious firmware, which invalidates everything about 'secure erase' being secure.

You don't necessarily have to take out the chips to read them, they should be all located on the HDD itself. And some good forensic software would read it. I would be happy if someone volunteered their SSD to be wiped using this then they could mail me the drive and I could plug it in and read it with some high Quality Forensic software and go through it and see if there is anything but random data left.

pogue, 2021-12-10:
Not sure what happened to my other post...

```
I still like this idea as an alternative. I've seen it recommended SO MANY times to encrypt an SSD with a random key then format it.

[Secure Wiping of SSD Drives using Encryption & Formatting](https://bitbin.it/aJOhwukR/)

Sorry I had to post this on bitbin, I keep getting the error from Github "You can't comment at this time."
```

pogue, 2021-12-10:
Also, I finally did find **NIST document 800-88** 
According to that secure wiping methods document "approves SE at a higher security level than external software block overwrite utilities like as **Norton Government Wipe**, and it meets the legal requirements of HIPAA, PIPEDA, GLBA, and Sarbanes-Oxley.

However, I can find almost NOTHING about this so called "Norton Government Wipe". I will email Norton and ask them about this and if it is available for the general public or if it's only sold to LE/gov agencies.

Core2DuoKing, 2022-11-16:
> But, [BlackBlaze](https://www.backblaze.com/blog/how-to-securely-recycle-or-dispose-of-your-ssd/) (an online cloud storage company) recommends encrypting the SSD with a random key and then formatting it.
> 
> Maybe you could borrow some code from [VeryCrypt](https://www.veracrypt.fr/code/VeraCrypt/) to encrypt the HDD, use a PSNG to generate the password to encrypt the drive, and then run a full format on it.

Hello,

This I think this here is the future of dealing with SSD's, it just makes sense. I remember I found a similar guide here: https://its.fsu.edu/sites/g/files/upcbnu1011/files/ISPO%20Pages/Sanitize%20Drive.pdf which looks like it was put together by the IT department of Florida State University. They give instructions of encrypting a drive with Bitlocker in Windows, formatting, then encrypting again to encrypt the first encryption key and formatting again. 

My question would be if even using something like Bitlocker made by the great Microsoft, does it get into all the nooks and crannies of SSD's where possible data may be hiding or inaccessible? You would hope so. But I guess if you really needed something you can securely wipe, you wouldn't use SSD's in the first place.

Anyways, I guess all that I'm saying is... having a 2-pass encryption wipe method in nwipe would be really cool, and I would be interested in seeing these methods tested forensically.

shiroscout, 2023-02-12:
This is a bit of a late response, but what an interesting thread topic here.
I have a hard time trying to understand with Governent security specialists, cyber forenstic specialists, and the near same hdd manufacturers for the last 30 years why we do not have a unified answer to wiping, erasing disks.
I agree with a comment above, we need 1-3 methods that should wipe/erase a drive in a secure, as fast as possible way.
I agree heavily on the topic of needing updated wipe methods. We have basically the same regurgatated info on the same methods from decades ago.
Nearly every company does not know either, so they implement booting from USB most of the time or CD and running Killdisk. I'm fond of killdisk actually...
I do IT Support and it is a massive dread to do Wipes/erase... create the USB image, boot from USB, wait for hours for it to complete, copy the certificate to a sharepoint site, etc... and then there are the PC's that do not boot or power on and we have to disassemble the pc/laptop, and now we are also faced with Tablets, iPADs, Mobile phones, etc.. and seemingly no answer from any vendor, hdd manfacturer, government entity on what the best and quickest options are.
I won't do more than a 3 pass wipe and verify. If it needs more than that, it is cheaper and more cost effective to just pull and physically destroy the drive, although we try to keep as many working units as possible for best prices from recyclers, which is a whole other topic. Try finding a recycler for Zebra label printers, forklift PC's and other oddities in IT Equipment.

I do guess more PC manfufacturers will make use of UEFI, and Secure-Erase/ATA type utilities, but they need to offer proof and verification options. I have to use 3rd party utilities that have a print out of proof of wipe/erase. There are risk management, Legal, and HR related associated issues that requires a printer copy of proof drives have been securely wiped and all data attempted best effort to erase/wipe.

Anywho, thanks for the interesting and relevant discussion. :)

Firminator, 2023-02-18:
You summarized pretty well the recent (decade) issues in Data Destruction. I share and agree on all the points you brought up. Also what history has shown us is that you can't even rely on UEFI or drive manufacturers to wipe correctly as their implementations are sometimes shortcuts and/or buggy, and their source code is not open-source and verifiable. For example if SecureErase is implemeneted in the UEFI of a current mainboard it might work on this specific UEFI/BIOS version, but another mainboard model from the same vendor might have a totally different implementation and it might be buggy. We just don't know as information is sparce and they don't share versioning numbers, changelog or code of their SecureErase implemenation.

Firminator, 2023-03-03:
Well low and behold Dell just released an update for one of their laptops in our fleet admitting that the UEFI wipe wasn't working properly confirming in the real world what we know only from academic research of the past.

`- Fixed the issue where the Data Wipe function does not work on certain solid state drives.
`
source: https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=vxgmt