PartialVolume/shredos.x86_64 #69: hot swap support

ITJamie, 2021-12-08:
Is / Would it be possible to have hot swap support with auto wiping?

We want to build a system that we can pop drives into and have them automatically wipe and unmount when complete. we would remove them once there is no more drive activity.

Binary-Bear, 2022-05-09:
Hot swap is an excellent idea!

XVII, 2025-04-29:
Part of our use case here would be optimising the wiping process for a large quantity of drive with varying sizes.

i.e. Drive 1 is 500 GB and is wiped (waiting on 100%) but Drive 2 is 4 TB and must complete before the 500 GB can be swapped in.
Ideally we could plug another drive in and have it start wiping rather than waiting idle.