PartialVolume/shredos.x86_64 #70: ShredOS fails to wake from suspend

PartialVolume, 2021-12-08:
rtcwake -m mem -s 5 does indeed put the device to sleep but upon waking reboots ShredOS rather than the expected behaviour of restoring the running ShredOS from RAM.

PartialVolume, 2022-02-14:
I've been looking at the issue of ShredOS not waking from sleep this evening, and specifically what part of the system specifically doesn't wake.  This is a DELL Optiplex 9010. I've found I can telnet into ShredOS and the drives are indeed 'not frozen' and I can do a secure erase via hdparm. What seems to happen is that the display hardware (Intel) doesn't wake from sleep. The keyboard is functional so it seems the consoles ALT-F1-F3 is working. It's just the display that isn't.

I've been looking at the various power related features in /sys/devices/ etc and toggled auto to on in control but no luck yet.

PartialVolume, 2022-02-14:
>For a standard installed Linux distribution, it's usually a matter of pushing options to the video driver. Interesting that both of my test cases have integrated Intel graphics, which your P4 doesn't. (Of course, integration itself may not be significant – other things will also have changed over time.)

And my Dell optiflex also having integrated Intel graphics.

I think I will get on with making the changes to nwipe so it supports ATA secure erase and controls putting the system to sleep then come back to this.

PartialVolume, 2022-02-15:
Some bedtime reading [https://www.intel.com › publicPDF
Linux Power Management -- IEGD Considerations - Intel](https://www.intel.com/content/dam/www/public/us/en/documents/white-papers/linux-power-mgmt-paper.pdf)

PartialVolume, 2022-02-17:
@hameau @Firminator Looks like I may have solved the suspend to RAM issue. I've added some drm [Direct Rendering Manager](https://dri.freedesktop.org/docs/drm/gpu/) code to ShredOS along with a selection of Nvidia, AMD and Intel DRM drivers and some configuration of ACPI and now ShredOS on my Nvidia laptop boots ShredOS with the native resolution of the screen. Previously it only ever displayed ShredOS in a low Res display.

Most importantly, the command `echo "mem" > /sys/power/state` powers down the drives and graphics system and then hitting the power button correctly restores ShredOS with all the drives 'Not Frozen'.  Within nwipe this process will be handled by `rtcwake -s 5 -m mem` to automatically put ShredOS to sleep for five seconds in order to unfreeze the drives prior to issuing the secure erase.

I can now start writing the secure erase code in nwipe and have a working ShredOS that will correctly suspend to RAM that I can use as a test bed.

Before I do a release I'll need some testing done on systems with lots of different graphics hardware to make sure ShredOS can always be sent to sleep and restored with fully working graphics.

If anybody is interested in testing the beta version of ShredOS with secure erase then let me know and I'll send you a link to download the .img in due course.

gorbiWTF, 2022-02-18:
> Before I do a release I'll need some testing done on systems with lots of different graphics hardware to make sure ShredOS can always be sent to sleep and restored with fully working graphics.
>
> If [...]

I've got a couple of different combinations of graphic cards and monitor resolutions, newer and older hardware, which I can test if you want.

PartialVolume, 2022-02-18:
The test procedure: Once ShredOS has booted, ALT F2 to bring up the 2nd virtual terminal.

ShredOS version: You will need to be testing against the pre-release  version v2021.08.2_21_x86-64_0.32.024 or higher. I'm not interested in the results from earlier revisions. This version will be pre-released either today or over the weekend.

Once ShredOS has booted run the following rtcwake & hdparm commands:

rtcwake -s 5 -m mem
  Expected Result: System should power off hard drives and display hardware, display goes blank, backlight switches off on laptops. Around 9 seconds elapse until drives are back up and running and display is on. Keyboard should be responsive.

hdparm -I /dev/sda
 Expected Result: In the Security section near the end of the listing it should say "not frozen". If it says only "frozen" the disc drive power off never worked correctly.

Repeat for each drive on the system.

The following information can be obtained from the nwipe log (after a Control C) and from lspci

| Graphics | Motherboard | Bios Notes etc | RTC Status | Frozen state (success=not frozen |
| ------------ | ----------------- | -------------------- | --------------- | -------------------------------------------- |
| Nvidia GM107 (GeForce GTX750) | Gigabyte H81M-DS2V | Bios version F5 06/19/2014 | SUCCESS | SUCCESS |
| Intel IGC Crystal Well | Apple MacBookPro 11,2 | Bios 430.0.0.0.0 12/17/202 | SUCCESS | SUCCESS |

Add you findings in the comments and I'll add your results to the table. Starting off with a I7 Nvidia laptop and a MacBook Pro that I'm currently testing.

PartialVolume, 2022-02-19:
A note about screen resolution: With the new DRM drivers, the kernel controls the resolution of the display using KMS (kernel mode setting) With all the systems I've tested so far the kernel chooses the native resolution of the monitor. On a high resolution monitor this can result in quite small text. If this isn't to your taste you can have ShredOS boot using the 640x480 (80x30) resolution, nwipe then fills the entire monitor with larger text.

To disable KMS and revert to low resolution display and larger text add the following to the kernel command line

### HOWEVER, if you disable KMS using the following method you will most likely find that rtcwake will no longer work as the kernel can't then control the display hardware to rewake it. So if you want secure erase to work properly DO NOT make the following additions to the linux kernel command line in grub.cfg

For Nvidia hardware, add nomodeset  and nouveau.modeset=0
```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 nomodeset nouveau.modeset=0
}
```

For Intel, add nomodeset and i915.modeset=0
```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 nomodeset i915.modeset=0
}
```
For Nvidia Optimus dual-graphics system, you need to add all the three kernel parameters (i.e. "nomodeset i915.modeset=0 nouveau.modeset=0").

```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 nomodeset i915.modeset=0 nouveau.modeset=0
}

PartialVolume, 2022-02-19:
With reference to the above comment about screen resolution, the preferred way to increase font size is to not mess with KMS and NOT set `nomodeset` and therefore let the kernel set the native resolution, however,  in order to increase the font size, use the command
```
setfont -d
```
This doubles the size of the existing font. I think I may put an option in nwipe that toggles between default and double size font, specifically for high resolution monitors

The images below show a monitor operating at it's native resolution of 1920x1200. The first image is the default 8x16 font. The second image is after having run the `setfont -d` command. In both cases the monitor is still operating at 1920x1200.

[image]
[image]

PartialVolume, 2022-02-19:
A little more info about manually doubling the font size of the default nwipe that runs in tty1.

So type ALT F2 (Fn ALT F2 on a MAC) to bring up the 2nd virtual console. Type the following `tty` command which will return the current console name. So from this result `/dev/tty2` we can deduce that the default nwipe in ALT F1 is /dev/tty1. For reference ALT F3 is /dev/console.
```
tty
/dev/tty2
```
So to set the font for the default nwipe in the first virtual console ALT F1 (/dev/tty1), type the following command in the 2nd virtual console (ALT F2)
```
setfont -d -C /dev/tty1
```

gorbiWTF, 2022-02-22:
<html><body>

Graphics | Motherboard | Bios Notes etc | rtcwake status | hdparm frozen state (success=not frozen
-- | -- | -- | -- | --
Intel UHD 630 | Lenovo 312A | M1UKT68A (2021-12-27) | SUCCESS | SUCCESS **(*)**
Intel HD 4600 | HP 1998 | L01 v02.71 (2017-05-09) | SUCCESS | SUCCESS 

</body>
</html>

**(*)** I've got two drives in this machine: SATA drive is "not frozen", but for the NVME drive `hdparm -I /dev/nvme0n1` and `hdparm -I /dev/nvme0` returns nothing.

Even resolutions up to 5120x1440 are working flawlessly for me.

However, I got two machines which don't want to boot: "HP Compaq 8100 Elite" (doesn't want to boot at all) and "HP EliteDesk 800 G4" (stuck on "booting 'shredos'").

PartialVolume, 2022-02-22:
>but for the NVME drive hdparm -I /dev/nvme0n1 and hdparm -I /dev/nvme0 returns nothing.

hdparm doesn't support nvme, with nvme drives you would use [nvme-cli](https://nvmexpress.org/open-source-nvme-management-utility-nvme-command-line-interface-nvme-cli/), however I have read that even nvme-cli doesn't provide security status 'frozen' information. Don't quote me on that though as I've got no nvme hardware to test on. With nvme I think it's a case of sending ShredOS to sleep then issuing the secure erase command via nvme-cli.

This is something I'm going to have to take care of in nwipe. When it recognises a nvme drive it switched to using nvme-cli.

nvme-cli should be available on ShredOS so you may want to try a wipe using that. Let me know how it works out. I think @Firminator uses nvme-cli and may be able to help here.

PartialVolume, 2022-02-22:
@gorbiWTF 
>Even resolutions up to 5120x1440 are working flawlessly for me.

On those high resolution screens did you get to try the /bin/setfont -d command? Was the text better for reading at a distance?

PartialVolume, 2022-02-22:
@gorbiWTF 
>However, I got two machines which don't want to boot: "HP Compaq 8100 Elite" (doesn't want to boot at all) and "HP EliteDesk 800 G4" (stuck on "booting 'shredos'").

Do those two systems have a secondary graphics card installed by any chance?

PartialVolume, 2022-02-22:
@gorbiWTF  Come to think of it, I think I had a HP Compaq 8100 Elite at some point and could never get the thing to boot from USB. It worked fine booting from CD though. Think I may have to build the .iso version of this pre-release.

gorbiWTF, 2022-02-22:
> > but for the NVME drive hdparm -I /dev/nvme0n1 and hdparm -I /dev/nvme0 returns nothing.
> 
> hdparm doesn't support nvme, with nvme drives you would use [nvme-cli](https://nvmexpress.org/open-source-nvme-management-utility-nvme-command-line-interface-nvme-cli/), however I have read that even nvme-cli doesn't provide security status 'frozen' information. Don't quote me on that though as I've got no nvme hardware to test on. With nvme I think it's a case of sending ShredOS to sleep then issuing the secure erase command via nvme-cli.
> 
> This is something I'm going to have to take care of in nwipe. When it recognises a nvme drive it switches to using nvme-cli.
> 
> nvme-cli should be available on ShredOS so you may want to try a wipe using that. Let me know how it works out. I think @Firminator uses nvme-cli and may be able to help here.

I see. When I first send the system to sleep, then use `nvme format /dev/nvme0n1 -s 1 -l 0` it says "_The LBA Format specified is not supported. This may be due to various condiditions(0x10a)_". However, `nvme id-ns /dev/nvme0n1 -n 1 -H` says "_LBA FORMAT 0 : Meadata Size: 0 bytes - Data Size: 512 bytes - Relative Performance: 0 Best (in use)_".

> @gorbiWTF
> 
> > Even resolutions up to 5120x1440 are working flawlessly for me.
> 
> On those high resolution screens did you get to try the /bin/setfont -d command? Was the text better for reading at a distance?

Text size is fine for me out of the box, with the command I think it's too big. The screen only has 109 ppi, but I will try it later with a 164 ppi screen.

> @gorbiWTF
> 
> > However, I got two machines which don't want to boot: "HP Compaq 8100 Elite" (doesn't want to boot at all) and "HP EliteDesk 800 G4" (stuck on "booting 'shredos'").
> 
> Do those two systems have a secondary graphics card installed by any chance?

Yes, they both do. 

> @gorbiWTF Come to think of it, I think I had a HP Compaq 8100 Elite at some point and could never get the thing to boot from USB. It worked fine booting from CD though. Think I may have to build the .iso version of this pre-release.

Haha, well, I got the same problem when I tried to install Windows 10 from an USB drive, but a BIOS update fixed that for me.

PartialVolume, 2022-02-22:
> I see. When I first send the system to sleep, then use `nvme format /dev/nvme0n1 -s 1 -l 0` it says "_The LBA Format specified is not supported. This may be due to various condiditions(0x10a)_". [...]

@gorbiWTF  Did you send shredos to sleep with a `rtcwake -s 5 -m mem` before running that command?

PartialVolume, 2022-02-22:
> @gorbiWTF Did you send shredos to sleep with a `rtcwake -s 5 -m mem` before running that command?

@gorbiWTF Sorry, just reread your comment, looks like you did send the system to sleep before running that command.

PartialVolume, 2022-02-22:
> @gorbiWTF Sorry, just reread your comment, looks like you did send the system to sleep before running that command.

Is the system shutting down so there is no display (or backlight) and all fans are off and if there are any spinning discs they have stopped spinning? One thing to check is whether the bios contains any ACPI settings, I found one of my systems was not powering down sufficiently because there was a setting that was set as S1 when it needed to be S3. Could be something like that. Maybe there is something specific in the bios for M2 or nmve power managment?

gorbiWTF, 2022-02-24:
> Is the system shutting down so there is no display (or backlight) and all fans are off and if there are any spinning discs they have stopped spinning? One thing to check is whether the bios contains [...]

Yes, the system is shutting down like all other system, the SATA drive correctly changes from "frozen" to "not frozen" but I can't use `nvme format` on the NVME drive.

gorbiWTF, 2022-02-24:
> > @gorbiWTF
> 
> > > However, I got two machines which don't want to boot: "HP Compaq 8100 Elite" (doesn't want to boot at all) and "HP EliteDesk 800 G4" (stuck on "booting 'shredos'").
> 
> > > Do those two systems have a secondary graphics card installed by any chance?
> 
> > Yes, they both do.
> 
> It would be interesting to see if pulling out the secondary graphics card fixed either of them, at least it would indicate a graphics related issue. I don't know if you have two monitors attached but it would be worth checking the output of both video outputs. I had a similar problem, the bios allowed you to set the default card but that appeared to have no affect. Yet other systems worked nicely with two graphics cards. Funnily enough the systems who's bios allowed you to set the default card didn't work and those bios that had no video setting worked just fine. Still, I guess it's a quick test to pull the secondary card out of the system and see if ShredOS boots properly.

I removed the graphics card from the HP EliteDesk 800 G4 and then ShredOS worked as expected. I can't remove the graphics card from the other system because the iGPU isn't working anymore.

I also tried it on my notebook with an external screen: "booting 'shredos'" is showing up on the integrated screen (and always stays there), while the UI is on the external screen.

gorbiWTF, 2022-02-24:
<html><body>

Graphics | Motherboard | Bios Notes etc | rtcwake status | hdparm frozen state (success=not frozen
-- | -- | -- | -- | --
Intel HD 4400 | Acer Veriton X4630G | P11-A0L (2013-08-15) | SUCCESS | SUCCESS
Intel HD 530 | Fujitsu D3400-A1 | v5.0.0.11 R1.29.0 for D3400-A1x (2020-01-27) | SUCCESS | SUCCESS
Intel HD 2000 | HP 1495 | J01 v02.06 (2011-06-09) | SUCCESS | SUCCESS
Intel HD 5500 + Radeon R9 M275X | Lenovo Z51-70 | C2CN19WWCU2.00 (2015-07-14) | system just reboots | -
Nvidia 1650 Ti Mobile + AMD Renoir (Ryzen 7 4800H) | SchenkerTechnologiesGmbH GK7NxxR M20 | N.1.20.A03 (2020-12-10) | system turns off but not back on | -

</body>
</html>

I also got a HP Compaq dc7900 Convertible MT which can't boot (it just flashes an underscore cursor).

PartialVolume, 2022-02-24:
>I also got a HP Compaq dc7900 Convertible MT which can't boot (it just flashes an underscore cursor).

Does this boot with the previous version:
.img [shredos-2021.08.2_21_x86-64_0.32.023_20220123.img](https://github.com/PartialVolume/shredos.x86_64/releases/download/v2021.08.2_21_x86-64_0.32.023/shredos-2021.08.2_21_x86-64_0.32.023_20220123.img)
or
.iso [shredos-2021.08.2_21_x86-64_0.32.023_20220126.iso](https://github.com/PartialVolume/shredos.x86_64/releases/download/v2021.08.2_21_x86-64_0.32.023/shredos-2021.08.2_21_x86-64_0.32.023_20220126.iso)

If it boots with the .iso then at least we know it's likely to be the HP USB boot issue.

Alphakilo, 2022-12-13:
Adding to the list a HPE ProLiant Microserver Gen8

| Graphics |	Motherboard| 	Bios Notes etc| 	rtcwake status| 	hdparm frozen state (success=not frozen|
| ----------- | ----------- | ----------- | ----------- | ----------- |
|Matrox MGA 200EH| 	Gen8 Microserver Option Number 819185-001 |	J06 05/21/2018 | 	SUCCESS| 	FAILURE|

```sh
for d in sd{a,b,c,d}; do hdparm -I /dev/$d | grep frozen; done
   frozen
   frozen
   frozen
   frozen
```

What does however work is hot-plugging the drives. That _should_ work on every platform. So far the best results I've achieved was by reconnecting the power cable, while the data cable stayed in place.

PartialVolume, 2023-04-02:
Closing, as with now have switched to Linux DRM and include all the necessary firmware for most graphics cards, `rtcwake -m mem -s 5` should work fine on the majority of systems. Please re-open if you have a system it doesn't work on.

closed 2023-04-02 by PartialVolume