PartialVolume/shredos.x86_64 #79: Stuck on "Booting `shredos`"

JasonHorkles, 2021-12-17:
When I attempt to boot using a USB stick, the screen says it's booting but when it gets to 100% it just sits there and does nothing. I've tried different USB sticks and it hasn't changed anything. It seems to also only do it on this one laptop for some reason, as it works fine on others. Is there something specific that might cause it to not work on a single device? (It should also be a 64 bit device)

PartialVolume, 2021-12-17:
What's the laptop make/model?

Firminator, 2021-12-17:
Jason,
Are you using the latest version of ShredOS ( https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_20_x86-64_0.32.014 )?

JasonHorkles, 2021-12-17:
> Jason, Are you using the latest version of ShredOS ( https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_20_x86-64_0.32.014 )?

Yes

> What's the laptop make/model?

https://www.dell.com/support/home/en-us/product-support/product/xps-15-9510-laptop/overview

PartialVolume, 2021-12-17:
Are you booting it in legacy mode with the legacy option ROMs enabled in the bios or UEFI with secure boot disabled?

According to the service manual for that laptop both methods are supported. If which ever method you are using to boot it, it may worth trying the other method to see if it boots up more quickly.

I've only every come across an old Sony laptop that took several minutes to boot. But that was fixed in a recent update.

PartialVolume, 2021-12-17:
[Service Manual](https://www.google.com/url?sa=t&source=web&rct=j&url=https://dl.dell.com/topicspdf/xps-15-9510-laptop_service-manual_en-us.pdf&ved=2ahUKEwi-8bi06-v0AhXMSsAKHekQCKoQFnoECAcQAQ&usg=AOvVaw2pRAT-P01Hmgy5thC3YTzy)

According to page 51 of the service manual it defaults to legacy. If that is actually how it's setup then it's worth giving UEFI a go, I'm assuming you get the choice of legacy or UEFI in the one off boot menu?

JasonHorkles, 2021-12-17:
> [Service [...]

This laptop only supports UEFI mode (I turned off secure boot) as there are no options in the BIOS to switch to legacy

Firminator, 2021-12-17:
Latest BIOS?
Switchable GPU? If yes which GPUs?

JasonHorkles, 2021-12-18:
Look like it's capped at 1 MB/s for some reason.
BIOS probably aren't the latest, but I can try updating them. No switchable GPUs

Firminator, 2021-12-18:
Try a different USB flash drive maybe to rule the one you are using has issues.

JasonHorkles, 2021-12-18:
I've tried with like 4 different drives lol

Firminator, 2021-12-18:
>As an update, I left it on that screen overnight and came back to it in the correct menu, so it seems to just be super slow for some reason

Once it's booted and presents the GUI try `ALT+F2` to bring up second console and then pipe out dmesg>dmesg.log, then mount another flash drive and copy the file over and attach it to this thread.
At least from there we could possibly see where the delay is happening to present the GUI.

The other issue (1MB throuput only on USB bus) I'm not sure how to troubleshoot since there is only a limited amount of packages in ShredOS.

PartialVolume, 2021-12-20:
> When I attempt to boot using a USB stick, the screen says it's booting but when it gets to 100% it just sits there and does nothing. I've tried different USB sticks and it hasn't changed anything. [...]

When ShredOS is booting and it reaches 100% did any text get displayed below that 100%, such as a few lines of text issued by the kernel?

I'm wondering whether it's a frame buffer issue, some other driver issue or maybe smartctl hanging the process. I come across a drive that caused smartctl to hang. As we call smartctl before we display the GUI it would appear to cause a hang, but in such a situation I would still expect to see a few lines of kernel messages after reaching the 100% message. Like @Firminator  says, the dmesg output may tell us more.

Also, what's the history of this laptop, is it known to work reliably with other Linux distros or Windows? Just trying to rule out any hardware fault.

JasonHorkles, 2021-12-20:
> When ShredOS is booting and it reaches 100% did any text get displayed below that 100%, such as a few lines of text issued by the kernel?
No additional text. I'll try and get the output sometime soon but it's a work laptop and I have a lot to do before we have our break for the rest of the year.

Not sure on the history exactly, but it was running Windows and haven't had any reports of it having issues.

JasonHorkles, 2022-01-05:
[Log](https://paste.gg/p/JasonHorkles/a2257fbaa7ab4abb8efc622c27d03ba4)

PartialVolume, 2022-01-05:
Having had a quick look at dmesg.log I can't see anything that looks like an issue.

This line is interesting.
`Console: switching to colour frame buffer device 240x75`

Could be that the graphics card doesn't support the resolution we are using. Is it booting legacy or EFI? If you have the option to boot using either method then try both and see if it works. If it does boot legacy you could add an entry to grub.cfg to specify a particular resolution. [How to change resolution](https://github.com/PartialVolume/shredos.x86_64#how-to-make-a-persistent-change-to-the-terminal-resolution)

JasonHorkles, 2022-01-05:
It's UEFI only

PartialVolume, 2022-01-05:
Are you able to plug a network lead into it and telnet in from another computer?. If you can and thereby run nwipe through a terminal session it proves it's booted and confirms it's a framebuffer issue. [Logging in via the network](https://github.com/PartialVolume/shredos.x86_64#how-to-wipe-drives-on-headless-systems-or-systems-with-faulty-display-hardware-for-use-on-secure-lans-only)

JasonHorkles, 2022-01-05:
I don't have a whole lot of time to experiment with something that only seems to happen on this singular laptop so I might just close the issue unless you'd like it to remain open. Thanks for your help though.

closed 2022-01-05 by JasonHorkles

Firminator, 2022-01-06:
It seems to hang while enumerating the nvme device(s). Takes a whole 13 minutes!!!
```
[    1.289998] ACPI: \_SB_.PR11: Found 3 idle states
[   62.492701] nvme nvme0: I/O 16 QID 0 timeout, completion polled
[  123.932706] nvme nvme0: I/O 28 QID 0 timeout, completion polled
[  185.372701] nvme nvme0: I/O 17 QID 0 timeout, completion polled
[  246.812676] nvme nvme0: I/O 29 QID 0 timeout, completion polled
[  308.252668] nvme nvme0: I/O 18 QID 0 timeout, completion polled
[  369.692671] nvme nvme0: I/O 28 QID 0 timeout, completion polled
[  369.692779] nvme nvme0: allocated 32 MiB host memory buffer.
[  431.132677] nvme nvme0: I/O 19 QID 0 timeout, completion polled
[  492.572670] nvme nvme0: I/O 29 QID 0 timeout, completion polled
[  554.012678] nvme nvme0: I/O 28 QID 0 timeout, completion polled
[  615.452677] nvme nvme0: I/O 29 QID 0 timeout, completion polled
[  615.452695] nvme nvme0: 1/0/0 default/read/poll queues
[  676.892675] nvme nvme0: I/O 28 QID 0 timeout, completion polled
[  738.332683] nvme nvme0: I/O 29 QID 0 timeout, completion polled
[  799.772671] nvme nvme0: I/O 28 QID 0 timeout, completion polled
[  861.212673] nvme nvme0: I/O 29 QID 0 timeout, completion polled
[  891.933034] nvme nvme0: I/O 0 QID 1 timeout, completion polled
[  891.957346] scsi host0: pata_legacy
```
That leads me to believe this is a problem with whatever NMVe drive is hanging in there as NVMe0. I'd replace the drive with another model and see if that fixes the issue (it should). Also recommend to check for FW updates for that NVMe drive.

Also try https://duckduckgo.com/?q="nvme0%3A+I%2FO"

PartialVolume, 2022-01-06:
@Firminator nice one, I wasn't paying any attention to the timestamps. But obvious now looking at the log. 👍

Firminator, 2022-01-07:
Yeah that theory of a bad NVMe drive indicated by the dmesg is kind of backed by what he's writing: "something that only seems to happen on this singular laptop". I wonder what make and model the NVMe drive is? I saw ADATA drives mentioned a few times in other poeples reports for NVMe IO QID timeouts...

JasonHorkles, 2022-01-07:
I'm just dealing with it wiping for a super long time first and once it's done I'll actually open it up and take a look inside. I can almost guarantee it's some computer/hardware issue rather than with ShredOS anyways.

JasonHorkles, 2022-01-21:
Booted into Ubuntu off the same drive just fine and it did well on a disk test, so not sure why that singular computer doesn't like this singular program. Either way I was able to wipe it in the end so not too concerned, but just thought I'd update you anyways.
If in the future I have more problems with that specific type of laptop I can re-open the issue.

JasonHorkles, 2022-01-26:
Seems like ShredOS does just have issues with these specific laptops. Everything else still seems to work as normal, too

Firminator, 2022-01-28:
Which newer Dell laptops? Anything besides the XPS15/9510 that you initially reported?

[I have no problems with the very latest Dell Precision (TigerLake and 3 NVMe slots).]

JasonHorkles, 2022-01-28:
Latitude 5320 also had the issue

Firminator, 2022-02-01:
What's the model of the NVMe drive on your XPS15/9510 and in the Lat5320?

yuxiliu1995, 2023-05-07:
I have the same issue with Dell Latitude 7490.

PartialVolume, 2023-05-07:
@yuxiliu1995 is AHCI enabled in the bios rather than RAID which is the default?

```
SATA Operation Allows you to configure the internal SATA hard-drive controller. The options are:
● Disabled
● AHCI 
● RAID On: This option is enabled by default.
```

PartialVolume, 2023-05-07:
On some systems where the  AHCI/RAID option exists you need to select AHCI for SATA operation which on some systems, not necessarily Dell is also required on systems with NVME drives.

Which version are you using?

Does booting reach 100% as per original post, then blank screen or does it still say 100% on screen?

JasonHorkles, 2023-05-07:
> What's the model of the NVMe drive on your XPS15/9510 and in the Lat5320?

Oop sorry I never replied to this, although it's hard to say at this point since it's been such a long time and the laptop was probably given out to someone else at this point