PartialVolume/shredos.x86_64 #84: Suggestion: Add ability to autonuke if inactive after so much time

JasonHorkles, 2022-01-07:
It would be nice if all connected drives could automatically start being erased (except the USB itself ofc) after a certain amount of time specified in the grub config. That way if I don't touch the machine after so long it'll just start wiping.

PartialVolume, 2022-02-12:
If you are planning on wiping the machine, why not just set `--autonuke` in the grub config. I suppose nwipe could have a 30 second on screen count down prior to auto-nuke, in that way if you booted ShredOS and had forgotten that you had configured it as auto-nuke it would give you 30 seconds to abort the auto-nuke by pressing any key.

I can imagine that would be an absolute life safer to some poor soul that didn't realise they were booting from a ShredOS USB stick that had been configured to autonuke their system.

The more I think about it, the more I think a count down should be added. Anybody else have an opinion on this especially if you routinely use autonuke?

@JasonHorkles Is that the use case you are thinking of?

JasonHorkles, 2022-02-12:
Yep