PartialVolume/shredos.x86_64 #86: Can't see USB devices and SATA Devices

ItsSour, 2022-01-21:
Computer Specifications
Dell Optiplex 3040
BIOS Version: 1.18.1

I have been messing with the BIOS for about 3 days now and I can't get Shred OS to have storage devices connected to the SATA ports inside the computer and the external HDDs I have connected through USB show up in the same menu. I have one HDD connected via SATA and 3 via USB (There is only 2 SATA connections), and the only way I can see the USB is if I change the SATA configuration to "Disabled", but then it takes away the HDD I have connected via SATA. The only way I have found to see the SATA drive is to turn the SATA Configuration back to AHCI. There are only 2 options in that menu, AHCI or Disabled. Is there any way to work around this or am I stuck having to switch between the two?

PartialVolume, 2022-01-21:
Just checking that you restarting nwipe after the initial boot by doing a control-C? Nwipe only shows the USB drives after a control-C.

Firminator, 2022-01-23:
Aslo AHCI needs to be on at all times. Setting it to disabled will effectively render the 2 SATA ports useless as the connected drives will not be able to talk to the chipset using the AHCI interface. What you experienced (missing HDDs on the SATA port) there when disabling AHCI is expected behavior.

USB drives not showing up could also be because the DELL BIOS is set to Fast Boot disabling initialization of devices that are not necessary for booting the OS. This could be HDD/SSD over USB. Check in the PowerManagement menu and disable Fast Boot. If it's not there is might be under 'POST Behaviour'.

Else what PartialVolume said (restart nwipe with CTRL+C).

ItsSour, 2022-02-07:
> Just checking that you restarting nwipe after the initial boot by doing a control-C? Nwipe only shows the USB drives after a control-C.

Yes, I tried CTRL+C and with AHCI enable, it only has the USB drive not the SATA drives.

ItsSour, 2022-02-07:
> Aslo AHCI needs to be on at all times. Setting it to disabled will effectively render the 2 SATA ports useless as the connected drives will not be able to talk to the chipset using the AHCI [...]

I did everything you said, Fast Boot only has 3 options, Minimal, Thorough and Auto. Minimal looks to enable it just by reading the descriptions and Thorough looks to do a full hardware scan of all devices. I set it to thorough and set it to AHCI, it only shows the USB drives and not the SATA drives. I did notice that initially it does not pick up the SATA drives on boot of ShredOS, and reboots itself, then picks up the USB drives.

PartialVolume, 2022-02-07:
If you boot ShredOS and once it's up, switch to a virtual terminal ALT F2 type `fdisk -l` what's the output?

PartialVolume, 2022-02-07:
According to the Dell specs for a Optiplex 3040m it uses a power adapter of 65 W | 19.5 V DC, 3.34 A. That's not a large  PSU for powering 5 disks from. Of course it all depends what those discs are. Are the two internal discs 2.5”? I would not trust any of those cheap Chinese USB adapters either except for very new or very small drives. Any spinning platter drive over a certain age pulls quite a startup current which those cheap adapters that supposedly can deliver 2 amps don't work. If I connect a lot of drives to the USB ports I tend to use an external ATX 500W power supply and a suitable USB to SATA adapter that has a data only SATA connector.

Only recently I had to fix one of those cheap USB to SATA power adapters, the drives kept loosing power from what appeared to be a bad molex connector (it was rubbish) cut it  off to find that where it fed back into a single cable with three cores. They were basically using a shared ground cable for 5v and 12v so if you had a large initial current for spinning up the platter this would affect the 5v stability. Not only that they used cable that from the outside looked like an appropriate guage but when you cut and stripped it, it's guage was far from appropriate. Would work fine for small drives but power a 2TB drive and the drive wouldn't power up properly.

PartialVolume, 2022-02-07:
Of course you may have a Optiplex 3040 tower rather than a mini or midi.

ItsSour, 2022-02-08:
> Another thing to look for is whether you are overloading the Dells PSU, or USB. Are the USB drives obtaining power from the USB bus or do you have a external adapter.
> 
> I'm just wondering whether the discs are spinning up? All depends on what you are connecting and how old. I think a lot more detail maybe required on makes and models of drives, how they are powered.

They are externally powered, the USB is purely for a data connection, it's going through USB 3.0, so that data transmission should be fine, even the HDD connected through SATA is powered externally. I know they are all spinning up because I can feel them each vibrating a little from the platter spinning inside, and they all show up on different SATA configurations so I don't believe it is a power issue :)

ItsSour, 2022-02-08:
> Of course you may have a Optiplex 3040 tower rather than a mini or midi.

It is the tower version

PartialVolume, 2022-02-09:
@ItsSour no problem, glad you solved the problem. 👍