PartialVolume/shredos.x86_64 #89: Need .iso file for burning to USB flash drives

xmha97, 2022-01-28:
Hi, I need .iso file for burning to USB flash drives. (instead of .img file)

PartialVolume, 2022-01-28:
Why do you need an .iso for burning to USB drive?

xmha97, 2022-01-29:
All the bootable programs I have are in ISO format.
There is also more support for the ISO format.

PartialVolume, 2022-01-29:
The .img file is a bootable disc image. It only needs to be transferred  block by block to a USB stick.

On Linux you would do this with the program dd. On Windows you could use dd, Rufus or etcher. [https://github.com/PartialVolume/shredos.x86_64#windows-users](https://github.com/PartialVolume/shredos.x86_64#windows-users)

PartialVolume, 2022-01-29:
[dd for Windows](http://www.chrysocome.net/dd)

korpx, 2024-01-04:
ISO:s exist for ShredOS version v2021.08.2_23_x86-64_0.34 (Previous Release).
It would be great with some ISO:s for the latest release too.
Using Ventoy for multi OS boot sticks.

PartialVolume, 2024-01-04:
@korpx you know you can use a ShredOS .img on Ventoy.

korpx, 2024-01-04:
@PartialVolume I wasn't aware.
When I try it I end up in the grub prompt.

PartialVolume, 2024-01-04:
To determine the cause, I'd start with checking the integrity of the .img file that has been copied onto the Ventoy stick. 

So before you copy the .img file onto the Ventoy stick run a sha1sum check on the file, it should return `c0e7c6a1260979280543b95d6faca6addf7195ac` for version v2023.08.2_25.0_x86-64_0.35 If it matches, then copy onto the Ventoy stick and rerun the check on the .img that's on the Ventoy stick. This will check for any copy errors.

If the checksums all match, I'd start looking at bios settings.
I'm assuming the kernel never loads, so you don't see the kernel progress ever reach 100% or even indicate ShredOS is loading?

PartialVolume, 2024-01-04:
See ShredOS .img running on Ventoy [Video](https://github.com/PartialVolume/shredos.x86_64/assets/22084881/ab7277ec-e5f7-4d25-bf39-2cf0a4e7aa72) Also see #184

PartialVolume, 2024-01-04:
In the bios section I would still switch secure boot off and if there is any mention of OS, select other and not windows. What's the make and model of computer?

korpx, 2024-01-04:
Hash of the file on the USB stick matches C0E7C6A1260979280543B95D6FACA6ADDF7195AC
I am testing it on an old HP 8440p which runs the v2021.08.2_23_x86-64_0.34 (Previous Release) ISO from Ventoy fine.
No error code from grub.

Will check out the video and #184

korpx, 2024-01-04:
Oh, and yes there is no progress indicator that I can see. Drops straight to grub prompt.

closed 2026-02-14 by PartialVolume