PartialVolume/shredos.x86_64 #91: Checking if my HDD has bad sectors

MineGamer69, 2022-02-14:
I was wondering if this tool checks weather my hdd's have errors or bad sectors and are unusable. if so what method do i use to verify this.

PartialVolume, 2022-02-14:
It will check for write/read I/O errors and therefore bad sectors in the process of doing a wipe and verification. What it won't do is a non destructive read/write. I.e As long as you are happy to lose all the data on your disc it will do a pretty good test.

Generally a zeros, ones or PRNG with last pass verification will be sufficient to tell you whether your disc is failing. Or use all three methods, one after the other if you suspect it's an intermittent problem.

Or if you really want give the drive a very long test do a Gutmann wipe with verification on every pass, probably over the top but it would certainly test the drive non stop for a very, very long time. Not suitable for a SSD unless you want to kill the drive or at least reduce its life span somewhat.

Smart data, i e ALT F2 to switch to a console then `smartctl -a /dev/sda` (or sdb,sdc etc) is also useful.

ItsSour, 2022-02-15:
> I was wondering if this tool checks weather my hdd's have errors or bad sectors and are unusable. if so what method do i use to verify this.

As PartialVolume said, you can press alt+f2 to switch to console and comb through the `smartctl` command

A good thing to note is that it might not tell you there is a fault but it can help you predict where one could be. There is no standardization for the attributes either so you might have to do some research depending on the manufacturer. to give you a general guide on it all, you can open the attachment I posted and see if that'll help you out any.
[SmartCtlCommand.zip](https://github.com/PartialVolume/shredos.x86_64/files/8073043/SmartCtlCommand.zip)

MineGamer69, 2022-02-15:
Can i verify around 40 HDDs at once using this command?

PartialVolume, 2022-02-15:
Yes.

PartialVolume, 2022-02-15:
Once they are all verifying you can use the up and down arrow keys on the keyboard to move up and down the list of 40 drives to see the progress.

PartialVolume, 2022-02-15:
On the selection screen you would normally select a drive then press 'space' to enable it for [wipe] or [vrfy] then use down arrow to select next drive etc. As you want to wipe/verify a lot of drives just do a Control A and they will all be set to wipe or verify.

MineGamer69, 2022-02-16:
But the throughput is very low when I do this I get around 600MB/s total and among 8 drives I get around 90MB which is very slow it takes 5mins to reach 3% is there a faster option or a way to increase my throughput

PartialVolume, 2022-02-16:
>But the throughput is very low when I do this I get around 600MB/s total and among 8 drives I get around 90MB which is very slow it takes 5mins to reach 3% is there a faster option or a way to increase my throughput

Sounds like you are hitting a bottleneck in the hardware. What's the processor, how much memory and motherboard model? What interface is being used, are they all SATA? What's the make/model/size of the drives and what wipe method are you using. What's the final MB/s when you are only wiping a single drive?

MineGamer69, 2022-02-16:
Im using a HP Elite desk i believe its a intel core i3 8gb ram ddr4 its OEM mobo all SATA all are WD greenpower hdd, im using the 1 pass check. its around 500mb for 1 hdd but im trying to make it work at a mass scale of around 40 hdds but the speeds are way too slow

MineGamer69, 2022-02-16:
how much is the usual throughput you are getting?

MineGamer69, 2022-02-16:
intel core I3-8100 3.6GHz cpu 8gb ram

PartialVolume, 2022-02-16:
The drives (assuming a WD5000AVDS) theoretical maximum sustained throughput is about 130MB/s, likely spec'ed at the begining of the disc. It's average sustained throughput will be less than this as the speed slows as it approaches the end of the disc. So 90MB/s is in the ballpark. You could squeeze a little more by setting sync=0 but be aware nwipe won't do synchronization during the wipe and a failed drive will just show as a MB/s that drops almost to zero.

What throughput were you expecting for these drives?

MineGamer69, 2022-02-16:
What if i put i5-10500 with 16GB DDR4 and 128gb ssd for the OS to stored on it.

PartialVolume, 2022-02-16:
>its around 500mb

Can you clarify whether you mean you have 500Mb drives or you are getting throughput of 500Mb/s when you wipe one drive?

The 128GB SSD will make no difference. ShredOS is completely loaded into RAM, once it's booted you can completely remove the USB stick or CDROM and it will carry on working.

The additional memory might improve things at least initially it may make it look faster but once the wipe has completed the overall speed probably won't be much different.

Don't forget that the initial high throughput you see a drive produce is simply the CPU memory disc cache filling up. So the initial 8GB written to disc will be lightening fast but once the disc cache is full that's when you start to see the real disc speed. 90MB/s - 130MB/s is a limitation of the drive. Looking at the sustained specs of the WD green drives the bigger the drive becomes the slower the sustained rate. And the drive progressively slows from start to finish. That normal for all HDDs.

PartialVolume, 2022-02-16:
If you leave nwipe at the drive selection window, i.e not wiping then ALT F2 to bring up the second virtual terminal. Type `nwipe --sync=0` then wipe a drive with zeros, last pass verification and switch off blanking pass, i.e it will right zeros then verify, what's the final MB/s for that drive?

PartialVolume, 2022-02-16:
As an example of what you might expect for the fastest hard drive (not SSD) that you can currently buy. The Seagate Exos 2x14 dual actuator has a sustained speed of 524Mb/s. The WD green drives are nearly five times slower than the Seagate.

https://tech.hindustantimes.com/tech/news/the-world-s-fastest-hard-disk-drive-meet-seagate-s-exos-2x14-hdd-71621680861678.html

ItsSour, 2022-02-16:
> The 128GB SSD will make no difference. ShredOS is completely loaded into RAM, once it's booted you can completely remove the USB stick or CDROM and it will carry on working.

Slightly off topic, but since it is loaded directly into the RAM, once it is booted, I should still be able to do CTRL+C and restart ShredOS even if the thumb drive I booted off of isn't plugged in right?

PartialVolume, 2022-02-16:
Yes, that's correct.

MineGamer69, 2022-02-16:
> > its around 500mb
> 
> Can you clarify whether you mean you have 500Mb drives or you are getting throughput of 500Mb/s when you wipe one drive?
> 
> The 128GB SSD will make no difference. ShredOS is completely loaded into RAM, once it's booted you can completely remove the USB stick or CDROM and it will carry on working.
> 
> The additional memory might improve things at least initially it may make it look faster but once the wipe has completed the overall speed probably won't be much different.
> 
> Don't forget that the initial high throughput you see a drive produce is simply the CPU memory disc cache filling up. So the initial 8GB written to disc will be lightening fast but once the disc cache is full that's when you start to see the real disc speed. 90MB/s - 130MB/s is a limitation of the drive. Looking at the sustained specs of the WD green drives the bigger the drive becomes the slower the sustained rate. And the drive progressively slows from start to finish. That's normal for all HDDs.

Throughput of total 500mb/s and its divided among 8 drives showing around 60-70mb/s per drive

PartialVolume, 2022-02-16:
>Throughput of total 500mb/s and its divided among 8 drives showing around 60-70mb/s per drive

Thanks, if you are doing a wipe that generates a stream such as DoD short, a random stream will be generated for each drive which can slow things down. If you are using wipes that include random streams, you could try downloading the 32 bit version of ShredOS. Oddly enough I've found this can speed things up by 5-10%, at least on the random pass and verification.

If your just doing a static pass such as zeros or ones and verification then stick to the 64 bit version.

What's the 'top' command showing during a wipe. Also how many cores has your I3? I'm just wondering if the CPU is slowing things down due to 8 instances of the random number generator running.

So in summary to maximize the MB/s use 32 bit ShredOS if you are using wipes that contain a random pass and start nwipe using the --sync=0 option. `nwipe --sync`, assuming you are running it in the second virtual terminal. You can also edit grub.cfg and add `nwipe_options="--sync=0` to the kernel command line as discussed in the README.md to have the default nwipe (in ALT-F1) run with no fdatasync until the end.

That might hopefully get you from 60-70MB/s to 80-100MB/s

PartialVolume, 2022-02-16:
One other thing which may be more relevant. You may be hitting the maximum limit of 600-700MB/s (6 gigabits per seconds) of your SATA controller. Assuming my numbers are correct. I don't know enough about all the different options on motherboards but maybe the bottleneck is the SATA controller. On more expensive boards or server boards that are designed to hold many drives they have independent SATA controllers. Maybe one of the other guys that knows about data centre servers can jump in here.

closed 2023-03-27 by PartialVolume