PartialVolume/shredos.x86_64 #98: Build failure: missing file package/shredos/Config.in

jeho2a, 2022-03-22:
Hello, I'm trying to build shredos following the instructions in the README, but I'm getting an error about a missing file:
```
shredos.x86_64$ make shredos_defconfig
package/Config.in:1995: can't open file "package/shredos/Config.in"
make: *** [Makefile:1025: shredos_defconfig] Error 1
shredos.x86_64$ 
```

Any ideas?

PartialVolume, 2022-03-22:
To run make shredos_defconfig you need to be running from the directory above package/

Also maksure you didn't clone ShredOS as root otherwise your permissions will be wrong. Download as a non root user and run make as the same user.

PartialVolume, 2022-03-22:
I'm going to write a more detailed build guide over the next few days probably based around the latest Ubuntu. Just out of interest what Linux distro are you trying to build ShredOS on?

jeho2a, 2022-03-22:
> To run make shredos_defconfig you need to be running from the directory above package/
> 
> Also maksure you didn't clone ShredOS as root otherwise your permissions will be wrong. Download as a non root user and run make as the same user.

Thank you. As far as I can tell, both of these are correct: I am running from the directory above package/ and I am downloading and running as non-root. In case it's helpful, I attach a transcript. 

[transcript.txt](https://github.com/PartialVolume/shredos.x86_64/files/8326394/transcript.txt)

PartialVolume, 2022-03-22:
# You need to create the directory package/shredos and then create the empty Config.in
mkdir package/shredos
touch package/shredos/Config.in

So the commands to build should look like this.

```
sudo apt install git
sudo apt install build-essential   pkg-config   automake   libncurses5-dev   autotools-dev   libparted-dev   dmidecode   coreutils   smartmontools
cd package
cd shredos
touch package/shredos/Config.in
make shredos_defconfig
make
cd output/images
$ dd if=shredos-20200412.img of=/dev/sdx (20200412 will be the day you compiled, sdx is the USB flash drive)
```
Let me know if that works for you. I'm currently building on KDE Neon (Ubuntu 20.04).

PartialVolume, 2022-03-22:
@jeho2a @Firminator

 [Build Procedure](https://github.com/PartialVolume/shredos.x86_64/blob/master/README.md#compiling-shredos-and-burning-to-usb-stick-the-harder-way-) 

Please take a look at the updated build procedure and in particular the prerequisite software that must be installed to avoid errors when running `make`

jeho2a, 2022-03-23:
> @jeho2a @Firminator
>
> [Build Procedure](https://github.com/PartialVolume/shredos.x86_64/blob/master/README.md#compiling-shredos-and-burning-to-usb-stick-the-harder-way-)
>
> Please take a look at the [...]

I've successfully built now on Debian 11.2. Thanks for updating the build procedure. 

I'll be trying out the second half - making changes - soon, so I'll let you know if your instructions for those work too.

PartialVolume, 2022-03-23:
That's great, I'll see if I can add some more detail to the key ShredOS related files, locations and what the files are responsible for, sometime later today.

However, for starters, the directories package/nwipe and board/ShredOS are good places to take a look at. I'll provide notes on the relevance of the files in these directories.

One thing I do tend to do if making any changes is to generally run 'make clean' before running 'make' for a clean build. I know you can skip the `make clean` in some circumstances to speed up the build process but for me, when I've done that, the build has failed on several occasions. So although it can take a couple of hours to build ShredOS I always do a `make clean` before doing a `make`. This always seems to be a more reliable method for ending up with a working .img file.

closed 2022-03-27 by jeho2a