PartialVolume/shredos.x86_64 #110: USB Drives not detected on Lenovo Flex 3-1570 Laptop

jackfaubshner, 2022-08-15:
Hello,

I am trying to run ShredOS on the Lenovo Flex 3-1570 Laptop.

When I boot off ShredOS from a USB Drive, it only shows me my SATA SSD and no other devices.
If I unplug the SSD by opening the laptop, it detects no storage devices. I have tried 5 different USB sticks and they don't  show up.

I can boot up ShredOS from a USB drive, but even that USB drive doesn't show up in ShredOS.

Previously, I was using the Dell E6430 laptop and ShredOS worked fine in it with the same USB sticks.

Anyone know anything about this issue?

I am willing to further diagnose this issue and provide logs if required. I really want to run ShredOS on this laptop.

Regards,
Jack.

PartialVolume, 2022-08-16:
You don't mention whether after booting ShredOS and no USB devices having shown up you do a Control-C to restart nwipe. On many systems USB devices don't initially show up on the first run of nwipe. So try a Control-C to restart nwipe. Do the USB devices then show up?

If you have done the above and the USB devices still don't show up can you ALT F2 to display the second virtual terminal and post the output of the commands `lsusb` and `lspci`.

jackfaubshner, 2022-08-16:
Thank you for your response and your amazing work on this project, @PartialVolume.

My apologies, I should have mentioned it when I opened the issue. I did try using Control+C to restart nwipe but the USB sticks were still not detected.
On a side note, apart from USB sticks, I also tried external hard drives, which were not detected as well.

The light on the USB sticks lights up, it's just that ShredOS does not detect them for some reason.

Here is the output of the two commands you requested:

[image]

For the picture above, I removed the SSD and had plugged in three USB sticks.
Samsung 32GB Drive - USB 2.0 Port (This one had ShredOS in it)
Sandisk Cruzer Glide 3.0 16GB - USB 3.0 Port
SanDisk Extreme Go 64GB - USB 3.0 Port

Changing ports of the ShredOS stick does not help.

Thank you for helping. Willing to provide more logs to help diagnose this issue.

PartialVolume, 2022-08-16:
>Thank you for your response and your amazing work on this project, @PartialVolume.

Thank you, glad it's useful.

Just one more question to get the easy stuff out of the way. Can you check /boot/grub/grub.cfg and also /EFI/BOOT/grub.cfg and make sure there is no mention of --nousb on the Linux command line. If that entry was present it would produce the same effect you are seeing.

If --nousb is not present in grub.cfg then I believe we might have a missing driver issue. Do you know the release date of this laptop?

The 8087:8001 is a Intel integrated hub, and the 1d6b:0002 is the Linux root hub.

I'll dig a little deeper regarding the drivers tonight. If it's a very new intel chipset we could be missing that driver.

Can you also post the output of `dmesg | grep -i : more`. You may have to take a couple of pictures as it might be a long list, hence piping it to more so it will page the display. Space bar to continue (sorry if I'm teaching you to suck eggs) let me know if you don't need that level of detail unless you ask for it! 😊

jackfaubshner, 2022-08-16:
Thank you for your response once again @PartialVolume.

> Thank you, glad it's useful.
> 
> Just one more question to get the easy stuff out of the way. Can you check /boot/grub/grub.cfg and also /EFI/BOOT/grub.cfg and make sure there is no mention of --nousb on the Linux command line. If that entry was present it would produce the same effect you are seeing.

There's no mention of --nousb in my grub.cfg, here's a copy of it just in case something is awry:

```
set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3
}
```

> Can you also post the output of `dmesg | grep -i | more`

Is that command correct? It tells me how to use grep.
[image]

> (sorry if I'm teaching you to suck eggs) let me know if you don't need that level of detail unless you ask for it! 😊

It's all good, I am not really an expert on these things. In fact, thanks to you, I now know what the grep command does on Linux :)

PartialVolume, 2022-08-16:
Sorry, my mistake, I missed out the USB. It should have read `dmesg | grep -i usb | more`

jackfaubshner, 2022-08-17:
> Sorry, my mistake, I missed out the USB. It should have read `dmesg | grep -i usb | more`

Here are the new screenshots:

[image]

[image]

The setup is the same as before,
No SSD
Samsung 32GB Drive - USB 2.0 Port (This one had ShredOS in it)
Sandisk Cruzer Glide 3.0 16GB - USB 3.0 Port
SanDisk Extreme Go 64GB - USB 3.0 Port

Let me know if you need anything else. And thank you again for trying to help a stranger out.

PartialVolume, 2022-08-17:
Thanks, can you run the command 'dmesg -W'. Make sure it's a upper case W. This command will initially not print anything however it's waiting for new messages. With the dmesg -W still running plug in or unplug and then plug in a USB disc drive, try both hard drive and a flash drive. If you could post the output. Hopefully dmesg -W displays something similar to the contents of the image shown below.
[image]

PartialVolume, 2022-08-17:
Can you also post the output of `lsmod | grep hcd; lspci -nnk | grep "USB controller" -A2`

For this command it doesn't matter whether you have anything plugged in to the USB ports or not.

The output should look something like this.
[image]

jackfaubshner, 2022-08-18:
Hello,

Apologies for the late reply, yesterday was a very busy day for me.

> can you run the command 'dmesg -W'.

I did a `dmesg -W` but plugging in any of my USB sticks or the hard drive did not print anything. The light on the USB Stick lit up and my hard drive did start spinning, but nothing on the dmesg.  Even plugging out and in the drive with ShredOS in it did nothing.

[image]

> Can you also post the output of `lsmod | grep hcd; lspci -nnk | grep "USB controller" -A2`

[image]

Sorry about the picture being on an angle. I had to take from the side due to the reflections from the window.

PartialVolume, 2022-08-19:
From above it looks like the v2 USB driver is loaded and possibly the xhci_pci drivers for V3 USB. I'm assuming at least one USB port on this system has V3 USB.

So the next bit of info is going to be a little more difficult to obtain. I need the entire output of dmesg. The way I would do this is to output dmesg to a file then mount the USB stick and copy the file onto the USB stick. Once copied you can then access the dmesg.txt file in another system and then drag and drop or click on the section at the bottom of the comment box that says "Attach files by dragging and dropping, selecting or pasting them."

PartialVolume, 2022-08-19:
Here's how to do that in more detail.
```
dmesg > dmesg.txt
fdisk -l  (to see what drive and partition your USB stick is, in the example I'm assuming /dev/sdb1 but it may be different for you depending on how many disks are in the system)
mkdir usbstick
mount /dev/sdb1 usbstick (there should be a partition number listed in fdisk, so not just sdb but sdb1)
cp dmesg.txt usbstick/
umount usbstick
```

Give it a couple of seconds then pull out the USB stick and transfer it to your usual PC/laptop and drag and drop the file into a comment window below. Rather than list the contents it will show a hyperlink. This is so we don't fill the comments with upto a thousand lines of output from dmesg.

Also can you take a good look at all the options in the systems bios. Are there any options that reference USB, sometimes this might not be obvious, may mention xhci or ehci not just USB, maybe some screenshots of the options might be useful.

and finally do the USB ports work when a live ubuntu distro is booted on this system. I'm sort of assuming yes, but wanted confirmation it's only shredos and not other distros that also have this problem.

PartialVolume, 2022-08-20:
With reference to your bios settings specifically XHCI handoff, EHCI handoff and IOMMU  (if they exist) you may find this thread useful 
https://unix.stackexchange.com/questions/72625/why-is-usb-not-working-in-linux-when-it-works-in-uefi-bios

It may or may not apply for your system but worth checking your bios to see if those settings or something similar exists.

I don't know whether your bios has legacy or UEFI capability, it may not considering its age but if it does then it might also be worth trying both boot methods to see if USB then works. In the thread they talk about enabling the above three options and booting in legacy mode.

jackfaubshner, 2022-08-20:
Thank you for your help so far.

> From above it looks like the v2 USB driver is loaded and possibly the xhci_pci drivers for V3 USB. I'm assuming at least one USB port on this system has V3 USB.

That's right, this laptop has one USB2.0 port and two USB3.0 ports.

> So the next bit of info is going to be a little more difficult to obtain. I need the entire output of dmesg. The way I would do this is to output dmesg to a file then mount the USB stick and copy the file onto the USB stick. Once copied you can then access the dmesg.txt file in another system and then drag and drop or click on the section at the bottom of the comment box that says "Attach files by dragging and dropping, selecting or pasting them."
> Here's how to do that in more detail.
> 
> ```
> dmesg > dmesg.txt
> fdisk -l  (to see what drive and partition your USB stick is, in the example I'm assuming /dev/sdb1 but it may be different for you depending on how many disks are in the system)
> mkdir usbstick
> mount /dev/sdb1 usbstick (there should be a partition number listed in fdisk, so not just sdb but sdb1)
> cp dmesg.txt usbstick/
> umount usbstick
> ```
> 
> Give it a couple of seconds then pull out the USB stick and transfer it to your usual PC/laptop and drag and drop the file into a comment window below. Rather than list the contents it will show a hyperlink. This is so we don't fill the comments with upto a thousand lines of output from dmesg.

Okay, so, the output of `fdisk -l` bears nothing cause the problem isn't that the USB sticks don't get mounted, the problem is that ShredOS does not know if a USB stick or hard drive was plugged in to the USB ports (their lights light up though).

Luckily, SATA drives are detected and I was able to find an old SSD, format it to ext4, mount it, and copy the dmesg.txt.
(For some reason I could not mount my current SSD's NTFS drive. I specifically mounted it with the read write parameter but I was always greeted with a `cp: cannot create regular file 'dmesg.txt': Read-only file system`. Mounting an ext4 partition and copying the file worked perfectly fine.

Here's the dmesg.txt:
[dmesg.txt](https://github.com/PartialVolume/shredos.x86_64/files/9387152/dmesg.txt)

> Also can you take a good look at all the options in the systems bios. Are there any options that reference USB, sometimes this might not be obvious, may mention xhci or ehci not just USB, maybe some screenshots of the options might be useful.
> With reference to your bios settings specifically XHCI handoff, EHCI handoff and IOMMU (if they exist) you may find this thread useful https://unix.stackexchange.com/questions/72625/why-is-usb-not-working-in-linux-when-it-works-in-uefi-bios
> 
> It may or may not apply for your system but worth checking your bios to see if those settings or something similar exists.
> 
> I don't know whether your bios has legacy or UEFI capability, it may not considering its age but if it does then it might also be worth trying both boot methods to see if USB then works. In the thread they talk about enabling the above three options and booting in legacy mode.

This BIOS is like bare bare bones, there are barely any options compared to desktop motherboards (or even some other laptops).
Here are all the pictures:

[image]

[image]
Toggling always on USB did not help with the issue.

[image]

[image]
There literally is no option for Legacy only mode. The two boot mode options are "UEFI" and "Legacy Support".
If I switch to Legacy Support, I get the options of "UEFI First" and "Legacy First".
Switching to "Legacy First" and booting ShredOS did not fix the issue.

[image]
Toggling OS Optimized Defaults did not help.

> and finally do the USB ports work when a live ubuntu distro is booted on this system. I'm sort of assuming yes, but wanted confirmation it's only shredos and not other distros that also have this problem.

I live booted Ubuntu 22.04.1 LTS and the USB Ports are working fine. They also work fine in Windows 11.

Regarding the dmesg, I am no expert but I see a few "New USB device found".
Is it possible that the USB Devices are detected but they are detected after ShredOS boots up? Like every time ShredOS restarts, they are detected after ShredOS is done loading the devices?

PartialVolume, 2022-08-20:
From dmesg it looks like an issue with xhci_hcd.
```
[    2.231808] xhci_hcd 0000:00:14.0: can't derive routing for PCI INT A
[    2.231810] xhci_hcd 0000:00:14.0: PCI INT A: no GSI
[    2.231829] xhci_hcd 0000:00:14.0: xHCI Host Controller
[    2.231861] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 2
[    2.232904] xhci_hcd 0000:00:14.0: hcc params 0x200077c1 hci version 0x100 quirks 0x000000000004b810
[    2.232911] xhci_hcd 0000:00:14.0: request interrupt -2147483648 failed
[    2.232912] xhci_hcd 0000:00:14.0: startup error -107
[    2.232914] xhci_hcd 0000:00:14.0: USB bus 2 deregistered
[    2.232947] xhci_hcd 0000:00:14.0: init 0000:00:14.0 fail, -107
[    2.232954] xhci_hcd: probe of 0000:00:14.0 failed with error -107
```

Can you place `amd_iommu=off` on the kernel command line in the two grub.cfg files and reboot ShredOS and try the USB ports again. Once nwipe appears it won't initially show any USB devices. Do a control c to exit nwipe, it will count down 4321..and restart nwipe. Hopefully the USB devices will show up.

jackfaubshner, 2022-08-21:
> From dmesg it looks like an issue with xhci_hcd.
> 
> ```
> [    2.231808] xhci_hcd 0000:00:14.0: can't derive routing for PCI INT A
> [    2.231810] xhci_hcd 0000:00:14.0: PCI INT A: no GSI
> [    2.231829] xhci_hcd 0000:00:14.0: xHCI Host Controller
> [    2.231861] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 2
> [    2.232904] xhci_hcd 0000:00:14.0: hcc params 0x200077c1 hci version 0x100 quirks 0x000000000004b810
> [    2.232911] xhci_hcd 0000:00:14.0: request interrupt -2147483648 failed
> [    2.232912] xhci_hcd 0000:00:14.0: startup error -107
> [    2.232914] xhci_hcd 0000:00:14.0: USB bus 2 deregistered
> [    2.232947] xhci_hcd 0000:00:14.0: init 0000:00:14.0 fail, -107
> [    2.232954] xhci_hcd: probe of 0000:00:14.0 failed with error -107
> ```
> 
> Can you place `amd_iommu=off` on the kernel command line in the two grub.cfg files and reboot ShredOS and try the USB ports again. Once nwipe appears it won't initially show any USB devices. Do a control c to exit nwipe, it will count down 4321..and restart nwipe. Hopefully the USB devices will then show up.

Quick question @PartialVolume, where do I place the `amd_iommu=off`?

I did:

`set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3
}

amd_iommu=off`

I also tried:
`set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3 amd_iommu=off
}`

USB Drives did not work in both cases.

Did I do it right or that's not where I was supposed to place the `amd_iommu=off`?

PartialVolume, 2022-08-21:
That looks good, however I gave you the wrong command, I just realised this is an Intel machine so the command should be `intel_iommu=off` and not `amd_iommu=off` making sure you do it in both /boot/grub/grub.cfg and /EFI/BOOT/grub.cfg.

Looking at the various posts on the internet this seems to be a predominately Lenovo issue with certain versions of Linux. If the above doesn't work then I'm wondering whether you should also try the previous version of ShredOS that used an earlier kernel to see if that makes any difference. [https://github.com/PartialVolume/shredos.i686/releases/tag/v2020.02.008_i686-0.30.001](https://github.com/PartialVolume/shredos.i686/releases/tag/v2020.02.008_i686-0.30.001) This is an older 32bit release running the 2021.02 version of buildroot. The issue may be in both but we have to check to rule it out.

I also double checked your BIOS version and you're running the current version so that rules out any BIOS updates to fix the problem as suggested in some posts.

PartialVolume, 2022-08-21:
One other thing, can you download [memtest86](https://www.memtest86.com/download.htm) or run it from ubuntu live if it appears on the initial boot screen and run a over night test on the memory. Just need to rule out any memory hardware errors as the cause.

jackfaubshner, 2022-08-22:
> That looks good, however I gave you the wrong command, I just realised this is an Intel machine so the command should be `intel_iommu=off` and not `amd_iommu=off` making sure you do it in both /boot/grub/grub.cfg and /EFI/BOOT/grub.cfg.

I changed it in both the grub.cfg

```set default="0"
set timeout="0"

menuentry "shredos" {
	linux /boot/shredos console=tty3 loglevel=3
}

intel_iommu=off```

Unfortunately, my USB drives are still not working.

> Looking at the various posts on the internet this seems to be a predominately Lenovo issue with certain versions of Linux.

When you said this is a Lenovo issue with certain version of Linux, I decided to try other Linux distros.
I came across [Tails](https://tails.boum.org) and as it turns out USB sticks do not work with it either. It boots up from the USB stick and is able to read / write to the USB stick, but the OS itself does not know the USB stick exists. Weird issue. Moreover, even the WiFi adapter is not detected.

> If the above doesn't work then I'm wondering whether you should also try the previous version of ShredOS that used an earlier kernel to see if that makes any difference. https://github.com/PartialVolume/shredos.i686/releases/tag/v2020.02.008_i686-0.30.001 This is an older 32bit release running the 2021.02 version of buildroot. The issue may be in both but we have to check to rule it out.

Tried it, does not work with this either. :(

> ne other thing, can you download [memtest86](https://www.memtest86.com/download.htm) or run it from ubuntu live if it appears on the initial boot screen and run a over night test on the memory. Just need to rule out any memory hardware errors as the cause.

memtes86 9.4 (Build 1000) reported no errors.

**May I know what exactly you searched for when looking up the issue with Lenovo laptops? Maybe I can find an answer somewhere hidden deep on the internet or maybe ask on the Lenovo forums.**

PartialVolume, 2022-08-22:
>May I know what exactly you searched for when looking up the issue with Lenovo laptops? Maybe I can find an answer somewhere hidden deep on the internet or maybe ask on the Lenovo forums.

Sure, basically do a search for `can't derive routing for PCI INT A` or `PCI INT A: no GSI`.

Here's a few links. The problem is none seem to get to the root cause of the issue that I can see.
[https://www.mail-archive.com/coreboot@coreboot.org/msg54684.html](https://www.mail-archive.com/coreboot@coreboot.org/msg54684.html)
[https://linux-kernel.vger.kernel.narkive.com/mjeosk8f/xhci-module-fails-when-booting-in-uefi-mode](https://linux-kernel.vger.kernel.narkive.com/mjeosk8f/xhci-module-fails-when-booting-in-uefi-mode)
[https://marc.info/?l=linux-kernel&m=136213525628621&w=2](https://marc.info/?l=linux-kernel&m=136213525628621&w=2)
[https://bugs.launchpad.net/dell-sputnik/+bug/1881121](https://bugs.launchpad.net/dell-sputnik/+bug/1881121)

There are many many more related links.

It's also not just Lenova but other makes too. The odd thing is I have around ten systems I test ShredOS on of various makes and vintages and the USB works on all of them. Very odd problem

>Any suggestions what can be tried next?

I need a better understanding of what xhci_hcd is doing and what `can't derive routing for PCI INT A` or `PCI INT A: no GSI` actually means. I guess it's an interrupt issue.

One thing I could do is rebuild ShredOS with the USB related drivers as modules rather than statically build into the kernel. Then we could try manually unloading and loading the drivers to see if it's a timing or sequence related issue.

jackfaubshner, 2022-08-23:
> Sure, basically do a search for `can't derive routing for PCI INT A` or `PCI INT A: no GSI`.
>
> Here's a few links. The problem is none seem to get to the root cause of the issue that I can see. [...]

Well, I tried looking for more and I was not able to find anyone who fixed it, only discussions of the problem.

> I need a better understanding of what xhci_hcd is doing and what `can't derive routing for PCI INT A` or `PCI INT A: no GSI` actually means. I guess it's an interrupt issue.
>
> One thing I could do is [...]

Is there any thing I could do to help?
Unfortunately, I have never coded at such a lower level before. I have only stuck with higher level programming.

PartialVolume, 2022-08-23:
I think the next step is probably me building ShredOS with the USB host controller as a separate module. This will allow us to load and unload the module and monitor how it behaves and what errors it produces.

I would then pass the Shredos .IMG over to you with a few tests to perform. This would involve loading the host controller module and monitoring dmesg for messages.

There may be a few days delay before I can get you that .IMG as I'm busy at work at the moment.

If anybody else wants to have a go at building ShredOS with the USB drivers as modules then please feel free.

jackfaubshner, 2022-08-24:
Well, thank you for all your help so far. So kind of you to help a stranger out.

Whenever you are ready, I will be ready to test things out. :)

PartialVolume, 2022-08-24:
No problem. I was just looking at the recent commits to this driver in the Linux kernel. Quite a few in 2022 so maybe I need to look at whether  I can update either the drivers or bump to the latest stable buildroot if that version includes those changes.

https://github.com/torvalds/linux/commits/master/drivers/usb/host/xhci.c

jackfaubshner, 2022-08-25:
> No problem. I was just looking at the recent commits to this driver in the Linux kernel. Quite a few in 2022 so maybe I need to look at whether I can update either the drivers or bump to the latest [...]

Are there any logs or other information I can provide from other distros? Cause USB drivers seem to be working fine in Ubuntu.

PartialVolume, 2022-08-25:
Yes, that would seem to point to it maybe not being a bug in the driver as the latest Ubuntu is using kernel 5.15 I believe, the latest being 5.19.

You could boot up Ubuntu and post the output of `/proc/modules` and `lsmod` which should show which USB drivers Ubuntu is using.

Also post the output of `uname -a` which will show and confirm the kernel version.

jackfaubshner, 2022-08-27:
> Yes, that would seem to point to it maybe not being a bug in the driver as the latest Ubuntu is using kernel 5.15 I believe, the latest being 5.19.
>
> You could boot up Ubuntu and post the output of [...]

Sorry about the late reply, been a busy week.

Here are all the logs:
[proc-modules.txt](https://github.com/PartialVolume/shredos.x86_64/files/9436984/proc-modules.txt)
[lsmod.txt](https://github.com/PartialVolume/shredos.x86_64/files/9436985/lsmod.txt)
[uname -a.txt](https://github.com/PartialVolume/shredos.x86_64/files/9436986/uname.-a.txt)

I have two 64GB USB sticks connected, one containing Ubuntu, other, blank (NTFS).

Have a nice weekend.

PartialVolume, 2022-08-27:
Interesting that the xhci_pci & xhci_pci_renesas are loaded under Ubuntu, but under ShredOS although these two drivers are build-in to the kernel there is no mention of them being loaded by dmesg. Instead we get ehci_pci (USB2.0) loaded.  Maybe I need to build those three as modules so we can manually unload echi_pci and load xhci_pci and xhci_pci_renesas manually. If that should get the USB working then at least we'll know we are on the right track and can then work out whether this is some quirk on this hardware.

However before I do anything can you change one thing in the BIOS. In theory it shouldn't make any difference, but I just want to be sure. Can you change 'always on USB' from disabled to enabled. Power cycle the laptop and boot ShredOS and see if the USB devices are recognised (after doing a Control-C on nwipe). (For others who might be reading this thread, USB devices **never** show in nwipe on the first run so you have to Control-C to restart nwipe to see USB devices)

PartialVolume, 2022-08-27:
@jackfaubshner Here's a link to ShredOS with loadable USB drivers. The link is available for 30 days. [ShredOS-Modified-with-loadable-USB-drivers](https://ufile.io/785arnka)

Please check the intergrity of the .img file with md5 or sha1.
```
For the file shredos-2021.08.2_22_x86-64_0.32.023_20220827.img
sha1 =  a1056772a6a4ad7085c3469154235c1632c25b08
md5  =  322d60e9d5d0d2bcb9d28415f0f6e9c7
```

Burn to a USB stick and boot.

Once it's running  go to the second virtual terminal ALT-F2 and type `modprobe xhci_pci` This command will load the USB driver.

Then type `lsmod`. You should see the following:
```
Module                         Size            Used by
xhci_pci                       16384           0
xhci_pci_renesas               16384           1 xhci_pci
xhci_hcd                       135168          1 xhci_pci
```

Makesure a USB drive is plugged in to a port and type `fdisk -l`. Hopefully your USB drive should be listed. If it is go back to ALT-F1 terminal where nwipe is running and Control C to exit it and press the spacebar to start the nwipe restart countdown 4,3,2,1 then nwipe will re-display. Fingers crossed the USB drive/s should be showing.

If the USB drives don't show up and you write dmesg to a file and attach to a comment like you did before.

jackfaubshner, 2022-08-28:
Hello, 

> However before I do anything can you change one thing in the BIOS. In theory it shouldn't make any difference, but I just want to be sure. Can you change 'always on USB' from disabled to enabled. [...]

I already tried this, it did not work unfortunately. :(

> @jackfaubshner Here's a link to ShredOS with loadable USB drivers. The link is available for 30 days. [ShredOS-Modified-with-loadable-USB-drivers](https://ufile.io/785arnka)
> 
> Please check the intergrity of the .img file with md5 or sha1.
> 
> ```
> For the file shredos-2021.08.2_22_x86-64_0.32.023_20220827.img
> sha1 =  a1056772a6a4ad7085c3469154235c1632c25b08
> md5  =  322d60e9d5d0d2bcb9d28415f0f6e9c7
> ```
> 
> Burn to a USB stick and boot.
> 
> Once it's running go to the second virtual terminal ALT-F2 and type `modprobe xhci_pci` This command will load the USB driver.
> 
> Then type `lsmod`. You should see the following:
> 
> ```
> Module                         Size            Used by
> xhci_pci                       16384           0
> xhci_pci_renesas               16384           1 xhci_pci
> xhci_hcd                       135168          1 xhci_pci
> ```
> 
> Makesure a USB drive is plugged in to a port and type `fdisk -l`. Hopefully your USB drive should be listed. If it is listed, then go back to the ALT-F1 terminal where nwipe is running and Control C to exit nwipe and then press the spacebar to start the nwipe restart countdown 4,3,2,1 then nwipe will re-display. Fingers crossed the USB drive/s should now be showing up in nwipe's drive selection window.
> 
> If the USB drives don't show up can you write dmesg to a file and attach to a comment like you did before.

I am sorry to be the bringer of bad news but this did not work. :( (Don't shoot the messenger).

lsmod does give the same output as the sample output you provided but drives still do not work.

Here is the dmesg:
[dmesg2.txt](https://github.com/PartialVolume/shredos.x86_64/files/9438878/dmesg2.txt)

PartialVolume, 2022-08-28:
From dmesg I can see we get the same error.
```
23.428963] xhci_hcd 0000:00:14.0: can't derive routing for PCI INT A
[   23.428968] xhci_hcd 0000:00:14.0: PCI INT A: no GSI
[   23.428994] xhci_hcd 0000:00:14.0: xHCI Host Controller
[   23.429040] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 1
[   23.430082] xhci_hcd 0000:00:14.0: hcc params 0x200077c1 hci version 0x100 quirks 0x000000000004b810
[   23.430088] xhci_hcd 0000:00:14.0: request interrupt -2147483648 failed
[   23.430090] xhci_hcd 0000:00:14.0: startup error -107
[   23.430093] xhci_hcd 0000:00:14.0: USB bus 1 deregistered
[   23.430139] xhci_hcd 0000:00:14.0: init 0000:00:14.0 fail, -107
[   23.430147] xhci_hcd: probe of 0000:00:14.0 failed with error -107
```

jackfaubshner, 2022-08-29:
> From dmesg I can see we get the same error.

Yeah, I saw that.

I guess we have reached a dead end, eh?

PartialVolume, 2022-08-29:
No, there are still a few more things we can try. There is one or two other USB drivers that can be built as modules so they don't get loaded.

We can also build a newer kernel to see if that fixes the problem. This in particular is more time consuming so may take a while.

Looking at a number of threads, its interesting that they seem to come to the conclusion it's a bug in the bios ACPI PCI interrupt table or code associated with that. Some resolved the problem by updating their bios, unfortunately for your system there is no bios update. So in the kernel code they seem to have workarounds to solve the issue.

At the moment I'm sort of working on the theory that some other USB driver that loads first is causing the problem so by minimising what drivers are actually loaded maybe we can figure out what's causing the interrupt error when xhci_pci loads.

However if that doesn't resolve the problem I need to build the latest buildroot and see what kernel they are using, hopefully 5.19, but we will see.

One thing's for sure, I rarely give up unless I've exhausted all possible avenues. I'm not into delving deep into the kernel code though as I have plenty to do with the nwipe code. I'm hoping that somebody that knows the USB drivers inside and out could shed some light on why this particular Lenovo laptop has a issue where most others don't.

I'll shortly give you another link to ShredOS that has one or two other USB drivers built as modules which don't appear in the Ubuntu driver list.

jackfaubshner, 2022-08-30:
> One thing's for sure, I rarely give up unless I've exhausted all possible avenues.

Man! You really are one committed individual. I like it.

> I'll shortly give you another link to ShredOS that has one or two other USB drivers built as modules which don't appear in the Ubuntu driver list.

Thank you kind sir, please, take all the time you need.

Side note:
I am not the expert here but would it be a better idea to see what workaround other Linux distros are using?
I don't know how one would extract that information though.

PartialVolume, 2022-08-30:
>Side note:
I am not the expert here but would it be a better idea to see what workaround other Linux distros are using?
I don't know how one would extract that information though.

Yes, that is certainly something to be checked assuming there is a workaround and the bug isn't due to some other driver that's triggering the issue. All these are things to explore.

I've looked at the latest buildroot and the default kernel version is currently 5.17, the ShredOS version is 5.15 so it's worth trying that version out as well. (I've not built it yet but will do). I know there are changes in xhci_pci between 5.17 and 5.19, 5.19 being the latest release [https://github.com/torvalds/linux/tags](https://github.com/torvalds/linux/tags) so I'd really like to upgrade to 5.19 but may have to wait for buildroot to update to that release. I could use a custom kernel however I don't want to deviate from the work buildroot does as it could introduce other bugs I'm not expecting.

There is one more check that I'd like you to do with the modified ShredOS and that is to load the EHCI driver. The USB ports with the 'blue' inserts won't work but the USB ports with black ports should work, even with a USB 3.x device. The device itself will be backwards compatible. This just gives us a little more information on the problem. I'll follow up with details once I've checked out this test.

jackfaubshner, 2022-08-31:
> There is one more check that I'd like you to do with the modified ShredOS and that is to load the EHCI driver. The USB ports with the 'blue' inserts won't work but the USB ports with black ports [...]

Oh I already tried this. The USB 2.0 port does not work in any ShredOS, not even the modified one that you provided.
I try all the USB ports with all version of ShredOS I have tried so far.

Does the 2.0 port use different drivers?

PartialVolume, 2022-08-31:
>Does the 2.0 port use different drivers?

ohci-pci handles usb V1
echi-pci handles usb V2 (and V1)
xhci-pci handles usb V3 (and V1, V2).

At least I think that's how it works, but I'm no expert on USB drivers.

I don't think you need all three to have a working mix of V1/2/3. xhci-pci works with V2 ports, certainly on my systems without the other drivers being loaded.

PartialVolume, 2022-08-31:
Having thought about this and looked at buildroot's release schedule. I think the next course of action is to wait for buildroot to release 2022.08 which includes the 5.17 Linux kernel. Hopefully this will be sometime in September. Currently ShredOS is using 5.13 and and ubuntu 22.04 LTS is using 5.15 so I'm hoping 5.17 will resolve this issue. If it doesn't then I've not wasted any time as ShredOS is due for a kernel update anyway. In this way I hopefully kill two birds with one stone and it may give me enough time in get the start of HPA/DSO (hidden disk area) detection into nwipe. So the new ShredOS release will include an updated kernel and new features to nwipe.

jackfaubshner, 2022-09-01:
> ohci-pci handles usb V1 echi-pci handles usb V2 (and V1) xhci-pci handles usb V3 (and V1, V2).
>
> At least I think that's how it works, but I'm no expert on USB drivers.
>
> I don't think you need all [...]

I see.

Quick question (might be stupid), what if we load the echi-pci driver? That way, we can confirm whether it is an issue with the xhci-pci driver, all USB drivers or maybe something entirely else?

> Having thought about this and looked at buildroot's release schedule. I think the next course of action is to wait for buildroot to release 2022.08 which includes the 5.17 Linux kernel. Hopefully [...]

I had a thought and I decided to try older versions of Ubuntu.

I tried,

Ubuntu 18.04.6 LTS -> USB Ports work.
Kernel Version: `5.4.0-84-generic`
[dmesg Ubuntu 18.04.6.txt](https://github.com/PartialVolume/shredos.x86_64/files/9469891/dmesg.Ubuntu.18.04.6.txt)
[dmesg Ubuntu 18.04.6 - grep xhci.txt](https://github.com/PartialVolume/shredos.x86_64/files/9469892/dmesg.Ubuntu.18.04.6.-.grep.xhci.txt)

Ubuntu 20.04.4 LTS -> USB Ports works
Kernel Version: `5.13.0-30-generic`
[dmesg Ubuntu 20.04.4.txt](https://github.com/PartialVolume/shredos.x86_64/files/9469896/dmesg.Ubuntu.20.04.4.txt)
[dmesg Ubuntu 20.04.4 - grep xhci.txt](https://github.com/PartialVolume/shredos.x86_64/files/9469897/dmesg.Ubuntu.20.04.4.-.grep.xhci.txt)

I don't know if this is relevant information or not to diagnose this problem.

PartialVolume, 2022-09-02:
Interesting that Ubuntu 18.04 & 20.04 both show the same error as shown in ShredOS yet the USB works. Here's a version of [ShredOS ](https://ufile.io/4q5g5x3m) with manually loadable modules for ohci-pci, ehci-pci and xhci-pci.

Once ShredOS has boot I would start with just loading xhci-pci. `modprobe xhci-pci` and then fdisk -l and see if the USB drives now show up. To unload the driver `rmmod xhci-pci`. Then you could try ehci-pci. For what it's worth ehci-pci doesn't work on my hardware, hardware that's very recent even though it has V3 & V2 USB. Could be different on your system. However I guess the point of this test is that after having booted ShredOS there are initially no USB host drivers loaded, and you then manually load `modprobe xhci-pci`. Use `lsmod` to see that it has loaded along with some other associated/related drivers. This is probably how Ubuntu is loading the drivers, i.e. it doesn't load ohci or ehci first, it simply loads xhci-pci and that handles all versions of USB. Maybe we should also ignore the `can't derive routing for PCI INT A and  PCI INT A: no GSI` error, as it seems to be a benign error on your system, even they I don't get that error on my hardware. So probably easiest to use 'fdisk -l' to list the drives to see if the USB drives have appeared or alternatively, restart nwipe with the control-C and see if the drives have shown up. 

If it still doesn't work then I think kernel 5.17 is the way to go.

PartialVolume, 2022-09-02:
btw, the checksums for checking the integrity of that link on file  `shredos-2021.08.2_22_x86-64_0.32.023_20220830.img`
```
md5 = ade899b384164a6cd306570c5d1cd32b
sha1 = 151ca543dbce65c8552bf237e1714c2c4426c898
```

jackfaubshner, 2022-09-03:
> Interesting that Ubuntu 18.04 & 20.04 both show the same error as shown in ShredOS yet the USB works. Here's a version of [ShredOS ](https://ufile.io/4q5g5x3m) with manually loadable modules for ohci-pci, ehci-pci and xhci-pci.
> 
> Once ShredOS has booted I would start with just loading xhci-pci. `modprobe xhci-pci` and then `fdisk -l` and see if the USB drives now show up. To unload the driver `rmmod xhci-pci`. Then you could try ehci-pci. For what it's worth ehci-pci doesn't work on my hardware, hardware that's very recent even though it has V3 & V2 USB. Could be different on your system. However I guess the point of this test is that after having booted ShredOS there are initially no USB host drivers loaded, and you then manually load `modprobe xhci-pci`. Use `lsmod` to see that it has loaded along with some other associated/related drivers. This is probably how Ubuntu is loading the drivers, i.e. it doesn't load ohci or ehci first, it simply loads xhci-pci and that handles all versions of USB. Maybe we should also ignore the `can't derive routing for PCI INT A and PCI INT A: no GSI` error, as it seems to be a benign error on your system, even though I don't get that error on my hardware. So probably easiest to use 'fdisk -l' to list the drives to see if the USB drives have appeared or alternatively, restart nwipe with the control-C and see if the drives have shown up.
> 
> If it still doesn't work then I think kernel 5.17 is the way to go.

Hello,

I tried this new build (checksum was correct after the download).

First, I tried, `modprobe xhci-pci` and got the same error as before, nothing on `fdisk -l`.

Then, I unloaded all the xhci-pci drivers that showed up in `lsmod` using `rmmod`.

Repeated the process for ehci-pci and then for ohci-pci (still nothing on `fdisk -l` or by soft rebooting ShredOS (Ctrl + C).

You can see me going step by step in the dmesg as per the timestamps.

Also, ohci-pci and ehci-pci seem to load fine, but nothing is detected on the USBs.

Heck, even if I completely power off ShredOS, start again (hard reboot) and load just the ehci-pci (without loading xhci-pci), even then it doesn't work.

Here is the dmesg:
[dmesg3.txt](https://github.com/PartialVolume/shredos.x86_64/files/9482795/dmesg3.txt)

alessandrouw, 2022-11-23:
Hi everyone. I have the (almost) very same issue as OP, except for me the laptops are Dell Latitude 7450. I was going to start a new thread but I found this one.

I tried different 7450 laptops with different USB drives. I am not trying to erase the USB, I just want to export the log file. I do this regularly for other models without issue, but this model is not recognizing the USB - it doesn't show anything with `fdisk -l`

Any ideas?

PartialVolume, 2022-11-23:
I believe that's running the Wildcat LP USB controller. Can you run the command lspci and confirm that.

I'll check we have the correct drivers loaded and get back to you.

CONFIG_USB_EHCI_HCD CONFIG_USB_OHCI_HCD CONFIG_USB_SL811HS CONFIG_USB_UHCI_HCD

PartialVolume, 2022-11-23:
I see that both the Dell Latitude 7450 (@alessandrouw ) and @jackfaubshner Lenovo Flex 3 1570 both use the Intel Wildcat Point LP USB controller. Maybe we can kill two birds with one stone.

PartialVolume, 2022-11-23:
Looks like all the correct drivers are installed including the SL811HS (SL811_HCD).

Just double checking but are these laptops being used without a docking station? I read somewhere on Dells website that the rear USB ports are disabled if the laptop is plugged into a docking station.

alessandrouw, 2022-11-24:
> @alessandrouw I believe that's running the Wildcat LP USB controller. Can you run the command lspci and confirm that.
> 
> I'll check we have the correct drivers loaded and get back to you.
> 
> CONFIG_USB_EHCI_HCD CONFIG_USB_OHCI_HCD CONFIG_USB_SL811HS CONFIG_USB_UHCI_HCD

I'll try it tomorrow, as I don't have said laptop here with me.

> Looks like all the correct drivers are installed including the SL811HS (SL811_HCD).
> 
> Just double checking but are these laptops being used without a docking station? I read somewhere on Dells website that the rear USB ports are disabled if the laptop is plugged into a docking station.

The 7450 is not connected to a docking station or anything, really, just the power supply. I also tried all ports, including the ones on the side. USB is working, as it's the same port I used to load ShredOS.

alessandrouw, 2022-11-24:
Hi @PartialVolume, please check `lspci` results below. I could not export the code for obvious reasons, haha, so I hope a picture is enough.

[image]

PartialVolume, 2022-11-24:
> Hi @PartialVolume, please check `lspci` results below. I could not export the code for obvious reasons, haha, so I hope a picture is enough.
> 
> [image]
> 

Thanks. That confirms the Wildcat chipset.

I was reading about a Mint user who had the same problem with the wildcat chipset. After he upgraded to kernel 5.15 the USB devices then worked.

The latest stable release of buildroot 22.08  I believe has the 5.15 kernel, although I need to double check that. Currently ShredOS is based on 5.13

ShredOS is due for an upgrade to buildroot 22.08 so all I can suggest at the moment is to wait for the next release which will have the 5.15 kernel.

PartialVolume, 2022-11-25:
I hoping to get a release out before Christmas, however if you need those certificates quickly this is how I would go about it. 

I would have ShredOS export those certificates via FTP to a system on you LAN that's running a FTP server using the procedure here. 

https://github.com/PartialVolume/shredos.x86_64#transferring-nwipe-log-files-to-a-ftp-server

I have used this method several times and if you're wiping lots of systems it saves all that copying to and from USB stick.

PartialVolume, 2022-11-25:
If you have a go at setting up a FTP transfer of the logs and can't get it working,  by all means let me know and I'll help you out.

One thing I did notice that probably isn't clear in the readme is that if you copy the lftp command verbatim into the grub.cfg it assumes your FTP server has a directory called 'data'. If this isn't the case you should create a directory called 'data' in your FTP server's root directory otherwise it will fail to transfer.

PartialVolume, 2023-11-08:
Please reopen this issue if not fixed by 2023.08.2_25.0_x86-64_0.35

closed 2023-11-08 by PartialVolume

jackfaubshner, 2024-04-01:
> Please reopen this issue if not fixed by 2023.08.2_25.0_x86-64_0.35

Whelp. Unfortunately, I tried version 2023.08.2_25.0_x86-64_0.35 and I am still facing the same issue :(