PartialVolume/shredos.x86_64 #112: Kernel panic booting into shredos on Dell r440

Cody-Coleman, 2022-10-21:
When using any version after Shredos shredos-2020.05.017_x86-64_0.32.003_20211111.img (or iso) on a Dell R440, after loading the shredos into memory (so after the countdown reaches 100%) I get Kernal panic: 
Kernel panic - not syncing: Attempted to kill the idle task! 

Started with the latest release and I just kept trying earlier versions until I landed on the 05.017 version and I have had no issues with that. Not sure what other information you could use / need but this is very easy for me to reproduce. The other kicker is it's not all the Dell r440s doing this, and the bios/ firmware versions on those do not seem to matter (as two boxes with the same versions, one will work the other will not no matter what). But the r440s that this does work on seem to be the exception to the rule. 

[image]

PartialVolume, 2022-10-21:
Just wondering if you have tried toggling the x2APIC bios in in the processor section. According to the document below the virtualisation technology setting needs to be enabled to change the state of x2APIC. Are these settings different on the R440s that work as opposed to those that panic? Maybe worth trying to toggle the state of both these values and see if that makes any difference.

https://www.dell.com/support/manuals/en-uk/poweredge-r440/per440_bios_pub/processor-settings-details?guid=guid-ac6a729a-e608-44b1-affc-e024f94b28c9&lang=en-us

c0nsaw, 2023-03-13:
Same issue with Dell R740xd2

PartialVolume, 2023-03-13:
Can you add this to the *two* grub.cfg files in the kernel command line on the USB stick `initcall_blacklist=dw_i2c_init_driver` then give it another try. This seems to solve some Dell specific kernel panics.

jarthur22, 2023-03-21:
Same issue on PowerEdge R720

PartialVolume, 2023-03-21:
>On the Dell PowerEdge T640/04WYPY, BIOS 2.4.8 11/27/2019, Linux 5.4.14 (and 4.19.57) with
> unset `IRQ_REMAP` and `X86_X2APIC` crashes on start-up, when x2APIC is enabled in the
> firmware.

Does disabling x2APIC in the firmware (I assume they are talking about the equivalent the BIOS??). Are you able to disable x2APIC on one of these Dell servers and see if it boots ShredOS?

PartialVolume, 2023-03-21:
There is a x86_X2APIC option in the kernel and I don't have that enabled in ShredOS. If I build the current master with this x86_X2APIC option enabled are you able to test it on one of this Dell servers that are showing this problem?

If somebody can then I'll upload a second .img that provides X2_APIC support.

PartialVolume, 2023-03-21:
For reference x86_APIC is enabled in ShredOS by first enabling `ScaleMP vSMP` and then enabling the now visible `Support x2apic` and maybe also `Numascale NumaChip`.

PartialVolume, 2023-03-23:
If you you able to try the latest pre-release version thanks. 

[v2021.08.2_23.2_x86-64_0.34](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_23.2_x86-64_0.34)

chmuri, 2023-04-14:
@PartialVolume I have similar problem on dell r720xd with boot I have downloaded pre release but I can't boot from usb stick UEFI mode. Below screen with error.
[image]

PartialVolume, 2023-04-14:
@chmuri can you Alt F2 to display a virtual terminal?. Just wondering if the main terminal that nwipe runs in is hanging because of the raid controller. If you can bring up a virtual terminal then it's nothing to do with the direct rendering manager or graphics driver but might be due to the raid controllers.

If you can't bring up a terminal then maybe it's graphics. Can you telnet into the system after you get that message as per the instructions for headless systems. If it is graphics, ShredOS would have booted up behind the scenes. You will need to enable the telnet server and if you can login run dmesg and post the output.

Sorry, but I don't have access to any of that hardware so somebody is going to have to spend some time digging into this issue starting with the above methods.

GrimDemon, 2023-06-14:
Hi @PartialVolume, I have the same problem on a DELL R640. Too bad you don't make a x64 ISO image for pre-releases. This would make it much easier to analyze the problem for people who remotely connect to their servers via iDRAC (it does not support remote booting via removable media). I was unable to convert the IMG file into a form that would allow me to run it as a bootable ISO image.

PartialVolume, 2023-06-14:
@GrimDemon If you feel it may be useful I can build an .iso for the latest pre-release. May take a couple of days as I'm busy with other stuff at the moment. I'll let you know once it's been uploaded.

PartialVolume, 2023-07-02:
@OlivierFrancais @GrimDemon @chmuri Can you try the latest pre-release. This contains the very latest linux firmware [v2021.08.2_23.4_x86-64_0.34](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_23.4_x86-64_0.34).

What graphics hardware is provided on these servers?

chmuri, 2023-07-02:
Default graphics

PartialVolume, 2023-07-02:
The default graphics card on a Dell R650xs is apparently a Matrox G200, I'm not sure what graphics are on the Dell R440, the spec I looked at did not mention anything about graphics.

Anyway, guess what, ShredOS is missing the G200 KMS (kernel mode setting) driver. It's also missing the AST driver for servers.

I'll add those two extra drivers and publish a new pre-release for testing.

PartialVolume, 2023-07-02:
@c0nsaw For reference the Dell EMC PowerEdge R740xd2 also has a Matrox G200eW3 graphics controller. So hopefully the pre-release I'm about to release today that includes the Matrox G200 KMS driver will also fix the blank screen on this server too.

PartialVolume, 2023-07-02:
@Cody-Coleman @OlivierFrancais @GrimDemon @chmuri Can you try the latest pre-release which now has the Matrox G200 KMS DRM drivers, which these Dell servers use. In addition I also added the AST server drivers. Hopefully if all works well, this should fix the blank screen issue. 

[Pre-release Details](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_23.5_x86-64_0.34)

[.img file for burning to USB](https://github.com/PartialVolume/shredos.x86_64/releases/download/v2021.08.2_23.5_x86-64_0.34/shredos-2021.08.2_23.5_x86-64_0.34_20230702.img)

OlivierFrancais, 2023-07-05:
I tried on R440 and R650xs with the new img, same result. I see Booting 'shredos', the loader going to 100% and nothing after. I'm stuck on this, nothing blink, no tty

PartialVolume, 2023-07-05:
@OlivierFrancais If  you've not already tried this can you append nomodeset to the linux command line in /EFI/BOOT/grub.cfg and also in /boot/grub/grub.cfg and rebooting.

This stops the kernel switching to DRM graphics mode.

Also, do you have the option to legacy boot on that system? On my Dell system it's F12 then I get the option to legacy boot or EFI boot.

PartialVolume, 2023-11-08:
Please reopen thsi issue if not fixed by 2023.08.2_25.0_x86-64_0.35

closed 2023-11-08 by PartialVolume