PartialVolume/shredos.x86_64 #115: No boot

bikerjay, 2022-11-26:
Hey. Appreciate the work you did on this. It worked great on a Asus laptop I have. But, I can't get it to boot up on my HP. Stuck on Booting "shredos"? Maybe I'm doing something wrong.
Secure boot: disabled
USB boot: enabled
Network boot: disabled 

Thanks for your time

PartialVolume, 2022-11-27:
What model is the HP laptop?

PartialVolume, 2022-11-27:
'''
Appreciate the work you did on this
'''
Thanks. One other question, does it say ShredOS Booting 100% or doesn't it display any percentage progress info.

If no percentage info ever gets displayed this is grub having an issue loading the ShredOS image from USB for some reason.

However if it shows 100% and hangs it's possibly a frame buffer issue or it's having trouble unpacking the image or starting the kernel.

Before doing anything else I'd probably burn a copy of memtest 86 and run at least a full pass just to rule out a memory issue especially if other OSs boot ok. There could be a single bit failure that affects one OS and not another depending on how it's failing.

I don't know how old the HP is but if it's a 32 bit processor and you're running 64bit ShredOS, grub which is 32 bit will load the OS and stop at 100% "Booting ShredOS 100%" as the 64bit code won't run, so whether there is a progress displayed or not has some significants.

Having said all that I've got an old HP desktop that just won't boot from USB, ok from CD, just not USB and I've not figured out why yet.

KindaGothish, 2022-12-06:
Hello, 

I am also having this issue where ShredOS hangs after showing 100% boot. i am trying to run it on and Dell Poweredge R6525.
any help I can get would be greatly appreciated! thank you

CoodyCZ, 2023-01-08:
I have the same problem on latest version [v2021.08.2_23_x86-64_0.34] - x86_64 iso stuck on words Booting Shredos and x86_64 .img loaded to 100% but nothing is going on... only screen with 100% loading.

In previous version - [v2021.08.2_21_x86-64_0.32.023] everything works great

PartialVolume, 2023-01-08:
> I have the same problem on latest version [v2021.08.2_23_x86-64_0.34] - x86_64 iso stuck on words Booting Shredos and x86_64 .img loaded to 100% but nothing is going on... only screen with 100% loading.
> 
> In previous version - [v2021.08.2_21_x86-64_0.32.023] everything works great

Can you boot using the working version then let me know the output of lscpu, lspci and lsusb. Thanks.

PartialVolume, 2023-03-18:
@CoodyCZ @bikerjay @KindaGothish Please see [Systems that don't apparently boot with the latest ShredOS v2021.08.2_23_x86-64_0.34 #133](https://github.com/PartialVolume/shredos.x86_64/discussions/133) for a method to identify the offending driver and also a quick solution by adding various nomodeset commands to the grub.cfg files depending upon the graphics card you have.

PartialVolume, 2023-03-21:
Please see pre-release [v2021.08.2_23.1_x86-64_0.34](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_23.1_x86-64_0.34) which contains missing GPU firmware that was causing some systems not to boot.

PartialVolume, 2023-03-23:
If you you able to try the latest pre-release version thanks. 

[v2021.08.2_23.2_x86-64_0.34](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_23.2_x86-64_0.34)

PartialVolume, 2023-03-27:
Closing, please reopen if this is still an issue when using the latest pre-release version.

closed 2023-03-27 by PartialVolume