PartialVolume/shredos.x86_64 #116: nwipe 0.34 integration

sphings79, 2022-12-18:
Hi,

nwipe 0.34 is out since july.
Will there be a new release from your side?

regards

sphings

PartialVolume, 2022-12-18:
Hi,

Yes, I've delayed doing an update to 0.34, as 0.32.023 which ShredOS currently uses, includes the majority of changes found in 0.34, i.e the changes between 0.32 to 0.34. 

There will be an update as soon as I've added a significant feature update to nwipe. Once I've done that I'll release what's currently the pre-release of ShredOS but update with nwipe 0.35 which should include HPA/DCO detection.

Unfortunately, I've finding my day job & life in general is taking up a lot more of my time and so I've had to prioritise to the detriment of nwipe development. I'm hoping I can find more time to work on it in 2023.  I tend to work in fits and starts so there may be lags and then a lot of activity in a short period of time. Rest assured , I've not abandoned work the project.

sphings79, 2022-12-18:
Thx for the amazing fast reply and thx also for your great work.

closed 2022-12-18 by sphings79

PartialVolume, 2022-12-18:
Thanks, much appreciated.

PartialVolume, 2023-01-02:
Update ShredOS with nwipe v0.34 https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_23_x86-64_0.34