PartialVolume/shredos.x86_64 #117: Remaining time calculation - could be improved

jokdesign, 2022-12-19:
All in all the perfect successor to DBAN I have been looking for - Thanks a lot!

Suggestion for improvement on the calculation of time for "Remaining":

It seems the remaining time in statistics is calculated using the current throughput measured right at the moment.
This leads to inconsistent readings while shreding large disks (for a 16 TB disk between 70h and 155h).
In my case, it showed around 70 hours initially, towards the end of pass 1 it gradually increased to 155 hours, only to jump back to about 70 hours at the beginning of pass 2.

It seems in the main window the individual disks throughput is an average value over the whole process.
(initially about 244 MB/s, at the end of pass 1 only 194 MB/s - then at the start of pass 2 only very gradually increasing again)
I think in Statistics, the time remaining should be calculated with this "average over all" value for throughput.
That would give more realistic (and less erratic) estimates for the time remaining in statistics display - especially for large disks and multiple passes / rounds.

Thanks for considering for next release!

PartialVolume, 2022-12-19:
>All in all the perfect successor to DBAN I have been looking for - Thanks a lot!

Thanks, glad it's useful.

 >I  think in Statistics, the time remaining should be calculated with this "average over all" value for throughput.
That would give more realistic (and less erratic) estimates for the time remaining in statistics display - especially for large disks and multiple passes / rounds.

Thanks, I'll take a look at the code, can't say I've really noticed it before but thanks for letting me know. What version of shredos/nwipe are you using? as shown on nwipe's title bar.

PartialVolume, 2022-12-19:
I guess I probably haven't noticed it before because I tend to test with very much smaller drives and often tiny loop devices. Biggest drive for wiping I have is a 4TB and it's rare I wipe that. With the size of disc in mind I'll take a look at the code and the time remaining and see if those jumps in time remaining become exaggerated the larger the drive . I'll try and roll this into 0.35 along with some other smaller changes.

jokdesign, 2022-12-20:
Thanks for looking into that!
I seems on large disks the throughput is significantly lower when writing to the outer sectors as compared to the first (inner) ones during a pass.
I was also a bit surprised by that...

Here is what the title bar says on the session I have running:
ShredOS v2021.08.2.21_x86-64_0.32.023

PartialVolume, 2022-12-20:
>I seems on large disks the throughput is significantly lower when writing to the outer sectors as compared to the first (inner) ones during a pass.
I was also a bit surprised by that...

Not just large drives, there's some nice graphs produced by the likes of HDTach, this graph shows a 50% reduction in speed between outer and inner tracks, on a 1TB drive.

[image]

Source:https://www.raymond.cc/blog/measure-actual-hard-disk-perfomance-under-windows/2/

>Here is what the title bar says on the session I have running:
ShredOS v2021.08.2.21_x86-64_0.32.023

Thanks :+1: