PartialVolume/shredos.x86_64 #125: Disk wipe fails with error code -1

InAUGral, 2023-02-15:
Hi all, cant see any info about error codes if I search for them. What does it mean if a wipe shows "FAILURE error code -1"?
I was running a wipe on a 2TB HDD using the Fill with Zeros method.

PartialVolume, 2023-02-16:
Failure error code -1 is just a generic code meaning it failed. To find out why it failed (most probably an I/O error on writing you need to look at the summary table displayed at the end of the wipe after you have pressed any key to exit.

If you are running nwipe in a distro from the command line with the command `sudo ./nwipe  /dev/xxxx` nwipe will output the log information and summary table and error table to stdout. It doesn't by default generate a logfile. To make it generate a logfile so you can look at it later, use the command `sudo ./nwipe -l nwipe_log.txt /dev/xxxx`.

It's different if you are using nwipe from within [ShredOS](https://github.com/PartialVolume/shredos.x86_64). ShredOS automatically has nwipe generate and store it's logfile. If you are using the latest ShredOS you will find the logfile on the USB stick that you booted ShredOS from.

Here's the sort of output you will see in stdout and the logfile. Look at the error log table for whether it was a pass(write), verification, or fdatasync error. Also in the text above the table it will probably say I/O error when writing.

Any drives that produces any errors I would scrap, assuming there is no other cause like bad PSU or cables.

Log below anonymized using nwipe's -q option, hence the ????? and XXXXXX
```
sudo ./nwipe -q /dev/loop23

Waiting for wipe thread to cancel for /dev/loop23
[2023/02/16 00:38:09]    info: nwipe 0.34
[2023/02/16 00:38:09]    info: Linux version 5.19.0-32-generic (buildd@lcy02-am
                               d64-026) (x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1
                               ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for
                                Ubuntu) 2.38) #33~22.04.1-Ubuntu SMP PREEMPT_D
                               YNAMIC Mon Jan 30 17:03:34 UTC 2
[2023/02/16 00:38:09] warning: /dev/loop23 USB bridge, no pass-through support
[2023/02/16 00:38:09]  notice: Found /dev/loop23, VIRT, Loopback device,   1 MB, S/N=???????????????
[2023/02/16 00:38:09]  notice: bios-version = 1.07.02TNO
[2023/02/16 00:38:09]  notice: bios-release-date = 02/22/2022
[2023/02/16 00:38:09]  notice: system-manufacturer = Novatech
[2023/02/16 00:38:09]  notice: system-product-name = NP5x_NP6x_NP7xPNP
[2023/02/16 00:38:09]  notice: system-version = Not Applicable
[2023/02/16 00:38:09]  notice: system-serial-number = XXXXXXXXXXXXXXX
[2023/02/16 00:38:09]  notice: system-uuid = XXXXXXXXXXXXXXX
[2023/02/16 00:38:09]  notice: baseboard-manufacturer = Novatech
[2023/02/16 00:38:09]  notice: baseboard-product-name = NP5x_NP6x_NP7xPNP
[2023/02/16 00:38:09]  notice: baseboard-version = Not Applicable
[2023/02/16 00:38:09]  notice: baseboard-serial-number = XXXXXXXXXXXXXXX
[2023/02/16 00:38:09]  notice: baseboard-asset-tag = XXXXXXXXXXXXXXX
[2023/02/16 00:38:09]  notice: chassis-manufacturer = Novatech
[2023/02/16 00:38:09]  notice: chassis-type = Notebook
[2023/02/16 00:38:09]  notice: chassis-version = N/A
[2023/02/16 00:38:09]  notice: chassis-serial-number = XXXXXXXXXXXXXXX
[2023/02/16 00:38:09]  notice: chassis-asset-tag = XXXXXXXXXXXXXXX
[2023/02/16 00:38:09]  notice: processor-family = Core i7
[2023/02/16 00:38:09]  notice: processor-manufacturer = Intel(R) Corporation
[2023/02/16 00:38:09]  notice: processor-version = 12th Gen Intel(R) Core(TM) i7-12700H
[2023/02/16 00:38:09]  notice: processor-frequency = 2178 MHz
[2023/02/16 00:38:09]  notice: Opened entropy source '/dev/urandom'.
[2023/02/16 00:38:09]  notice: hwmon: Module drivetemp loaded, drive temperatures available
[2023/02/16 00:38:45]  notice: Program options are set as follows...
[2023/02/16 00:38:45]  notice:   autonuke = 0 (off)
[2023/02/16 00:38:45]  notice:   autopoweroff = 0 (off)
[2023/02/16 00:38:45]  notice:   banner   = nwipe 0.34
[2023/02/16 00:38:45]  notice:   prng     = Isaac64
[2023/02/16 00:38:45]  notice:   method   = PRNG Stream
[2023/02/16 00:38:45]  notice:   quiet    = 1
[2023/02/16 00:38:45]  notice:   rounds   = 1
[2023/02/16 00:38:45]  notice:   sync     = 100000
[2023/02/16 00:38:45]  notice:   verify   = 1 (last pass)
[2023/02/16 00:38:45]  notice: /dev/loop23 has serial number ???????????????
[2023/02/16 00:38:45]  notice: /dev/loop23, sect/blk/dev 512/4096/1048576
[2023/02/16 00:38:45]  notice: Invoking method 'PRNG Stream' on /dev/loop23
[2023/02/16 00:38:45]  notice: Starting round 1 of 1 on /dev/loop23
[2023/02/16 00:38:45]  notice: Starting pass 1/1, round 1/1, on /dev/loop23
[2023/02/16 00:38:45]  notice: Initialising ISAAC-64 prng
[2023/02/16 00:38:45]  notice: prng stream is active
[2023/02/16 00:38:45]  notice: 1048576 bytes written to /dev/loop23
[2023/02/16 00:38:45]  notice: Finished pass 1/1, round 1/1, on /dev/loop23
[2023/02/16 00:38:45]  notice: Finished final round 1 of 1 on /dev/loop23
[2023/02/16 00:38:45]  notice: Blanking device /dev/loop23
[2023/02/16 00:38:45]  notice: 1048576 bytes written to /dev/loop23
[2023/02/16 00:38:45]  notice: Verifying that /dev/loop23 is empty.
[2023/02/16 00:38:45]  notice: 1048576 bytes read from /dev/loop23
[2023/02/16 00:38:45]  notice: [SUCCESS] Verified that /dev/loop23 is empty.
[2023/02/16 00:38:45]  notice: [SUCCESS] Blanked device /dev/loop23

******************************** Error Summary *********************************
!   Device | Pass Errors | Verifications Errors | Fdatasync I\O Errors
--------------------------------------------------------------------------------
    loop23 |           0 |                    0 |                    0
[... cut ...]
!   Device | Status | Thru-put | HH:MM:SS | Model/Serial Number
--------------------------------------------------------------------------------
    loop23 | Erased |   0 B/s | 00:00:00 | Loopback device/???????????????
--------------------------------------------------------------------------------
[2023/02/16 00:38:48] Total Throughput   0 B/s, PRNG Stream, 1R+B+VL
********************************************************************************

[2023/02/16 00:38:48]    info: Nwipe successfully completed. See summary table for details
```

PartialVolume, 2023-02-16:
And if you are running the latest ShredOS by booting from a CD/DVD you can still save the logfile by either, plugging a Fat formatted USB stick into the computer before you hut the spacebar to exit nwipe. Nwipe will find the USB fat formatted drive and write the logfile to it. As long as you haven't rebooted ShredOS the logfiles will still exist in the virtual RAM disc. ALT F2 to switch to the virtual terminal, then type `ls` and you should see the log files. ALT F1 to get back to nwipe.

InAUGral, 2023-02-16:
OK thank you for the information. I had a feeling it might be I/O error related as between the time of me posting this and your response I remembered it a HDD with existing I/O Issues. Will run the ShredOS wipe again and use the above steps to see the log for curiosity sake.

closed 2023-02-16 by InAUGral