PartialVolume/shredos.x86_64 #131: No boot as well (v0.34)

JakobPP, 2023-03-08:
Hello

Thanks for your effort and project.

As many others Ive had issue with booting the newest version. Its just stuck at no boot and then nothing. ALT+F2 or F3 doesnt bring up terminal.
Secure boot disabled and all that other yada yada. Ive tried both 32 and 64 bit, IMG and ISO files.
The same second i booted v0.32.023 it worked without issues. Im using a USB with VENTOY.

Ive only tried with one machine, AMD Ryzen 5 2600, 32GB RAM, WD BLACK NVMe and the disk to be wiped connected via USB SATA adapter.

PartialVolume, 2023-03-08:
>Thanks for your effort and project.

No problem, hope we can figure out the issue.

> As many others Ive had issue with booting the newest version. Its just stuck at no boot and then nothing. ALT+F2 or F3 doesnt bring up terminal. Secure boot disabled and all that other yada yada. [...]

When you say v0.32.023 worked, was that this release v2021.08.2_**22**_x86-64_0.32.023 (the pre-release)
[https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_22_x86-64_0.32.023](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_22_x86-64_0.32.023)

or this release v2021.08.2_**21**_x86-64_0.32.023? (I'm guessing probably this one).
[https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_21_x86-64_0.32.023](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_21_x86-64_0.32.023)

The difference being that `v2021.08.2_22_x86-64_0.32.023` as opposed to `v2021.08.2_21_x86-64_0.32.023` has lots of changes to the display side of things with the use of DRM drivers, as does the latest 0.34 release.

- Do you see the progress percentage on screen as grub is loading ShredOS?
- What's the graphics hardware on your system, has it got more than one graphics card, i.e onboard the motherboard or a discrete graphics card or both?
- How many video output ports do you have, VGA, DVI, HDMI?
- If you have a graphics with dual DVI ports have you switched the video lead to the other port?
- If you have both onboard and discrete graphics hardware have you tried the other, i.e VGA onboard, DVI discrete.
- If you follow the instructions on accessing nwipe via telnet on a headless system, this would be really useful to know, if you can login via telnet and run nwipe, it would tell me that ShredOS is booting fully and it's a display hardware issue rather than linux just failing to boot. [https://github.com/PartialVolume/shredos.x86_64#how-to-wipe-drives-on-headless-systems-or-systems-with-faulty-display-hardware-for-use-on-secure-lans-only](https://github.com/PartialVolume/shredos.x86_64#how-to-wipe-drives-on-headless-systems-or-systems-with-faulty-display-hardware-for-use-on-secure-lans-only)

PartialVolume, 2023-03-19:
@JakobPP will be fixed in the next release, we were missing some AMD GPU firmware.

JakobPP, 2023-03-21:
Hello

Sorry I havent gotten back to you. I finished my wipe sessions with the v2021.08.2_21_x86-64_0.32.023 version. Ill give the new version a try at some point when it gets released.

Thank you for your service.

PartialVolume, 2023-03-21:
@JakobPP 
>Ill give the new version a try at some point when it gets released.

It's already released, it's [here](https://github.com/PartialVolume/shredos.x86_64/releases/download/v2021.08.2_23.1_x86-64_0.34/shredos-2021.08.2_23.1_x86-64_0.34_20230320.img)

CoodyCZ, 2023-03-24:
Shredos version v2021.08.2_23.1_x86-64_0.34 in pre-release works with Ventoy and standalone from USB without any issues, the interface loaded successfully. It has been tested on AMD Ryzen, HP, Lenovo, and Triline (an older model).

PartialVolume, 2023-03-24:
> Shredos version v2021.08.2_23.1_x86-64_0.34 in pre-release works with Ventoy and standalone from USB without any issues, the interface loaded successfully. It has been tested on AMD Ryzen, HP, Lenovo, and Triline (an older model).

@CoodyCZ That's great ! The feedback and you time testing that it worked is much appreciated.

closed 2023-03-27 by PartialVolume