PartialVolume/shredos.x86_64 #140: ShredOS booting on non existing screen

MinecollYT, 2023-04-06:
The first time I used ShredOS on my HP Notebook I was hooked up to a docking station, everything seemed okay for that matter except that the GUI was only displayed on the monitors connected via DP. 

When I tried to run it for the again, this time without the docking station, there was no video output on my notebooks internal monitor. So I needed to hook up the docking station again, and there it was, the GUI only showing up on the second screen. 

My BIOS, Ventoy, ... is showing up on my internal monitor like expected. Is there a way to change the display output?

PartialVolume, 2023-04-07:
This sounds like it's caused by the Linux digital rendering manager DRM/KMS. Can you plug into the docking station then once ShredOS has booted switch to the virtual terminal ALT F2 and pipe dmesg to a file `dmesg > dmesg.txt` and copy to the USB stick. Then post that file here. That will hopefully tell us what's going on and why it's not polling the displays and switching screens.

I don't know how much you know about the Linux command line but if you are not familiar with Linux take a look in the ShredOS README.md file in the section about copying log files to the USB stick.

I'm only guessing here as I've not used Ventoy but maybe it's using the basic frame buffer with DRM/KMS which is why it renders to the internal screen. I'm guessing it doesn't show up on the second screen ?

We could be missing some driver specific to your internal screen maybe, dmesg should hopefully give us a clue as to what's going on.

PartialVolume, 2023-04-07:
One other thing it might be,  in the laptops bios is there a setting for designating which is the primary video output. If there is try switching it and see if ShredOS then renders to the internal screen.

MinecollYT, 2023-04-07:
> One other thing it might be, in the laptops bios is there a setting for designating which is the primary video output.

Sadly, there is no setting for changing the primary display in the bios.

> We could be missing some driver specific to your internal screen maybe, dmesg should hopefully give us a clue as to what's going on.

I'm not sure, it worked on my internal screen before I connected it to my docking station for the first time.

Here are the log files including dmesg one:
[dmesg.txt](https://github.com/PartialVolume/shredos.x86_64/files/11177040/dmesg.txt)
[nwipe_log_20230407-083205.txt](https://github.com/PartialVolume/shredos.x86_64/files/11177042/nwipe_log_20230407-083205.txt)

PartialVolume, 2023-04-07:
Can you confirm what graphics hardware is on your laptop. In dmesg it's complaining about missing i915 firmware but then mentions Nvidia.

>3.968837] nouveau 0000:01:00.0: NVIDIA GP108 (138000a1)
[    4.000471] nouveau 0000:01:00.0: bios: version 86.08.3b.00.0a
[    4.109243] fbcon: i915drmfb (fb0) is primary device
[    4.675846] i915 0000:00:02.0: Direct firmware load for i915/kbl_dmc_ver1_04.bin failed with error -2
[    4.675853] i915 0000:00:02.0: [drm] Failed to load DMC firmware i915/kbl_dmc_ver1_04.bin. Disabling runtime power management.
[    4.675855] i915 0000:00:02.0: [drm] DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915

MinecollYT, 2023-04-07:
My CPU (Intel Core i7-10510U-Prozessor) seems to have built in graphics "Intel UHD Graphics 620" and a seperat one "NVIDIA GeForce MX250 (2 GB DDR5 dediziert)"

PartialVolume, 2023-04-07:
That could explain it, unfortunately I've not got access to my development system until next week so can't progress it until then. But does look like missing firmware.

MinecollYT, 2023-04-07:
No problem, I'm done with my work. I just wanted to report it to you so others don't have the same problem. Happy Easter! :)

PartialVolume, 2023-04-10:
@MinecollYT Can you confirm which version you are using. The latest pre-release, published 3 weeks ago contains the i915 firmware. [https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_23.2_x86-64_0.34](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_23.2_x86-64_0.34)

MinecollYT, 2023-04-10:
I think I was using the latest stable release v2021.08.2_23_x86-64_0.34

PartialVolume, 2023-04-10:
> I think I was using the latest stable release v2021.08.2_23_x86-64_0.34

Yes, that version is missing the i915/kbl_dmc_ver1_04.bin firmware. If you could try the latest pre-release that includes that firmware and let us if that fixed the issue. Thanks.

MinecollYT, 2023-04-18:
Hey, I'm sorry for not responding for a while. I'm currently a bit low on time, I might be able to test it this week