PartialVolume/shredos.x86_64 #146: ShredOS does not boot on Acer Aspire E1-510

multi-vit, 2023-05-15:
Hi,

I made a bootable USB stick with ShredOS, but it gets stuck whilst loading.

I see the black screen, the text announcing `Booting 'shredos'` followed by a new line showing the loading taking place and ending with `[ shredos 101.75MiB 100% 6.44MiB/s]` then it hangs.

I get no error messages and is not responsive to any key presses, it just stops at this point.

I have tried the following versions with the same result on all of them (the file size and transfer rates differ on the loading section but always reaches 100%):

shredos-2021.08.2_21_i586_0.32.023_20220126
shredos-2021.08.2_21_x86-64_0.32.023_20220123
shredos-2021.08.2_23.1_x86-64_0.34_20230320  (after reading about errors with Intel Firmware missing in 2021.08.2_21)
shredos-2021.08.2_23_i586_0.34_32-bit_20221231 (just in case the laptop is not 64 bit and that's why the above didn't work)

I am not 100% sure if the laptop is 64 bit or not, hence the attempts with different versions. The operating system (Windows - what looks like 7) boots but I can't login and find the system info (or erase it in other ways) as I've long forgotten the password, and just want to dispose of the laptop.

multi-vit, 2023-05-15:
Never mind! 
Decided to try one more version, just in case and sod's law - it worked!
Definitely more an issue with the laptop rather than your app.
Thank you so much for all your work on this as DBAN was not an option for me due to UEFI!

closed 2023-05-15 by multi-vit

PartialVolume, 2023-05-15:
Glad you got it working, sounds like it may have been a graphics issue. Which version of ShredOS did you get it working with in the end?

multi-vit, 2023-05-16:
shredos-2021.08.2_21_x86-64_0.32.023_20220123

Which ironically I think I listed as already tried - I re-downloaded (which I only realised because a (1) was appended to the file name) it thinking it was one I hadn't tried yet. You're possibly right on the graphics issue, when I tried to do Legacy BIOS mode (when trying to run DBAN before I found your great program!), there was an error about the graphics card (unrelated to DBAN or ShredOS because it was appearing before boot options), but in UEFI mode it wasn't happening and as I said, the OS was booting!

PartialVolume, 2023-05-17:
shredos-2021.08.2_21_x86-64_0.32.023_20220123 (v21) is the release prior to the introduction of the Linux direct rendering manager (DRM) being used for handling the screen. All versions after the version you found works use DRM for handling the display which allows ShredOS to send the hardware to sleep for unfreezing drives when commanding the drive to do a erase that's initiated by the drives firmware.

The version you found works uses the basic frame buffer for display.

If you have the time it would be really useful if you could take the lastest pre-release and boot using that then login remotely via the network into ShredOS and capture the dmesg output. This will tell us why the DRM graphics is failing. If you could take a look at the thread #133 it explains all about blank screens, with a possible workaround and how to capture the dmesg output which will help to identify the issue.

multi-vit, 2023-06-04:
Apologies for the slow reply, but I finally found some time to have a play with this today. I have had no luck I'm afraid!

I grabbed the latest pre-release and followed the instructions in the linked thread, but the device does not appear to be connected to the network as not only does the nmap command return no results for port 23 being open, but I also can't find the device on the network in general, so it's possible the ethernet port doesn't work on the device (I don't know if I have ever used it to know if it's working to be honest!). 

I don't know much about networking, but I had to run `nmap -p23 192.168.1.0/24 --open` to have any success. I know that's partly because my network is 192.168.1 instead of 192.168.0 but if I ran the stated command (from the linked thread) with the correct ip: `nmap -p23 192.168.1/24 --open`, nmap shows an error of `Failed to resolve "192.168.1". WARNING: No targets were specified, so 0 hosts scanned.`. Running `nmap 192.168.1.0/24` is how I know the laptop isn't showing on the network at all as I can see all the devices connected and account for all of them.

Let me know if I'm doing something wrong with nmap as I've used it before, or if you want me to try anything else!

PartialVolume, 2023-06-04:
Looks like you were doing everything ok, so `nmap -p23 192.168.1.0/24 --open` would have searched for the full range of 256 host IPs on your 192.168.1 subnet and identified any hosts with an open telnet port.

There may be an easier solution. I'm publishing a new pre-release sometime tomorrow. This release will have a new feature that copies the dmesg output to the USB stick, even if the screen appears blank.

All you will need to do is boot ShredOS then when it's finished loading give it a minute or so then type Control C wait 10 seconds then pull the stick and see if the flash drive has a dmesg.txt file on it.

If it does have a dmesg.txt file then post the contents here, if not then Linux may be crashing very early on in the boot sequence. Hopefully that dmesg.txt file will be present.

multi-vit, 2023-06-22:
Have just given this a go a few time but afraid I'm not getting a dmesg.txt file. Assuming it would write the file to the root of the stick but have checked all the directories too, just in case. Sorry I can't be more help!