PartialVolume/shredos.x86_64 #149: initial NTP or some clock time sync?

jamenlang, 2023-06-30:
All of these events happened within a matter of minutes, but the datetime changed drastically during the session. What can I do to prevent this? Can I instruct ShredOS to reach out to an ntp server before it starts logging things?

[2023/06/29 14:53:52]  notice: Found /dev/nvme0n1, NVME, CX2-8B256-Q11 NVMe LITEON 256GB, 256 GB, S/N=TW09F8D15508567M03J5
[2023/06/29 14:53:52]    info: Automatically enumerated 1 devices.
[2023/06/29 14:53:52]  notice: bios-version = 1.34.3
[2023/06/29 14:53:52]  notice: bios-release-date = 11/20/2022
[2023/06/29 14:53:52]  notice: system-manufacturer = Dell Inc.
[2023/06/29 14:53:52]  notice: system-product-name = Precision 3510
[2023/06/29 14:53:52]  notice: system-version = Not Specified
[2023/06/29 14:53:52]  notice: system-serial-number = 71YQMC2
[2023/06/29 14:53:52]  notice: system-uuid = 4c4c4544-0031-5910-8051-b7c04f4d4332
[2023/06/29 14:53:52]  notice: baseboard-manufacturer = Dell Inc.
[2023/06/29 14:53:52]  notice: baseboard-product-name = 0XC72F
[2023/06/29 14:53:52]  notice: baseboard-version = A00
[2023/06/29 14:53:52]  notice: baseboard-serial-number = /71YQMC2/CN1296368K004E/
[2023/06/29 14:53:52]  notice: baseboard-asset-tag = Not Specified
[2023/06/29 14:53:52]  notice: chassis-manufacturer = Dell Inc.
[2023/06/29 14:53:52]  notice: chassis-type = Laptop
[2023/06/29 14:53:52]  notice: chassis-version = Not Specified
[2023/06/29 14:53:52]  notice: chassis-serial-number = 71YQMC2
[2023/06/29 14:53:52]  notice: chassis-asset-tag = Not Specified
[2023/06/29 14:53:52]  notice: processor-family = Core i7
[2023/06/29 14:53:52]  notice: processor-manufacturer = Intel(R) Corporation
[2023/06/29 14:53:52]  notice: processor-version = Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz
[2023/06/29 14:53:52]  notice: processor-frequency = 2600 MHz
[2023/06/29 14:53:52]  notice: Opened entropy source '/dev/urandom'.
[2023/06/29 14:53:52]  notice: hwmon: Module drivetemp loaded, drive temperatures available
[2023/06/29 14:53:52]  notice: hwmon: nvme0n1 has temperature monitoring
[2023/06/29 14:54:10]  notice: Program options are set as follows...
[2023/06/29 14:54:10]  notice:   autonuke = 0 (off)
[2023/06/29 14:54:10]  notice:   autopoweroff = 0 (off)
[2023/06/29 14:54:10]  notice:   banner   = ShredOS v2021.08.2_23.3_x86-64_0.34
[2023/06/29 14:54:10]  notice:   prng     = Isaac64
[2023/06/29 14:54:10]  notice:   method   = DoD Short
[2023/06/29 14:54:10]  notice:   quiet    = 0
[2023/06/29 14:54:10]  notice:   rounds   = 1
[2023/06/29 14:54:10]  notice:   sync     = 100000
[2023/06/29 14:54:10]  notice:   verify   = 1 (last pass)
[2023/06/29 14:54:10]  notice: /dev/nvme0n1 has serial number TW09F8D15508567M03J5
[2023/06/29 14:54:10]  notice: /dev/nvme0n1, sect/blk/dev 512/4096/256060514304
[2023/06/29 14:54:10]  notice: Invoking method 'DoD Short' on /dev/nvme0n1
[2023/06/29 14:54:10]  notice: Starting round 1 of 1 on /dev/nvme0n1
[2023/06/29 14:54:10]  notice: Starting pass 1/3, round 1/1, on /dev/nvme0n1
[2023/06/29 15:21:03]  notice: 256060514304 bytes written to /dev/nvme0n1
[2023/06/29 15:21:03]  notice: Finished pass 1/3, round 1/1, on /dev/nvme0n1
[2023/06/29 15:21:03]  notice: Starting pass 2/3, round 1/1, on /dev/nvme0n1
[2023/06/29 15:55:34]  notice: 256060514304 bytes written to /dev/nvme0n1
[2023/06/29 15:55:34]  notice: Finished pass 2/3, round 1/1, on /dev/nvme0n1
[2023/06/29 15:55:34]  notice: Starting pass 3/3, round 1/1, on /dev/nvme0n1
[2023/06/29 15:55:34]  notice: Initialising ISAAC-64 prng
[2023/06/29 15:55:34]  notice: prng stream is active
[2023/06/29 16:14:25]  notice: 256060514304 bytes written to /dev/nvme0n1
[2023/06/29 16:14:25]  notice: Finished pass 3/3, round 1/1, on /dev/nvme0n1
[2023/06/29 16:14:25]  notice: Finished final round 1 of 1 on /dev/nvme0n1
[2023/06/29 16:14:25]  notice: Blanking device /dev/nvme0n1
[2023/06/29 16:43:51]  notice: 256060514304 bytes written to /dev/nvme0n1
[2023/06/29 16:43:51]  notice: Verifying that /dev/nvme0n1 is empty.
[2023/06/29 17:00:19]  notice: 256060514304 bytes read from /dev/nvme0n1
[2023/06/29 17:00:19]  notice: [SUCCESS] Verified that /dev/nvme0n1 is empty.
[2023/06/29 17:00:19]  notice: [SUCCESS] Blanked device /dev/nvme0n1

******************************** Error Summary *********************************
!   Device | Pass Errors | Verifications Errors | Fdatasync I\O Errors
--------------------------------------------------------------------------------
   nvme0n1 |           0 |                    0 |                    0
********************************************************************************

********************************* Drive Status *********************************
!   Device | Status | Thru-put | HH:MM:SS | Model/Serial Number
--------------------------------------------------------------------------------
   nvme0n1 | Erased | 169 MB/s | 02:06:09 | CX2-8B256-Q11 NVM/TW09F8D15508567M03J5
--------------------------------------------------------------------------------
[2023/06/30 08:42:09] Total Throughput 169 MB/s, DoD Short, 1R+B+VL
********************************************************************************

[2023/06/30 08:42:09]    info: Nwipe successfully completed. See summary table for details.

jamenlang, 2023-06-30:
My apologies, I think this was a different machine that did not have the --nowait option.

closed 2023-06-30 by jamenlang

PartialVolume, 2023-06-30:
No problem, NTP was added in the pre-release published 20/March/2023 https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2021.08.2_23.1_x86-64_0.34

After booting ShredOS the NTP server is contacted and usually the date & time is updated within a few seconds. However if you are very quick at starting nwipe or there is some issue contacting the server you can get a jump in the date/time when NTP updates.

I plan on having the date & time updating in real time some where on the title bar of nwipe. This would allow you to glance at the date/time and if it's not accurate you can wait for NTP to update it or manually update it if you are not connected to the internet. I think that would be useful especially if you require accurate date/time in the logs.